Zxilly/UA2F #2: 编译出错

pxw139, 2020-12-08:
4.19内核
```
make[3]: Entering directory '/workdir/openwrt/package/UA2F'
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/.built
touch /workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/.built_check
x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0=UA2F-2.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -o /workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/ua2f.o -c /workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/ua2f.c
/workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/ua2f.c: In function 'queue_cb':
/workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/ua2f.c:216:12: warning: implicit declaration of function 'nfq_nlmsg_put'; did you mean 'nfq_nlmsg_parse'? [-Wimplicit-function-declaration]
     nlh2 = nfq_nlmsg_put(buf, NFQNL_MSG_VERDICT, 10010);
            ^~~~~~~~~~~~~
            nfq_nlmsg_parse
/workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/ua2f.c:216:10: warning: assignment to 'struct nlmsghdr *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
     nlh2 = nfq_nlmsg_put(buf, NFQNL_MSG_VERDICT, 10010);
          ^
/workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/ua2f.c: In function 'main':
/workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/ua2f.c:414:9: warning: assignment to 'struct nlmsghdr *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
     nlh = nfq_nlmsg_put(buf, NFQNL_MSG_CONFIG, queue_number);
         ^
/workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/ua2f.c:422:9: warning: assignment to 'struct nlmsghdr *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
     nlh = nfq_nlmsg_put(buf, NFQNL_MSG_CONFIG, queue_number);
         ^
x86_64-openwrt-linux-musl-gcc -L/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib -L/workdir/openwrt/staging_dir/target-x86_64_musl/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -lmnl -lnetfilter_queue -znow -zrelro -o /workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/ua2f /workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/ua2f.o
/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/ua2f.o: in function `queue_cb':
ua2f.c:(.text+0x2ff): undefined reference to `nfq_nlmsg_put'
/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/ua2f.o: in function `main':
ua2f.c:(.text.startup+0x205): undefined reference to `nfq_nlmsg_put'
/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ua2f.c:(.text.startup+0x252): undefined reference to `nfq_nlmsg_put'
collect2: error: ld returned 1 exit status
Makefile:43: recipe for target '/workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/.built' failed
make[3]: Leaving directory '/workdir/openwrt/package/UA2F'
make[3]: *** [/workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/.built] Error 1
time: package/UA2F/compile#0.22#0.05#0.27
package/Makefile:111: recipe for target 'package/UA2F/compile' failed
make[2]: *** [package/UA2F/compile] Error 2
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2
Error: Process completed with exit code 2.
```
5.4内核
```
make[3]: Entering directory '/workdir/openwrt/package/UA2F'
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/.built
touch /workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/.built_check
x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0=UA2F-2.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -o /workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/ua2f.o -c /workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/ua2f.c
/workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/ua2f.c: In function 'queue_cb':
/workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/ua2f.c:216:12: warning: implicit declaration of function 'nfq_nlmsg_put'; did you mean 'nfq_nlmsg_parse'? [-Wimplicit-function-declaration]
     nlh2 = nfq_nlmsg_put(buf, NFQNL_MSG_VERDICT, 10010);
            ^~~~~~~~~~~~~
            nfq_nlmsg_parse
/workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/ua2f.c:216:10: warning: assignment to 'struct nlmsghdr *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
     nlh2 = nfq_nlmsg_put(buf, NFQNL_MSG_VERDICT, 10010);
          ^
/workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/ua2f.c: In function 'main':
/workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/ua2f.c:414:9: warning: assignment to 'struct nlmsghdr *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
     nlh = nfq_nlmsg_put(buf, NFQNL_MSG_CONFIG, queue_number);
         ^
/workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/ua2f.c:422:9: warning: assignment to 'struct nlmsghdr *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
     nlh = nfq_nlmsg_put(buf, NFQNL_MSG_CONFIG, queue_number);
         ^
x86_64-openwrt-linux-musl-gcc -L/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib -L/workdir/openwrt/staging_dir/target-x86_64_musl/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -lmnl -lnetfilter_queue -znow -zrelro -o /workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/ua2f /workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/ua2f.o
/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/ua2f.o: in function `queue_cb':
ua2f.c:(.text+0x2ff): undefined reference to `nfq_nlmsg_put'
Makefile:43: recipe for target '/workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/.built' failed
/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/ua2f.o: in function `main':
ua2f.c:(.text.startup+0x205): undefined reference to `nfq_nlmsg_put'
/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ua2f.c:(.text.startup+0x252): undefined reference to `nfq_nlmsg_put'
collect2: error: ld returned 1 exit status
make[3]: *** [/workdir/openwrt/build_dir/target-x86_64_musl/UA2F-2.0/.built] Error 1
make[2]: *** [package/UA2F/compile] Error 2
make[3]: Leaving directory '/workdir/openwrt/package/UA2F'
time: package/UA2F/compile#0.23#0.06#0.29
package/Makefile:111: recipe for target 'package/UA2F/compile' failed
make[2]: Leaving directory '/workdir/openwrt'
package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [world] Error 2
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
Error: Process completed with exit code 2.
```

pxw139, 2020-12-08:
使用的源码https://github.com/coolsnowwolf/lede
云编译的固件https://github.com/P3TERX/Actions-OpenWrt

Zxilly, 2020-12-19:
link出错了，这个依赖libnetfilter-queue和libmnl，你要确保这些也编译了

pxw139, 2020-12-19:
> link出错了，这个依赖libnetfilter-queue和libmnl，你要确保这些也编译了

我看到博客里写的会自动勾选....我再去试试看

Zxilly, 2020-12-19:
> > link出错了，这个依赖libnetfilter-queue和libmnl，你要确保这些也编译了
> 
> 我看到博客里写的会自动勾选....我再去试试看

这是openwrt的锅，他把这两个lib移到openwrt/packages去了，建议你本地编译

pxw139, 2020-12-19:
[image]
[image]
[image]
奇怪了,我看了看,前面两个是自动勾选,第三个手动勾选了,但是编译还是出错
我先尝试只编译这三个试试

Zxilly, 2020-12-19:
很有可能是编译顺序的问题，所以我建议你本地编译
第一次不行第二次就行了（

pxw139, 2020-12-19:
终于编译过去了
看来真的是编译顺序的问题
目前云编译是不能直接编译这玩意了
(但是云编译是真的舒服啊,自己玩游戏他不香吗x

pxw139, 2020-12-19:
> 终于编译过去了
> 看来真的是编译顺序的问题
> 目前云编译是不能直接编译这玩意了
> (但是云编译是真的舒服啊,自己玩游戏他不香吗x

艹,编译过去了这个插件,但是最后出错了,还得重新-J1编译一次

pxw139, 2020-12-19:
> 很有可能是编译顺序的问题，所以我建议你本地编译
> 第一次不行第二次就行了（

夭寿,第一次把前三个插件编译完了
第二次编译这个插件还是炸了
错误同一个

Zxilly, 2020-12-19:
你是不是没有更新源码，这10多天我一直在写这个，改了挺多地方的。

pxw139, 2020-12-19:
> 你是不是没有更新源码，这10多天我一直在写这个，改了挺多地方的。

源码一直追着更新的，惨啊
（我为了云编译正常每次都要把插件全部更新一遍的

pxw139, 2020-12-19:
是不是只需要几个依赖就可以跑了
大佬能不能直接发一下IPK文件?(x86,或者mt7621也有设备

Zxilly, 2020-12-19:
> 是不是只需要几个依赖就可以跑了
> 大佬能不能直接发一下IPK文件?(x86,或者mt7621也有设备

https://zxilly.coding.net/public/static/static/git/files
你看看行不行

pxw139, 2020-12-19:
测试完了，还是出错
我估计是我们用的源码不同的问题
依赖都有
提示这个
大佬你用的是哪一个，我编译一个固件先用上再说

Zxilly, 2020-12-19:
我知道什么问题了，你看这个
https://github.com/coolsnowwolf/lede/blob/71e335c0bef32249ea56f5a398f65ce600264217/package/libs/libnetfilter-queue/Makefile
这个库版本太低了，你用openwrt的官方源就没问题

pxw139, 2020-12-19:
终于成功了！
我  惨  我
直接把openwrt官方的包替换了过来
编译成功
不过之前用op官方sdk编译也出错
我还以为是同一个问题😂

Zxilly, 2020-12-20:
libnetfilter-queue的1.0.5更新上个月才合并到openwrt，这些三方的源码还没更新估计

closed 2020-12-20 by Zxilly

arbaleast, 2021-04-13:
> […](#)

你好，请问你替换的是openwrt/package/feeds/packages/libnetfilter-queue
这个文件夹下的文件吗？

Zxilly, 2021-04-13:
@arbaleast 现在应该不会再遇到这个问题了，不管你用的哪个源这个库的版本都应该跟上来了

Zxilly, 2021-04-13:
@arbaleast 我错了，lede还是没更新，建议你用openwrt

arbaleast, 2021-04-13:
好的