Zxilly/UA2F #6: mnl_cb_run 出错

SunBK201, 2021-02-18:
`Thu Feb 18 20:06:35 2021 user.notice : Try to start UA2F processor at [2445].
Thu Feb 18 20:06:35 2021 user.notice : UA2F processor start at [2445].
Thu Feb 18 20:06:35 2021 syslog.notice UA2F[2445]: Pipset inited.
Thu Feb 18 20:06:35 2021 syslog.notice UA2F[2445]: UA2F has inited successful.
Thu Feb 18 20:06:35 2021 syslog.err UA2F[2445]: Exit at breakpoint 10.
Thu Feb 18 20:06:35 2021 user.err : Child sucide.
Thu Feb 18 20:06:35 2021 user.notice : Try to start UA2F processor at [2447].
Thu Feb 18 20:06:35 2021 user.notice : UA2F processor start at [2447].
Thu Feb 18 20:06:35 2021 syslog.notice UA2F[2447]: Pipset inited.
Thu Feb 18 20:06:35 2021 syslog.notice UA2F[2447]: UA2F has inited successful.
Thu Feb 18 20:06:35 2021 syslog.err UA2F[2447]: Exit at breakpoint 10.
Thu Feb 18 20:06:35 2021 user.err : Child sucide.
Thu Feb 18 20:06:35 2021 user.notice : Try to start UA2F processor at [2449].
Thu Feb 18 20:06:35 2021 user.notice : UA2F processor start at [2449].
Thu Feb 18 20:06:35 2021 syslog.notice UA2F[2449]: Pipset inited.
Thu Feb 18 20:06:35 2021 syslog.notice UA2F[2449]: UA2F has inited successful.
Thu Feb 18 20:06:35 2021 syslog.err UA2F[2449]: Exit at breakpoint 10.
Thu Feb 18 20:06:35 2021 user.err : Child sucide.
Thu Feb 18 20:06:35 2021 user.notice : Try to start UA2F processor at [2451].
Thu Feb 18 20:06:35 2021 user.notice : UA2F processor start at [2451].
Thu Feb 18 20:06:35 2021 syslog.notice UA2F[2451]: Pipset inited.
Thu Feb 18 20:06:35 2021 syslog.notice UA2F[2451]: UA2F has inited successful.
Thu Feb 18 20:06:35 2021 syslog.err UA2F[2451]: Exit at breakpoint 10.
Thu Feb 18 20:06:35 2021 user.err : Child sucide.
Thu Feb 18 20:06:35 2021 user.notice : Try to start UA2F processor at [2453].
Thu Feb 18 20:06:35 2021 user.notice : UA2F processor start at [2453].
Thu Feb 18 20:06:35 2021 syslog.notice UA2F[2453]: Pipset inited.
Thu Feb 18 20:06:35 2021 syslog.notice UA2F[2453]: UA2F has inited successful.
Thu Feb 18 20:06:35 2021 syslog.err UA2F[2453]: Exit at breakpoint 10.
Thu Feb 18 20:06:35 2021 user.err : Child sucide.
Thu Feb 18 20:06:35 2021 user.notice : Try to start UA2F processor at [2455].
Thu Feb 18 20:06:35 2021 user.notice : UA2F processor start at [2455].
Thu Feb 18 20:06:35 2021 syslog.notice UA2F[2455]: Pipset inited.
Thu Feb 18 20:06:36 2021 syslog.notice UA2F[2455]: UA2F has inited successful.
Thu Feb 18 20:06:36 2021 syslog.err UA2F[2455]: Exit at breakpoint 10.
Thu Feb 18 20:06:36 2021 user.err : Child sucide.
Thu Feb 18 20:06:36 2021 user.notice : Try to start UA2F processor at [2457].
Thu Feb 18 20:06:36 2021 user.notice : UA2F processor start at [2457].
Thu Feb 18 20:06:36 2021 syslog.notice UA2F[2457]: Pipset inited.
Thu Feb 18 20:06:36 2021 syslog.notice UA2F[2457]: UA2F has inited successful.
Thu Feb 18 20:06:36 2021 syslog.err UA2F[2457]: Exit at breakpoint 10.
Thu Feb 18 20:06:36 2021 user.err : Child sucide.
Thu Feb 18 20:06:36 2021 user.notice : Try to start UA2F processor at [2459].
Thu Feb 18 20:06:36 2021 user.notice : UA2F processor start at [2459].
Thu Feb 18 20:06:36 2021 syslog.notice UA2F[2459]: Pipset inited.
Thu Feb 18 20:06:36 2021 syslog.notice UA2F[2459]: UA2F has inited successful.
Thu Feb 18 20:06:36 2021 syslog.err UA2F[2459]: Exit at breakpoint 10.
Thu Feb 18 20:06:36 2021 user.err : Child sucide.
Thu Feb 18 20:06:36 2021 user.notice : Try to start UA2F processor at [2461].
Thu Feb 18 20:06:36 2021 user.notice : UA2F processor start at [2461].
Thu Feb 18 20:06:36 2021 syslog.notice UA2F[2461]: Pipset inited.
Thu Feb 18 20:06:36 2021 syslog.notice UA2F[2461]: UA2F has inited successful.
Thu Feb 18 20:06:36 2021 syslog.err UA2F[2461]: Exit at breakpoint 10.
Thu Feb 18 20:06:36 2021 user.err : Child sucide.
Thu Feb 18 20:06:36 2021 user.notice : Try to start UA2F processor at [2463].
Thu Feb 18 20:06:36 2021 user.notice : UA2F processor start at [2463].
Thu Feb 18 20:06:36 2021 syslog.notice UA2F[2463]: Pipset inited.
Thu Feb 18 20:06:36 2021 syslog.notice UA2F[2463]: UA2F has inited successful.
Thu Feb 18 20:06:36 2021 syslog.err UA2F[2463]: Exit at breakpoint 10.
Thu Feb 18 20:06:36 2021 user.err : Child sucide.
Thu Feb 18 20:06:36 2021 user.notice : Try to start UA2F processor at [2465].
Thu Feb 18 20:06:36 2021 user.notice : UA2F processor start at [2465].
Thu Feb 18 20:06:36 2021 syslog.notice UA2F[2465]: Pipset inited.
Thu Feb 18 20:06:36 2021 syslog.notice UA2F[2465]: UA2F has inited successful.
Thu Feb 18 20:06:36 2021 syslog.err UA2F[2465]: Exit at breakpoint 10.
Thu Feb 18 20:06:36 2021 user.err : Child sucide.
Thu Feb 18 20:06:36 2021 user.notice : Try to start UA2F processor at [2467].
Thu Feb 18 20:06:36 2021 user.notice : UA2F processor start at [2467].
Thu Feb 18 20:06:36 2021 syslog.notice UA2F[2467]: Pipset inited.
Thu Feb 18 20:06:36 2021 syslog.notice UA2F[2467]: UA2F has inited successful.
Thu Feb 18 20:06:36 2021 syslog.err UA2F[2467]: Exit at breakpoint 10.
Thu Feb 18 20:06:36 2021 user.err : Child sucide.
Thu Feb 18 20:06:36 2021 user.notice : Try to start UA2F processor at [2469].
Thu Feb 18 20:06:36 2021 user.notice : UA2F processor start at [2469].
Thu Feb 18 20:06:36 2021 syslog.notice UA2F[2469]: Pipset inited.
Thu Feb 18 20:06:36 2021 syslog.notice UA2F[2469]: UA2F has inited successful.
Thu Feb 18 20:06:36 2021 syslog.err UA2F[2469]: Exit at breakpoint 10.
Thu Feb 18 20:06:36 2021 user.err : Child sucide.
Thu Feb 18 20:06:36 2021 user.notice : Try to start UA2F processor at [2471].
Thu Feb 18 20:06:36 2021 user.notice : UA2F processor start at [2471].
Thu Feb 18 20:06:36 2021 syslog.notice UA2F[2471]: Pipset inited.
Thu Feb 18 20:06:36 2021 syslog.notice UA2F[2471]: UA2F has inited successful.
Thu Feb 18 20:06:36 2021 syslog.err UA2F[2471]: Exit at breakpoint 10.
Thu Feb 18 20:06:36 2021 user.err : Child sucide.
Thu Feb 18 20:06:36 2021 user.err : Meet too many fatal error, no longer try to recover.
Thu Feb 18 20:06:36 2021 user.err : Exit at breakpoint 3.`

SunBK201, 2021-02-18:
`ret = mnl_cb_run(buf, ret, 0, portid, (mnl_cb_t) queue_cb, NULL);` 调用报错

Zxilly, 2021-02-19:
请阅读README

> 由于新加入的 CONNMARK 影响，编译内核时需要添加 NETFILTER_NETLINK_GLUE_CT flag，否则会出现 mnl_cb_run:Not supported 错误

closed 2021-02-19 by SunBK201

WordlessEcho, 2021-04-22:
不好意思，我在`make kernel_menuconfig`的时候发现Network packet filtering framework (Netfilter)是空的
[image]

Zxilly, 2021-04-22:
@WordlessEcho 好像是要先打开某个选项这里才会有东西，你找找
实在不行手动改KConfig吧

WordlessEcho, 2021-04-22:
这个选项里没有东西，就要返回上一层，进入菜单之前先按一下y选中Network packet filtering framework (Netfilter)本身。左边的`[ ]`变成`[*]`之后，再进来看。
[image]

第一次调整内核参数，还不太熟悉，吃了没有经验的亏。

之后看[作者的文章](https://learningman.top/archives/304)，找到Leaf发表的这条回复：
> 安装后执行 ua2f 出现 mnl_cb_run:Not supported，不知道问题在哪
你可以看见作者的回复：
> 你先 make kernel_menuconfig
> 然后 Networking support -> Networking options -> Network packet filtering framework (Netfilter) -> Core Netfilter Configuration -> NFQUEUE and NFLOG integration with Connection Tracking 打开

再看[Leaf的文章](https://blog.doreoom.com/736.html)
>这里要在`Core Netfilter Configuration`把下面这些选了之后才会出现`NFQUEUE and NFLOG integration with Connection Tracking`
```
-*- Netfilter NFNETLINK interface                                                                                                                
<*> Netfilter LOG over NFNETLINK interfacer ip_tables)                                                                                           
<*> Netfilter connection tracking support 
<*> Connection tracking netlink interface
```

希望能帮到后来者