Zxilly/UA2F #7: 与ShadowSocksR Plus+模块冲突

SunBK201, 2021-02-21:
开启ssr+后此模块无法修改UA

Zxilly, 2021-02-21:
因为这个模块修改了包的mark，而且因为设计原因，不可能像隔壁的xmurp-ua一样避开ssr plus所占用的mark位

Zxilly, 2021-02-21:
你想用UA2F的话大概只能二选一了

closed 2021-02-21 by SunBK201

izeroo, 2021-09-10:
luci-app-ssr-plus测试是兼容的，gfwlist模式

izeroo, 2021-09-10:
> luci-app-ssr-plus测试是兼容的，gfwlist模式

协议是Vless+rprx

SunBK201, 2021-09-10:
@izeroo 旧版本与ssrp冲突，新版本的ua2f修改了mark位，应该是避开了冲突

Zxilly, 2021-09-11:
@SunBK201 事实上我并没有修改。。。他们都只需要一个标记位，所以直接位运算就好了，但是我需要很多个标记，所以我是直接修改整个int的，应该不会有任何位被保护

SunBK201, 2021-09-11:
我测试实际的冲突情况是：80端口流量ua2f处理不到，其他端口可以，而ssrp的运行是正常的，@izeroo，你说的兼容是指ssrp运行正常，ua2f所有端口也能全部处理？

izeroo, 2021-09-11:
> 我测试实际的冲突情况是：80端口流量ua2f处理不到，其他端口可以，而ssrp的运行是正常的，@izeroo，你说的兼容是指ssrp运行正常，ua2f所有端口也能全部处理？
对。ssrp代理常用端口，gfwlist和绕过中国大陆IP都正常。我对eth0.2接口抓包，所有http协议的ua成功被修改了，包括1234之类的非标准端口和oicq协议。走代理的部分没有被修改，当然也不需要修改

SunBK201, 2021-09-11:
> > 我测试实际的冲突情况是：80端口流量ua2f处理不到，其他端口可以，而ssrp的运行是正常的，@izeroo，你说的兼容是指ssrp运行正常，ua2f所有端口也能全部处理？
> 
> 
> 
> 对。ssrp代理常用端口，gfwlist和绕过中国大陆IP都正常。我对eth0.2接口抓包，所有http协议的ua成功被修改了，包括1234之类的非标准端口和oicq协议。走代理的部分没有被修改，当然也不需要修改
> 
> 可能是因为vless+xtls-rprx-splice这个协议比较特殊吧，其他协议没有试过

那我的情况是跟你一样了