Zxilly/UA2F #8: exit at breakpoint 10

DFlinmo, 2021-03-19:
```
Fri Mar 19 22:04:27 2021 user.notice : Try to start UA2F processor at [9486].
Fri Mar 19 22:04:27 2021 user.notice : UA2F processor start at [9486].
Fri Mar 19 22:04:27 2021 syslog.notice UA2F[9486]: Pipset inited.
Fri Mar 19 22:04:27 2021 syslog.notice UA2F[9486]: UA2F has inited successful.
Fri Mar 19 22:04:34 2021 syslog.err UA2F[9486]: Catch SIGSEGV at breakpoint 9 and breakpoint2 4
Fri Mar 19 22:04:34 2021 syslog.alert UA2F[9486]: Meet fatal error, try to restart.
Fri Mar 19 22:04:34 2021 user.err : Child sucide.
Fri Mar 19 22:04:34 2021 user.notice : Try to start UA2F processor at [9487].
Fri Mar 19 22:04:34 2021 user.notice : UA2F processor start at [9487].
Fri Mar 19 22:04:34 2021 syslog.notice UA2F[9487]: Pipset inited.
Fri Mar 19 22:04:34 2021 syslog.notice UA2F[9487]: UA2F has inited successful.
Fri Mar 19 22:04:48 2021 syslog.info UA2F[9487]: UA2F has handled 8 http, 0 http 1.0, 0 noua http, 133 tcp. Set 0 mark and 1 nohttp mark in 14 seconds
Fri Mar 19 22:05:21 2021 syslog.info UA2F[9487]: UA2F has handled 16 http, 0 http 1.0, 0 noua http, 730 tcp. Set 1 mark and 1 nohttp mark in 47 seconds
Fri Mar 19 22:05:26 2021 syslog.info UA2F[9487]: UA2F has handled 32 http, 0 http 1.0, 0 noua http, 1292 tcp. Set 1 mark and 3 nohttp mark in 52 seconds
Fri Mar 19 22:06:04 2021 syslog.info UA2F[9487]: UA2F has handled 64 http, 3 http 1.0, 4 noua http, 2316 tcp. Set 1 mark and 4 nohttp mark in 1 minutes and 30 seconds
Fri Mar 19 22:07:01 2021 syslog.info UA2F[9487]: UA2F has handled 128 http, 5 http 1.0, 15 noua http, 7405 tcp. Set 2 mark and 5 nohttp mark in 2 minutes and 27 seconds
Fri Mar 19 22:10:03 2021 syslog.info UA2F[9487]: UA2F has handled 256 http, 7 http 1.0, 295 noua http, 30744 tcp. Set 65 mark and 16 nohttp mark in 5 minutes and 29 seconds
Fri Mar 19 22:15:06 2021 daemon.notice netifd: wan (6133): udhcpc: sending renew to 10.0.0.3
Fri Mar 19 22:15:06 2021 daemon.notice netifd: wan (6133): udhcpc: lease of 10.108.6.146 obtained, lease time 28800
Fri Mar 19 22:16:41 2021 syslog.info UA2F[9487]: UA2F has handled 512 http, 10 http 1.0, 407 noua http, 85062 tcp. Set 74 mark and 37 nohttp mark in 12 minutes and 7 seconds
Fri Mar 19 22:19:43 2021 syslog.info UA2F[9487]: UA2F has handled 1024 http, 11 http 1.0, 413 noua http, 95202 tcp. Set 76 mark and 49 nohttp mark in 15 minutes and 9 seconds
Fri Mar 19 22:21:38 2021 daemon.err uhttpd[2120]: luci: accepted login on / for root from 192.168.1.239
Fri Mar 19 22:21:43 2021 syslog.info UA2F[9487]: UA2F has handled 2048 http, 11 http 1.0, 582 noua http, 99681 tcp. Set 90 mark and 93 nohttp mark in 17 minutes and 9 seconds
Fri Mar 19 22:23:03 2021 user.notice : Try to start UA2F processor at [10161].
Fri Mar 19 22:23:03 2021 user.notice : UA2F processor start at [10161].
Fri Mar 19 22:23:03 2021 syslog.notice UA2F[10161]: Pipset inited.
Fri Mar 19 22:23:03 2021 syslog.notice UA2F[10161]: UA2F has inited successful.
Fri Mar 19 22:23:03 2021 syslog.err UA2F[10161]: Exit at breakpoint 10.
Fri Mar 19 22:23:03 2021 user.err : Child sucide.
Fri Mar 19 22:23:03 2021 user.notice : Try to start UA2F processor at [10162].
Fri Mar 19 22:23:03 2021 user.notice : UA2F processor start at [10162].
Fri Mar 19 22:23:03 2021 syslog.notice UA2F[10162]: Pipset inited.
Fri Mar 19 22:23:03 2021 syslog.notice UA2F[10162]: UA2F has inited successful.
Fri Mar 19 22:23:03 2021 syslog.err UA2F[10162]: Exit at breakpoint 10.
Fri Mar 19 22:23:03 2021 user.err : Child sucide.
Fri Mar 19 22:23:03 2021 user.notice : Try to start UA2F processor at [10163].
Fri Mar 19 22:23:03 2021 user.notice : UA2F processor start at [10163].
Fri Mar 19 22:23:03 2021 syslog.notice UA2F[10163]: Pipset inited.
Fri Mar 19 22:23:03 2021 syslog.notice UA2F[10163]: UA2F has inited successful.
Fri Mar 19 22:23:03 2021 syslog.err UA2F[10163]: Exit at breakpoint 10.
Fri Mar 19 22:23:03 2021 user.err : Child sucide.
Fri Mar 19 22:23:03 2021 user.notice : Try to start UA2F processor at [10164].
Fri Mar 19 22:23:03 2021 user.notice : UA2F processor start at [10164].
Fri Mar 19 22:23:03 2021 syslog.notice UA2F[10164]: Pipset inited.
Fri Mar 19 22:23:03 2021 syslog.notice UA2F[10164]: UA2F has inited successful.
Fri Mar 19 22:23:03 2021 syslog.err UA2F[10164]: Exit at breakpoint 10.
Fri Mar 19 22:23:03 2021 user.err : Child sucide.
Fri Mar 19 22:23:03 2021 user.notice : Try to start UA2F processor at [10165].
Fri Mar 19 22:23:03 2021 user.notice : UA2F processor start at [10165].
Fri Mar 19 22:23:03 2021 syslog.notice UA2F[10165]: Pipset inited.
Fri Mar 19 22:23:03 2021 syslog.notice UA2F[10165]: UA2F has inited successful.
Fri Mar 19 22:23:03 2021 syslog.err UA2F[10165]: Exit at breakpoint 10.
Fri Mar 19 22:23:03 2021 user.err : Child sucide.
Fri Mar 19 22:23:03 2021 user.notice : Try to start UA2F processor at [10166].
Fri Mar 19 22:23:03 2021 user.notice : UA2F processor start at [10166].
Fri Mar 19 22:23:03 2021 syslog.notice UA2F[10166]: Pipset inited.
Fri Mar 19 22:23:03 2021 syslog.notice UA2F[10166]: UA2F has inited successful.
Fri Mar 19 22:23:03 2021 syslog.err UA2F[10166]: Exit at breakpoint 10.
Fri Mar 19 22:23:03 2021 user.err : Child sucide.
Fri Mar 19 22:23:03 2021 user.notice : Try to start UA2F processor at [10167].
Fri Mar 19 22:23:03 2021 user.notice : UA2F processor start at [10167].
Fri Mar 19 22:23:03 2021 syslog.notice UA2F[10167]: Pipset inited.
Fri Mar 19 22:23:03 2021 syslog.notice UA2F[10167]: UA2F has inited successful.
Fri Mar 19 22:23:03 2021 syslog.err UA2F[10167]: Exit at breakpoint 10.
Fri Mar 19 22:23:03 2021 user.err : Child sucide.
Fri Mar 19 22:23:03 2021 user.notice : Try to start UA2F processor at [10168].
Fri Mar 19 22:23:03 2021 user.notice : UA2F processor start at [10168].
Fri Mar 19 22:23:03 2021 syslog.notice UA2F[10168]: Pipset inited.
Fri Mar 19 22:23:03 2021 syslog.notice UA2F[10168]: UA2F has inited successful.
[... 238 lines cut ...]
Fri Mar 19 22:23:03 2021 syslog.notice UA2F[10208]: Pipset inited.
Fri Mar 19 22:23:03 2021 syslog.notice UA2F[10208]: UA2F has inited successful.
Fri Mar 19 22:23:03 2021 syslog.err UA2F[10208]: Exit at breakpoint 10.
Fri Mar 19 22:23:03 2021 user.err : Child sucide.
Fri Mar 19 22:23:03 2021 user.notice : Try to start UA2F processor at [10209].
Fri Mar 19 22:23:03 2021 user.notice : UA2F processor start at [10209].
Fri Mar 19 22:23:03 2021 syslog.notice UA2F[10209]: Pipset inited.
Fri Mar 19 22:23:03 2021 syslog.notice UA2F[10209]: UA2F has inited successful.
Fri Mar 19 22:23:03 2021 syslog.err UA2F[10209]: Exit at breakpoint 10.
Fri Mar 19 22:23:03 2021 user.err : Child sucide.
Fri Mar 19 22:23:03 2021 user.notice : Try to start UA2F processor at [10210].
Fri Mar 19 22:23:03 2021 user.notice : UA2F processor start at [10210].
Fri Mar 19 22:23:03 2021 syslog.notice UA2F[10210]: Pipset inited.
Fri Mar 19 22:23:03 2021 syslog.notice UA2F[10210]: UA2F has inited successful.
Fri Mar 19 22:23:03 2021 syslog.err UA2F[10210]: Exit at breakpoint 10.
Fri Mar 19 22:23:03 2021 user.err : Child sucide.
Fri Mar 19 22:23:03 2021 user.notice : Try to start UA2F processor at [10211].
Fri Mar 19 22:23:03 2021 user.notice : UA2F processor start at [10211].
Fri Mar 19 22:23:03 2021 syslog.notice UA2F[10211]: Pipset inited.
Fri Mar 19 22:23:03 2021 syslog.notice UA2F[10211]: UA2F has inited successful.
Fri Mar 19 22:23:03 2021 syslog.err UA2F[10211]: Exit at breakpoint 10.
Fri Mar 19 22:23:03 2021 user.err : Child sucide.
Fri Mar 19 22:23:03 2021 user.err : Meet too many fatal error, no longer try to recover.
Fri Mar 19 22:23:03 2021 user.err : Exit at breakpoint 3.
```

Zxilly, 2021-03-19:
UA2F 已经在运行了，你又开了一个，第二个UA2F拿不到socket就炸了

Zxilly, 2021-03-20:
你应该是checkout了一个错误的版本，https://github.com/Zxilly/UA2F/commit/462cbc4deaff735b3485cdc0de3de8fd34d7493a 这个版本修了内存越界

closed 2021-03-21 by Zxilly