Zxilly/UA2F #10: 编译失败

kinglasy, 2021-04-18:
make[3]: Entering directory '/home/k/lede/package/UA2F'
rm -f /home/k/lede/build_dir/target-mipsel_24kc_musl/UA2F-2.2/.built
touch /home/k/lede/build_dir/target-mipsel_24kc_musl/UA2F-2.2/.built_check
mipsel-openwrt-linux-musl-gcc -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -ffile-prefix-map=/home/k/lede/build_dir/target-mipsel_24kc_musl/UA2F-2.2=UA2F-2.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -o /home/k/lede/build_dir/target-mipsel_24kc_musl/UA2F-2.2/ua2f.o -c /home/k/lede/build_dir/target-mipsel_24kc_musl/UA2F-2.2/ua2f.c
In file included from /home/k/lede/build_dir/target-mipsel_24kc_musl/UA2F-2.2/ua2f.c:16:
/home/k/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include/wait.h:1:2: warning: #warning redirecting incorrect #include <wait.h> to <sys/wait.h> [-Wcpp]
 #warning redirecting incorrect #include <wait.h> to <sys/wait.h>
  ^~~~~~~
/home/k/lede/build_dir/target-mipsel_24kc_musl/UA2F-2.2/ua2f.c: In function 'nfq_send_verdict':
/home/k/lede/build_dir/target-mipsel_24kc_musl/UA2F-2.2/ua2f.c:170:11: warning: implicit declaration of function 'nfq_nlmsg_put'; did you mean 'nfq_nlmsg_parse'? [-Wimplicit-function-declaration]
     nlh = nfq_nlmsg_put(buf, NFQNL_MSG_VERDICT, queue_num);
           ^~~~~~~~~~~~~
           nfq_nlmsg_parse
/home/k/lede/build_dir/target-mipsel_24kc_musl/UA2F-2.2/ua2f.c:170:9: warning: assignment to 'struct nlmsghdr *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
     nlh = nfq_nlmsg_put(buf, NFQNL_MSG_VERDICT, queue_num);
         ^
/home/k/lede/build_dir/target-mipsel_24kc_musl/UA2F-2.2/ua2f.c: In function 'main':
/home/k/lede/build_dir/target-mipsel_24kc_musl/UA2F-2.2/ua2f.c:501:9: warning: assignment to 'struct nlmsghdr *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
     nlh = nfq_nlmsg_put(buf, NFQNL_MSG_CONFIG, queue_number);
         ^
/home/k/lede/build_dir/target-mipsel_24kc_musl/UA2F-2.2/ua2f.c:510:9: warning: assignment to 'struct nlmsghdr *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
     nlh = nfq_nlmsg_put(buf, NFQNL_MSG_CONFIG, queue_number);
         ^
mipsel-openwrt-linux-musl-gcc -L/home/k/lede/staging_dir/target-mipsel_24kc_musl/usr/lib -L/home/k/lede/staging_dir/target-mipsel_24kc_musl/lib -L/home/k/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/lib -L/home/k/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib -lmnl -lnetfilter_queue -lipset -znow -zrelro -o /home/k/lede/build_dir/target-mipsel_24kc_musl/UA2F-2.2/ua2f /home/k/lede/build_dir/target-mipsel_24kc_musl/UA2F-2.2/ua2f.o
/home/k/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib/gcc/mipsel-openwrt-linux-musl/8.4.0/../../../../mipsel-openwrt-linux-musl/bin/ld: /home/k/lede/build_dir/target-mipsel_24kc_musl/UA2F-2.2/ua2f.o: in function `queue_cb':
ua2f.c:(.text+0x2a2): undefined reference to `nfq_nlmsg_put'
/home/k/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib/gcc/mipsel-openwrt-linux-musl/8.4.0/../../../../mipsel-openwrt-linux-musl/bin/ld: /home/k/lede/build_dir/target-mipsel_24kc_musl/UA2F-2.2/ua2f.o: in function `main':
ua2f.c:(.text.startup+0x148): undefined reference to `nfq_nlmsg_put'
/home/k/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib/gcc/mipsel-openwrt-linux-musl/8.4.0/../../../../mipsel-openwrt-linux-musl/bin/ld: ua2f.c:(.text.startup+0x17a): undefined reference to `nfq_nlmsg_put'
collect2: error: ld returned 1 exit status
Makefile:46: recipe for target '/home/k/lede/build_dir/target-mipsel_24kc_musl/UA2F-2.2/.built' failed
make[3]: *** [/home/k/lede/build_dir/target-mipsel_24kc_musl/UA2F-2.2/.built] Error 1
make[3]: Leaving directory '/home/k/lede/package/UA2F'
time: package/UA2F/compile#0.22#0.07#0.27
package/Makefile:111: recipe for target 'package/UA2F/compile' failed
make[2]: *** [package/UA2F/compile] Error 2
make[2]: Leaving directory '/home/k/lede'
package/Makefile:107: recipe for target '/home/k/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/k/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/k/lede'
/home/k/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

用的是lean的源码

arbaleast, 2021-04-19:
用openwrt官方源能编译成功

blackyau, 2021-04-19:
有个解决方法
用 https://minhaskamal.github.io/DownGit/#/home 下载 OpenWrt 官方用的新版 `libnetfilter-queue` 
下载地址是这个 https://github.com/openwrt/packages/tree/master/libs/libnetfilter-queue
然后覆盖到 `~/lede/package/libs/libnetfilter-queue` 目录里面去，就可以编译成功了。
唯一的问题是，这样操作后就不能直接 `git pull` 更新前需要自己解决一下冲突的问题。

Zxilly, 2021-04-19:
@blackyau 你把变更存成commit就能pull了

kinglasy, 2021-04-20:
还有个问题，如果我想暂时放行某一台电脑，要如何处理呢，作业系统不放行就登录不了

Zxilly, 2021-04-20:
@kinglasy 你看iptables规则嘛，你加一条src是你的PC的就return的规则就好

closed 2021-04-23 by Zxilly

kinglasy, 2021-05-08:
说实话，我真滴不会，救救我，有个交作业的网站必须用电脑的UA，

SunBK201, 2021-05-08:
你

SunBK201, 2021-05-08:
@kinglasy 你可以针对你所要访问的网站放行
```
iptables -t mangle -A ua2f -d 网站ip -j RETURN
```

Zxilly, 2021-05-08:
@SunBK201 `-A` 不行，包已经到`NFQUEUE`去了，应该用`-I`

SunBK201, 2021-05-08:
@Zxilly 好的，已修改