Zxilly/UA2F #9: 存在User-Agent没有修改

arbaleast, 2021-04-18:
GET /1113200827040098/super/security HTTP/1.1
User-agent: Easemob-SDK(Android) 3.7.1
Connection: Keep-Alive
Authorization: Bearer YWMtyTCqCJ_pEeuwtKsbzgi5wJ18klvgRkGgrWaRAzRwjhqgQ_WgJ5kR65Jvp6jL71eeAwMAAAF44rTHdQBPGgA3RlChzLdw7foQYuYZDjKFR1sXO7jOC1ezDPo_lYt55A
Host: a4.easemob.com
Accept-Encoding: gzip

--------
POST /mmtls/7ae571b3 HTTP/1.1
Accept: */*
Cache-Control: no-cache
Connection: close
Content-Length: 611
Content-Type: application/octet-stream
Host: dns.weixin.qq.com
Upgrade: mmtls
User-Agent: MicroMessenger Client

--------
在使用wireshark对路由wan口抓包，发现这两个UA在使用UA2F后没有修改掉

Zxilly, 2021-04-18:
谢谢！我检查一下

Zxilly, 2021-04-18:
[image]
未能复现，能提供更多信息吗？

Zxilly, 2021-04-18:
@arbaleast 如果可能的话，可否提供wireshark抓包数据，我看下二进制

arbaleast, 2021-04-18:
[捕获到的UA](https://victoriaca-my.sharepoint.com/%3Af%3A/g/personal/arbalest_ncov_office_gy/EogOpyuFc45PmqMIYe2xPPUBAuCQWEsV0vfbFumC1Z9nWg?e=9GBrzw)

arbaleast, 2021-04-18:
暂时捕获到User-agent: Easemob-SDK(Android) 3.7.1,另外一个还没抓到（上午抓的时候忘记保存了）

arbaleast, 2021-04-18:
另外UA2F最近显示的日志
     Sun Apr 18 10:12:42 2021 syslog.info UA2F[4105]: UA2F has handled 16384 http, 828 http 1.0, 3644 noua http, 10381752 tcp. Set 12656 mark and 10553 nohttp mark in 1 days, 5 hours, 10 minutes and 15 seconds

看起来运行正常

Zxilly, 2021-04-18:
@arbaleast 我看了一下，原因大概知道了
[image]
按照http标准，每个header字段应该以`\r\n`结束，UA2F也是基于这个判断的
很明显，发这个请求的应用没有按标准来。。。。
然后就是微信的那个，你可以看到，iptables规则里有
`iptables -t mangle -A ua2f -p tcp --dport 443 -j RETURN`
这是为了效率考虑，不会处理443端口的数据，但是你抓到的这个请求就是一个端口为443的http。。。

arbaleast, 2021-04-18:
好的，谢谢

arbaleast, 2021-04-18:
[image]
一款明叫超级课表的软件，和微信(╥﹏╥)

Zxilly, 2021-04-18:
emmm....这个没办法，不可能为了一个不按标准来的改结构

arbaleast, 2021-04-19:
> emmm....这个没办法，不可能为了一个不按标准来的改结构

没事，我只是吐槽一下这两个软件。

SunBK201, 2021-04-22:
[image]
这个包没被改掉，貌似不是因为http header没有\r\n, 因为这个包是有\r\n

Zxilly, 2021-04-22:
@SunBK201 因为标准是`User-Agent`。。。写这个的人真不如去引用个OKHttp。。。

SunBK201, 2021-04-22:
啊这，没想到是因为这个。。

arbaleast, 2021-04-23:
https://victoriaca-my.sharepoint.com/:u:/g/personal/arbalest_ncov_office_gy/EcT0k0YXE2VFjkznNuwPJp0BpRypCT5PaRPHkhBdsuETYw?e=aIV79p
https://victoriaca-my.sharepoint.com/:u:/g/personal/arbalest_ncov_office_gy/ETPSn_CqbCdLnnJ-HcWF7HEBaANerb58NcXlHqYXhUl32A?e=dF9pab
https://victoriaca-my.sharepoint.com/:u:/g/personal/arbalest_ncov_office_gy/EbY2b2757ixPnHOxSW8T7QwBDkj12oBBTq630amvgFdNIg?e=iljCbn

---
请问一下，这些也是因为User-Agent使用不规范吗？

Zxilly, 2021-04-23:
@arbaleast 这个应该是网卡分包了，UA2F目前处理不了这种情况

xmurp-ua 的 dev 版实现可以处理

如果要支持这种情况，可能整个项目都要推倒重来。。。从面向包改成面向流的处理

Zxilly, 2021-04-23:
我有一个实验式的实现的想法，但是有可能会污染正常数据。。。基本就是放弃对于HTTP header的解析，只考虑`User-Agent`这个串

Zxilly, 2021-05-01:
https://github.com/Zxilly/UA2F/releases/tag/20210502T003557 这个release包含了一个测试性的实现

Zxilly, 2021-05-01:
顺便以后会自动release mt7620的ipk

closed 2021-05-01 by Zxilly