Zxilly/UA2F #12: 不太会玩，但是看上去感觉工作不正常（ua.zhezhiyu.xyz显示的ua都没有更改）

Matt-wzy, 2021-05-13:
‘’‘
Wed May 12 23:41:45 2021 kern.warn kernel: [   37.901180] rkp-ipid: Successfully modified 2048 packages, in which 2048 IDs are in increasing order, 0 IDs are random. There are 0 packages not writable.
Wed May 12 23:42:51 2021 syslog.info UA2F[1298]: UA2F has handled 8 ua http, 605 tcp. Set 0 mark and 0 noUA mark in 1 minutes and 19 seconds
Wed May 12 23:43:31 2021 kern.warn kernel: [  144.202643] rkp-ipid: Successfully modified 4096 packages, in which 4096 IDs are in increasing order, 0 IDs are random. There are 0 packages not writable.
Wed May 12 23:46:30 2021 syslog.info UA2F[1298]: UA2F has handled 16 ua http, 1483 tcp. Set 1 mark and 0 noUA mark in 4 minutes and 58 seconds
Wed May 12 23:47:03 2021 daemon.err uhttpd[1322]: luci: accepted login on / for root from 192.168.1.124
Wed May 12 23:50:11 2021 syslog.info UA2F[1298]: UA2F has handled 32 ua http, 2140 tcp. Set 3 mark and 0 noUA mark in 8 minutes and 39 seconds
Thu May 13 00:03:18 2021 kern.warn kernel: [ 1328.820638] rkp-ipid: Successfully modified 8192 packages, in which 8192 IDs are in increasing order, 0 IDs are random. There are 0 packages not writable.
Thu May 13 00:53:07 2021 syslog.info UA2F[1298]: UA2F has handled 64 ua http, 5541 tcp. Set 4 mark and 0 noUA mark in 1 hours, 11 minutes and 35 seconds
Thu May 13 00:54:35 2021 kern.warn kernel: [ 4405.852755] rkp-ipid: Successfully modified 16384 packages, in which 16384 IDs are in increasing order, 0 IDs are random. There are 0 packages not writable.
Thu May 13 02:23:35 2021 syslog.info UA2F[1298]: UA2F has handled 128 ua http, 10940 tcp. Set 4 mark and 1 noUA mark in 2 hours, 42 minutes and 3 seconds
Thu May 13 03:00:42 2021 kern.warn kernel: [11973.588624] rkp-ipid: Successfully modified 32768 packages, in which 32768 IDs are in increasing order, 0 IDs are random. There are 0 packages not writable.
Thu May 13 06:18:43 2021 syslog.info UA2F[1298]: UA2F has handled 256 ua http, 18320 tcp. Set 4 mark and 1 noUA mark in 6 hours, 37 minutes and 11 seconds
Thu May 13 07:57:39 2021 syslog.info UA2F[1298]: UA2F has handled 512 ua http, 23112 tcp. Set 4 mark and 1 noUA mark in 8 hours, 16 minutes and 7 seconds
Thu May 13 09:01:47 2021 daemon.notice netifd: Network device 'eth1' link is down
Thu May 13 09:01:47 2021 kern.info kernel: [33639.755689] br-lan: port 1(eth1) entered disabled state
Thu May 13 09:01:47 2021 kern.info kernel: [33639.758744] r8152 4-1:1.0 eth1: carrier off
Thu May 13 09:01:48 2021 daemon.notice netifd: bridge 'br-lan' link is down
Thu May 13 09:01:48 2021 daemon.notice netifd: Interface 'lan' has link connectivity loss
Thu May 13 09:01:54 2021 daemon.notice netifd: Network device 'eth1' link is up
Thu May 13 09:01:54 2021 daemon.notice netifd: bridge 'br-lan' link is up
Thu May 13 09:01:54 2021 daemon.notice netifd: Interface 'lan' has link connectivity
Thu May 13 09:01:54 2021 kern.notice kernel: [33647.116453] r8152 4-1:1.0 eth1: Promiscuous mode enabled
Thu May 13 09:01:54 2021 kern.info kernel: [33647.117119] r8152 4-1:1.0 eth1: carrier on
Thu May 13 09:01:54 2021 kern.info kernel: [33647.117924] br-lan: port 1(eth1) entered blocking state
Thu May 13 09:01:54 2021 kern.info kernel: [33647.118386] br-lan: port 1(eth1) entered forwarding state
’‘’
看日志工作正常，但是浏览器看ua显示发现没有修改
[image]
手机端同样
[image]

（虽然截图上面clash是开启状态，但是我在昨晚已经开启和关闭clash（电脑和手机都）都测试了好几次，还是没有ua修改）
（但是昨晚测试的时候开着clash是能改成ffffff的……但是就不知道为什么今天突然不能）

路由器使用rockchip3328芯片的Orange pi R1Plus
固件采用Fork别人的GitHub自动编译然后自己改配置
[本次测试所使用的固件链接](https://github.com/Matt-wzy/Actions-immortalWrt-UA2F/actions/runs/835183346)
昨晚刚编译好的固件（（（（

Matt-wzy, 2021-05-13:
```

iptables -t nat -A PREROUTING -p udp --dport 53 -j REDIRECT --to-ports 53
iptables -t nat -A PREROUTING -p tcp --dport 53 -j REDIRECT --to-ports 53

# 通过 rkp-ipid 设置 IPID
iptables -t mangle -N IPID_MOD
iptables -t mangle -A FORWARD -j IPID_MOD
iptables -t mangle -A OUTPUT -j IPID_MOD
iptables -t mangle -A IPID_MOD -d 0.0.0.0/8 -j RETURN
iptables -t mangle -A IPID_MOD -d 127.0.0.0/8 -j RETURN
#由于本校局域网是A类网，所以我将这一条注释掉了，具体要不要注释结合你所在的校园网
# iptables -t mangle -A IPID_MOD -d 10.0.0.0/8 -j RETURN
iptables -t mangle -A IPID_MOD -d 172.16.0.0/12 -j RETURN
iptables -t mangle -A IPID_MOD -d 192.168.0.0/16 -j RETURN
iptables -t mangle -A IPID_MOD -d 255.0.0.0/8 -j RETURN
iptables -t mangle -A IPID_MOD -j MARK --set-xmark 0x10/0x10

# ua2f 改UA
iptables -t mangle -N ua2f
#iptables -t mangle -A ua2f -d 10.0.0.0/8 -j RETURN
iptables -t mangle -A ua2f -d 127.0.0.0/8 -j RETURN
iptables -t mangle -A ua2f -d 192.168.0.0/16 -j RETURN # 不处理流向保留地址的包
iptables -t mangle -A ua2f -p tcp --dport 443 -j RETURN
iptables -t mangle -A ua2f -p tcp --dport 22 -j RETURN # 不处理 SSH 和 https
iptables -t mangle -A ua2f -p tcp --dport 80 -j CONNMARK --set-mark 44
iptables -t mangle -A ua2f -m connmark --mark 43 -j RETURN # 不处理标记为非 http 的流 (实验性)
iptables -t mangle -A ua2f -m set --set nohttp dst,dst -j RETURN
iptables -t mangle -A ua2f -j NFQUEUE --queue-num 10010

iptables -t mangle -A FORWARD -p tcp -m conntrack --ctdir ORIGINAL -j ua2f
iptables -t mangle -A FORWARD -p tcp -m conntrack --ctdir REPLY

# 防时钟偏移检测
iptables -t nat -N ntp_force_local
iptables -t nat -I PREROUTING -p udp --dport 123 -j ntp_force_local
iptables -t nat -A ntp_force_local -d 0.0.0.0/8 -j RETURN
iptables -t nat -A ntp_force_local -d 127.0.0.0/8 -j RETURN
iptables -t nat -A ntp_force_local -d 192.168.0.0/16 -j RETURN
iptables -t nat -A ntp_force_local -s 192.168.0.0/16 -j DNAT --to-destination 192.168.1.1

# 通过 iptables 修改 TTL 值
iptables -t mangle -A POSTROUTING -j TTL --ttl-set 64

# iptables 拒绝 AC 进行 Flash 检测
iptables -I FORWARD -p tcp --sport 80 --tcp-flags ACK ACK -m string --algo bm --string " src=\"http://1.1.1." -j DROP

```

防火墙配置，不知道对问题分析有没有用

Zxilly, 2021-05-13:
你本地开了代理，指不定走代理了。。。

Matt-wzy, 2021-05-13:
> 你本地开了代理，指不定走代理了。。。

[image]

上面说了开了代理关上代理一个样了（先取消全局然后退出的）

Zxilly, 2021-05-13:
没法判断，只能靠你自己了。。。怕不是和ipid冲突了？

Matt-wzy, 2021-05-13:
> 没法判断，只能靠你自己了。。。怕不是和ipid冲突了？

电脑、手机都关了代理跑的时候抓的包，
在路由器上运行 ` tcpdump -i eth0 -w test_Noproxy.cap ` 得到的，其中10.184.184.***是我所使用的设备在校园网内的ip地址

[test_Noproxy.zip](https://github.com/Zxilly/UA2F/files/6470800/test_Noproxy.zip)

Matt-wzy, 2021-05-13:
> > 没法判断，只能靠你自己了。。。怕不是和ipid冲突了？
> 
> 电脑、手机都关了代理跑的时候抓的包，
> 
> 电脑、手机访问几次ua.zhezhiyu.xyz
> 
> 在路由器上运行 `tcpdump -i eth0 -w test_Noproxy.cap` 得到的，其中10.184.184.***是我所使用的设备在校园网内的ip地址
> 
> [test_Noproxy.zip](https://github.com/Zxilly/UA2F/files/6470800/test_Noproxy.zip)

``` 
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
Thu May 13 13:03:27 2021 kern.notice kernel: [    0.000000] Linux version 5.4.117 (runner@fv-az162-443) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r18571-c97d4ee646)) #0 SMP PREEMPT Tue May 11 20:19:25 2021
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] Machine model: Xunlong Orange Pi R1 Plus
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff130000 (options '')
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] printk: bootconsole [uart8250] enabled
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] cma: Reserved 8 MiB at 0x000000003f800000
Thu May 13 13:03:27 2021 kern.debug kernel: [    0.000000] On node 0 totalpages: 261632
Thu May 13 13:03:27 2021 kern.debug kernel: [    0.000000]   DMA32 zone: 4088 pages used for memmap
Thu May 13 13:03:27 2021 kern.debug kernel: [    0.000000]   DMA32 zone: 0 pages reserved
Thu May 13 13:03:27 2021 kern.debug kernel: [    0.000000]   DMA32 zone: 261632 pages, LIFO batch:63
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] psci: probing for conduit method from DT.
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] psci: PSCIv1.0 detected in firmware.
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] psci: Using standard PSCI v0.2 function IDs
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] psci: MIGRATE_INFO_TYPE not supported.
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] psci: SMC Calling Convention v1.0
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] percpu: Embedded 21 pages/cpu s45912 r8192 d31912 u86016
Thu May 13 13:03:27 2021 kern.debug kernel: [    0.000000] pcpu-alloc: s45912 r8192 d31912 u86016 alloc=21*4096
Thu May 13 13:03:27 2021 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] Detected VIPT I-cache on CPU0
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] CPU features: detected: ARM erratum 845719
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 257544
Thu May 13 13:03:27 2021 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xff130000 root=PARTUUID=5452574f-02 rw rootwait
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] Memory: 1005008K/1046528K available (7422K kernel code, 536K rwdata, 2308K rodata, 1664K init, 717K bss, 33328K reserved, 8192K cma-reserved)
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] rcu: Preemptible hierarchical RCU implementation.
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] rcu: 	RCU event tracing is enabled.
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] 	Tasks RCU enabled.
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] GIC: Using split EOI/Deactivate mode
Thu May 13 13:03:27 2021 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x29c/0x39c with crng_init=0
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
Thu May 13 13:03:27 2021 kern.info kernel: [    0.000009] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
Thu May 13 13:03:27 2021 kern.info kernel: [    0.001566] Console: colour dummy device 80x25
Thu May 13 13:03:27 2021 kern.info kernel: [    0.002027] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
Thu May 13 13:03:27 2021 kern.info kernel: [    0.002959] pid_max: default: 32768 minimum: 301
Thu May 13 13:03:27 2021 kern.info kernel: [    0.003651] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
Thu May 13 13:03:27 2021 kern.info kernel: [    0.004329] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
Thu May 13 13:03:27 2021 kern.info kernel: [    0.007209] ASID allocator initialised with 32768 entries
[... cut ...]
Thu May 13 14:58:20 2021 authpriv.notice dropbear[4842]: Pubkey auth succeeded for 'root' with key sha1!! de:0e:a6:01:a7:c8:d4:6d:c7:f3:ae:03:38:13:04:38:4c:95:42:7a from 192.168.1.124:56011
Thu May 13 14:59:13 2021 authpriv.info dropbear[4842]: Exit (root) from <192.168.1.124:56011>: Exited normally
Thu May 13 15:08:49 2021 kern.warn kernel: [ 7540.481142] rkp-ipid: Successfully modified 16384 packages, in which 16384 IDs are in increasing order, 0 IDs are random. There are 0 packages not writable.
``` 
对应此时路由器内日志（删除部分校园网脚本登录信息）

Matt-wzy, 2021-05-13:
中午抓的这个包，在电脑开启clash代理的时候莫名其妙成功了几次，不过大多数都没成功
[image]

[test.zip](https://github.com/Zxilly/UA2F/files/6470923/test.zip)

Zxilly, 2021-05-13:
看起来UA2F没有接收到这些包，检查一下你的拓扑吧。。。你这环境太复杂了我没看明白

Zxilly, 2021-05-13:
我去看了一下rkp-ipid的设计，emmm。。。不太兼容
那边已经给包的mark位加上标记了，会导致ua2f不处理这些包，ua2f是默认包上没有任何标记的

XingNian-www, 2021-05-13:
(怕不是开启了硬件加速或者Flow Offloading 转发加速的锅

closed 2021-05-14 by Zxilly

Matt-wzy, 2021-06-05:
由于本人实操后发现改ua+ipid+ttl+时间偏移等一套组合拳打完还是莫得用处，该禁网还是禁网，算了算了，直接住没网大学得了，也不纠结这些细节了，把有关的抓包文件和编辑历史都删掉了，所有编辑仅此而已，并无它改动，望后人看到此issue（当然没人看）时周知，也十分感谢仓库作者对于我问题的耐心分析和解答，虽然没有在实际上帮到我成功使用整个互联网，但是也对仓库作者做出十足的敬意，感谢您写出这样的代码给我们一代又一代人用，让我们在看到一丝希望的光时也能了解到相关的网络原理。
只可惜，我无幸成为那一批被圣光普照到的人，只希望，更多的人能通过高考中多考一两分钟，使自己以后生活在没有恶心的网络限制的地方。

bling-yshs, 2023-09-02:
同款问题，最后发现跟openclash还有本地的开着的cfw有关系，最后我是用的tun模式好像才没问题，但是我不是很确定

Zxilly, 2023-09-03:
@bling-yshs 开了clash以后流量本身流经代理时已经是加密流量，不需要修改UA了，是正常的。网关同样读取不到UA。