Zxilly/UA2F #14: UA2F中系统日志显示异常

tonytt0624, 2021-05-29:
Fri May 28 08:12:01 2021 syslog.info UA2F[1984]: UA2F has handled 385024 ua http, 103544050 tcp. Set 385024 mark and 7 noUA mark in 5 days, 20 hours, 3 minutes and 53 seconds
Fri May 28 08:13:33 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:33 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:33 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:33 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:34 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:36 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:36 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:36 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:37 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:37 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:37 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:37 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:37 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:38 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:38 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:40 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:41 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:41 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:41 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:41 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:42 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:42 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:42 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:43 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:43 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:44 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:44 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:44 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:45 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:45 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:46 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:47 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:49 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:49 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:53 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:53 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:57 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:14:01 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:14:09 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:14:14 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:14:17 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:14:41 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:14:47 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:14:49 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:58:13 2021 daemon.notice netifd: wan (7097): udhcpc: sending renew to 10.217.31.254
Fri May 28 08:58:13 2021 daemon.notice netifd: wan (7097): udhcpc: lease of 10.217.20.137 obtained, lease time 7200
Fri May 28 09:00:17 2021 daemon.notice netifd: wan2 (12127): udhcpc: sending renew to 10.217.31.254
Fri May 28 09:00:17 2021 daemon.notice netifd: wan2 (12127): udhcpc: lease of 10.217.20.138 obtained, lease time 7200
Fri May 28 09:58:14 2021 daemon.notice netifd: wan (7097): udhcpc: sending renew to 10.217.31.254
Fri May 28 09:58:14 2021 daemon.notice netifd: wan (7097): udhcpc: lease of 10.217.20.137 obtained, lease time 7200
Fri May 28 10:00:17 2021 daemon.notice netifd: wan2 (12127): udhcpc: sending renew to 10.217.31.254
Fri May 28 10:00:17 2021 daemon.notice netifd: wan2 (12127): udhcpc: lease of 10.217.20.138 obtained, lease time 7200
Fri May 28 10:58:14 2021 daemon.notice netifd: wan (7097): udhcpc: sending renew to 10.217.31.254
Fri May 28 10:58:14 2021 daemon.notice netifd: wan (7097): udhcpc: lease of 10.217.20.137 obtained, lease time 7200
Fri May 28 11:00:17 2021 daemon.notice netifd: wan2 (12127): udhcpc: sending renew to 10.217.31.254
Fri May 28 11:00:17 2021 daemon.notice netifd: wan2 (12127): udhcpc: lease of 10.217.20.138 obtained, lease time 7200
Fri May 28 11:58:14 2021 daemon.notice netifd: wan (7097): udhcpc: sending renew to 10.217.31.254
Fri May 28 11:58:14 2021 daemon.notice netifd: wan (7097): udhcpc: lease of 10.217.20.137 obtained, lease time 7200
Fri May 28 12:00:18 2021 daemon.notice netifd: wan2 (12127): udhcpc: sending renew to 10.217.31.254
Fri May 28 12:00:18 2021 daemon.notice netifd: wan2 (12127): udhcpc: lease of 10.217.20.138 obtained, lease time 7200
Fri May 28 12:58:14 2021 daemon.notice netifd: wan (7097): udhcpc: sending renew to 10.217.31.254
Fri May 28 12:58:14 2021 daemon.notice netifd: wan (7097): udhcpc: lease of 10.217.20.137 obtained, lease time 7200
Fri May 28 13:00:18 2021 daemon.notice netifd: wan2 (12127): udhcpc: sending renew to 10.217.31.254
Fri May 28 13:00:18 2021 daemon.notice netifd: wan2 (12127): udhcpc: lease of 10.217.20.138 obtained, lease time 7200
Fri May 28 13:58:14 2021 daemon.notice netifd: wan (7097): udhcpc: sending renew to 10.217.31.254
Fri May 28 13:58:14 2021 daemon.notice netifd: wan (7097): udhcpc: lease of 10.217.20.137 obtained, lease time 7200
Fri May 28 14:00:18 2021 daemon.notice netifd: wan2 (12127): udhcpc: sending renew to 10.217.31.254
Fri May 28 14:00:18 2021 daemon.notice netifd: wan2 (12127): udhcpc: lease of 10.217.20.138 obtained, lease time 7200
Fri May 28 14:28:35 2021 syslog.info UA2F[1984]: UA2F has handled 393216 ua http, 103860581 tcp. Set 393216 mark and 7 noUA mark in 6 days, 2 hours, 20 minutes and 27 seconds
[... 201 lines cut ...]
Fri May 28 20:27:56 2021 syslog.warn UA2F[15324]: UA overflow
Fri May 28 20:27:56 2021 syslog.warn UA2F[15324]: UA overflow
Fri May 28 20:27:57 2021 syslog.warn UA2F[15324]: UA overflow
Fri May 28 20:58:15 2021 daemon.notice netifd: wan (7097): udhcpc: sending renew to 10.217.31.254
Fri May 28 20:58:15 2021 daemon.notice netifd: wan (7097): udhcpc: lease of 10.217.20.137 obtained, lease time 7200
Fri May 28 21:00:18 2021 daemon.notice netifd: wan2 (12127): udhcpc: sending renew to 10.217.31.254
Fri May 28 21:00:18 2021 daemon.notice netifd: wan2 (12127): udhcpc: lease of 10.217.20.138 obtained, lease time 7200
Fri May 28 21:01:47 2021 syslog.info UA2F[15324]: UA2F has handled 16384 ua http, 2570311 tcp. Set 16384 mark and 0 noUA mark in 4 hours, 28 minutes and 6 seconds
Fri May 28 21:58:15 2021 daemon.notice netifd: wan (7097): udhcpc: sending renew to 10.217.31.254
Fri May 28 21:58:15 2021 daemon.notice netifd: wan (7097): udhcpc: lease of 10.217.20.137 obtained, lease time 7200
Fri May 28 22:00:18 2021 daemon.notice netifd: wan2 (12127): udhcpc: sending renew to 10.217.31.254
Fri May 28 22:00:18 2021 daemon.notice netifd: wan2 (12127): udhcpc: lease of 10.217.20.138 obtained, lease time 7200
Fri May 28 22:58:15 2021 daemon.notice netifd: wan (7097): udhcpc: sending renew to 10.217.31.254
Fri May 28 22:58:16 2021 daemon.notice netifd: wan (7097): udhcpc: lease of 10.217.20.137 obtained, lease time 7200
Fri May 28 23:00:18 2021 daemon.notice netifd: wan2 (12127): udhcpc: sending renew to 10.217.31.254
Fri May 28 23:00:18 2021 daemon.notice netifd: wan2 (12127): udhcpc: lease of 10.217.20.138 obtained, lease time 7200
Fri May 28 23:58:15 2021 daemon.notice netifd: wan (7097): udhcpc: sending renew to 10.217.31.254
Fri May 28 23:58:15 2021 daemon.notice netifd: wan (7097): udhcpc: lease of 10.217.20.137 obtained, lease time 7200
Sat May 29 00:00:18 2021 daemon.notice netifd: wan2 (12127): udhcpc: sending renew to 10.217.31.254
Sat May 29 00:00:18 2021 daemon.notice netifd: wan2 (12127): udhcpc: lease of 10.217.20.138 obtained, lease time 7200
Sat May 29 00:58:15 2021 daemon.notice netifd: wan (7097): udhcpc: sending renew to 10.217.31.254
Sat May 29 00:58:15 2021 daemon.notice netifd: wan (7097): udhcpc: lease of 10.217.20.137 obtained, lease time 7200
Sat May 29 01:00:18 2021 daemon.notice netifd: wan2 (12127): udhcpc: sending renew to 10.217.31.254
Sat May 29 01:00:18 2021 daemon.notice netifd: wan2 (12127): udhcpc: lease of 10.217.20.138 obtained, lease time 7200
Sat May 29 01:57:36 2021 syslog.info UA2F[15324]: UA2F has handled 24576 ua http, 3436524 tcp. Set 24576 mark and 0 noUA mark in 9 hours, 23 minutes and 55 seconds

Zxilly, 2021-05-29:
UA Overflow是报错，说明遇到了预料之外的情况
设计中，这个提示应该永远不会出现
如果可以的话，能发一下你的路由器抓包吗？用tcpdump，我才有可能去排查

tonytt0624, 2021-05-29:
请稍等一会，安装tcpdump遇到了一些未知问题，正在解决。。。

tonytt0624, 2021-05-29:
[tonytt.zip](https://github.com/Zxilly/UA2F/files/6564443/tonytt.zip)

closed 2021-05-29 by tonytt0624