Zxilly/UA2F #15: UA2F系统日志显示异常

tonytt0624, 2021-05-29:
[tonytt.zip](https://github.com/Zxilly/UA2F/files/6564446/tonytt.zip)

tonytt0624, 2021-05-29:
不好意思，手误关掉了

tonytt0624, 2021-05-29:
Fri May 28 08:12:01 2021 syslog.info UA2F[1984]: UA2F has handled 385024 ua http, 103544050 tcp. Set 385024 mark and 7 noUA mark in 5 days, 20 hours, 3 minutes and 53 seconds
Fri May 28 08:13:33 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:33 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:33 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:33 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:34 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:36 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:36 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:36 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:37 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:37 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:37 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:37 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:37 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:38 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:38 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:40 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:41 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:41 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:41 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:41 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:42 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:42 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:42 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:43 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:43 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:44 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:44 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:44 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:45 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:45 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:46 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:47 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:49 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:49 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:53 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:53 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:13:57 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:14:01 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:14:09 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:14:14 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:14:17 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:14:41 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:14:47 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:14:49 2021 syslog.warn UA2F[1984]: UA overflow
Fri May 28 08:58:13 2021 daemon.notice netifd: wan (7097): udhcpc: sending renew to 10.217.31.254
Fri May 28 08:58:13 2021 daemon.notice netifd: wan (7097): udhcpc: lease of 10.217.20.137 obtained, lease time 7200
Fri May 28 09:00:17 2021 daemon.notice netifd: wan2 (12127): udhcpc: sending renew to 10.217.31.254
Fri May 28 09:00:17 2021 daemon.notice netifd: wan2 (12127): udhcpc: lease of 10.217.20.138 obtained, lease time 7200
Fri May 28 09:58:14 2021 daemon.notice netifd: wan (7097): udhcpc: sending renew to 10.217.31.254
Fri May 28 09:58:14 2021 daemon.notice netifd: wan (7097): udhcpc: lease of 10.217.20.137 obtained, lease time 7200
Fri May 28 10:00:17 2021 daemon.notice netifd: wan2 (12127): udhcpc: sending renew to 10.217.31.254
Fri May 28 10:00:17 2021 daemon.notice netifd: wan2 (12127): udhcpc: lease of 10.217.20.138 obtained, lease time 7200
Fri May 28 10:58:14 2021 daemon.notice netifd: wan (7097): udhcpc: sending renew to 10.217.31.254
Fri May 28 10:58:14 2021 daemon.notice netifd: wan (7097): udhcpc: lease of 10.217.20.137 obtained, lease time 7200
Fri May 28 11:00:17 2021 daemon.notice netifd: wan2 (12127): udhcpc: sending renew to 10.217.31.254
Fri May 28 11:00:17 2021 daemon.notice netifd: wan2 (12127): udhcpc: lease of 10.217.20.138 obtained, lease time 7200
Fri May 28 11:58:14 2021 daemon.notice netifd: wan (7097): udhcpc: sending renew to 10.217.31.254
Fri May 28 11:58:14 2021 daemon.notice netifd: wan (7097): udhcpc: lease of 10.217.20.137 obtained, lease time 7200
Fri May 28 12:00:18 2021 daemon.notice netifd: wan2 (12127): udhcpc: sending renew to 10.217.31.254
Fri May 28 12:00:18 2021 daemon.notice netifd: wan2 (12127): udhcpc: lease of 10.217.20.138 obtained, lease time 7200
Fri May 28 12:58:14 2021 daemon.notice netifd: wan (7097): udhcpc: sending renew to 10.217.31.254
Fri May 28 12:58:14 2021 daemon.notice netifd: wan (7097): udhcpc: lease of 10.217.20.137 obtained, lease time 7200
Fri May 28 13:00:18 2021 daemon.notice netifd: wan2 (12127): udhcpc: sending renew to 10.217.31.254
Fri May 28 13:00:18 2021 daemon.notice netifd: wan2 (12127): udhcpc: lease of 10.217.20.138 obtained, lease time 7200
Fri May 28 13:58:14 2021 daemon.notice netifd: wan (7097): udhcpc: sending renew to 10.217.31.254
Fri May 28 13:58:14 2021 daemon.notice netifd: wan (7097): udhcpc: lease of 10.217.20.137 obtained, lease time 7200
Fri May 28 14:00:18 2021 daemon.notice netifd: wan2 (12127): udhcpc: sending renew to 10.217.31.254
Fri May 28 14:00:18 2021 daemon.notice netifd: wan2 (12127): udhcpc: lease of 10.217.20.138 obtained, lease time 7200
[... cut ...]
Sat May 29 00:58:15 2021 daemon.notice netifd: wan (7097): udhcpc: sending renew to 10.217.31.254
Sat May 29 00:58:15 2021 daemon.notice netifd: wan (7097): udhcpc: lease of 10.217.20.137 obtained, lease time 7200
Sat May 29 01:00:18 2021 daemon.notice netifd: wan2 (12127): udhcpc: sending renew to 10.217.31.254
Sat May 29 01:00:18 2021 daemon.notice netifd: wan2 (12127): udhcpc: lease of 10.217.20.138 obtained, lease time 7200
Sat May 29 01:57:36 2021 syslog.info UA2F[15324]: UA2F has handled 24576 ua http, 3436524 tcp. Set 24576 mark and 0 noUA mark in 9 hours, 23 minutes and 55 seconds

tonytt0624, 2021-05-29:
已上传.zip文件（tcpdump抓包）

Zxilly, 2021-05-31:
@tonytt0624 你随便抓个包给我没用的。。。
只有你出错的时候抓的包才有用
挺麻烦的，但是抓不到我也不知道出啥情况了

tonytt0624, 2021-05-31:
确实 我看看有没有有啥办法能抓到这包，这段时间我研究一下

Zxilly, 2021-09-12:
看一下最新版本还会吗

tonytt0624, 2021-09-12:
今天刚更新的版本是吗，感谢，我会尽快测试。

lionop, 2021-09-18:
@Zxilly 仰望大佬，这个ua2f需要特别设置才能开启吗？还是安装后就自动随系统运行？另外上哪里能找到集成该固件的K2P 的固件，小白，感谢！

closed 2022-04-21 by Zxilly