Zxilly/UA2F #17: 编译失败 源码为最新的lean大源码

bfsg, 2021-06-02:
make[3]: *** [/home/bfsg/lede/build_dir/target-mipsel_24kc_musl/UA2F-3.7/.built] Error 1
make[3]: Leaving directory '/home/bfsg/lede/package/UA2F'
time: package/UA2F/compile#0.17#0.04#0.20
package/Makefile:111: recipe for target 'package/UA2F/compile' failed
make[2]: *** [package/UA2F/compile] Error 2
make[2]: Leaving directory '/home/bfsg/lede'
package/Makefile:107: recipe for target '/home/bfsg/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/bfsg/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/bfsg/lede'
/home/bfsg/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

Zxilly, 2021-06-02:
你刚好把出的错截掉了，判断不了
lean的源码libnetfilter-queue版本太低了，用openwrt

bfsg, 2021-06-02:
好的，谢谢

closed 2021-06-03 by Zxilly