Zxilly/UA2F #20: 不想用FFF...修改ua，怎么改源码。

DH-FU, 2021-06-05:
抱歉，我大部分代码看不懂，能帮帮我怎么把FFFF
换成其它的，我觉得学校看到我ua为FFFF...觉得不正常，可能会把我号给封了。我想换成正常的ua，这样感觉更安全，我上学期校园网掉了可以秒登，所以用脚本写了一个重登代码，这个学期就不行了。
上个学期一天重登40多次，应该是被学校看出异常。
我怕学校这次又看出异常，把不正常ua也加入扫描范围
这样就不行了。

Matt-wzy, 2021-06-05:
> 抱歉，我大部分代码看不懂，能帮帮我怎么把FFFF
> 换成其它的，我觉得学校看到我ua为FFFF...觉得不正常，可能会把我号给封了。我想换成正常的ua，这样感觉更安全，我上学期校园网掉了可以秒登，所以用脚本写了一个重登代码，这个学期就不行了。
> 上个学期一天重登40多次，应该是被学校看出异常。
> 我怕学校这次又看出异常，把不正常ua也加入扫描范围
> 这样就不行了。

亲亲如果您是抓包抓到请求然后用请求反复登录而不是模拟鼠标点击这种方式登录的话建议您今年重新抓包，也许会解决一天重登40多次的问题。

至于其他问题请留给作者。

别问我为什么这样给你建议，鬼知道为什么一个登录请求还带变得。

Zxilly, 2021-06-05:
你说的没法实现，UA2F设计之初就是面向包的处理，所以只能替换UA为等长的字符串。
你说的`学校看到我ua为FFFF...觉得不正常`基本不可能，校园网流量这么大只可能是设备审计，人工审计不可能的

Zxilly, 2021-06-05:
如果你想换个字母的话，改这里
https://github.com/Zxilly/UA2F/blob/661334b48ed2b3a9ad992a7af894e8b7cba011a8/src/ua2f.c#L445

closed 2021-06-07 by Zxilly