Zxilly/UA2F #22: 如果你想换个字母的话，改这里

DH-FU, 2021-06-05:
如果你想换个字母的话，改这里
https://github.com/Zxilly/UA2F/blob/661334b48ed2b3a9ad992a7af894e8b7cba011a8/src/ua2f.c#L445

_Originally posted by @Zxilly in https://github.com/Zxilly/UA2F/issues/20#issuecomment-855260521_

这样改行不行， 

static char str1[74]={"Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko)"};
.
.
.
int i;
.
.
.
 str = malloc(sizeof_buf);
  
     for(i=0; i<sizeof_buf; i++)
	{
		memset(str+i, str1[i] ,1);
	}
    if (sizeof_buf > 74)
        {
          for(i=75; i<sizeof_buf; i++)
	{
		memset(str+i, ' ' ,1);
	}
        }

Zxilly, 2021-06-05:
请把代码格式化好，我看不明白

Zxilly, 2021-06-05:
还有，没必要每次回复都开个新issue，请在同一个issue中完成

closed 2021-06-05 by Zxilly

DH-FU, 2021-06-05:
static char str1[74]={"Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko)"}; #添加一个电脑UA；

int i;#在int main（）中添加一个计数变量i；

str = malloc(sizeof_buf);
 for(i=0; i<sizeof_buf; i++){
	memset(str+i, str1[i] ,1);  #sizeof_buf长度的赋值为电脑ua
}

if (sizeof_buf > 74)  {
      for(i=75; i<sizeof_buf; i++)
	memset(str+i, ' ' ,1); #超过电脑UA的全部赋值为空格
    }

DH-FU, 2021-06-05:
总之就是把一个电脑ua一个字符一个字符的进行赋值，不够的补空格。

DH-FU, 2021-06-05:
> 还有，没必要每次回复都开个新issue，请在同一个issue中完成

static char str1[74]={"Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko)"}; #添加一个电脑UA；

int i;#在int main（）中添加一个计数变量i；

str = malloc(sizeof_buf);
for(i=0; i<sizeof_buf; i++){
memset(str+i, str1[i] ,1); #sizeof_buf长度的赋值为电脑ua
}

if (sizeof_buf > 74) {
for(i=75; i<sizeof_buf; i++)
memset(str+i, ' ' ,1); #超过电脑UA的全部赋值为空格
}

DH-FU, 2021-06-05:
> 还有，没必要每次回复都开个新issue，请在同一个issue中完成

总之就是把一个电脑ua一个字符一个字符的进行赋值，不够的补空格。

Zxilly, 2021-06-05:
你直接memcpy字符串不好吗，我觉得你的代码能力不太够，最好不要乱动