Zxilly/UA2F #26: 编译异常

Subdue0, 2021-07-07:
https://paste.ubuntu.com/p/3nDwZ828yr/

Zxilly, 2021-07-07:
`undefined reference to `nfq_nlmsg_put'`
libnetfilter-queue 版本低了，至少要1.0.5

Subdue0, 2021-07-07:
有没有的办法能直接升级软件包的

Zxilly, 2021-07-07:
啥叫直接升级软件包，你编译之前从官方仓库复制这个库到你自己的packages里面覆盖不就好了

closed 2021-07-11 by Zxilly

Subdue0, 2021-07-13:
为什么系统日志里没有UA2F相关日志，我已经把它编译进系统了。
打开UA检测网站，UA没变化

开机自启
ipset create nohttp hash:ip,port hashsize 16384 timeout 300

防火墙规则
iptables -t mangle -N ua2f
iptables -t mangle -A ua2f -d 10.0.0.0/8 -j RETURN
iptables -t mangle -A ua2f -d 127.0.0.0/8 -j RETURN
iptables -t mangle -A ua2f -d 192.168.0.0/16 -j RETURN # 不处理流向保留地址的包
iptables -t mangle -A ua2f -p tcp --dport 443 -j RETURN
iptables -t mangle -A ua2f -p tcp --dport 22 -j RETURN # 不处理 SSH 和 https
iptables -t mangle -A ua2f -p tcp --dport 80 -j CONNMARK --set-mark 44
iptables -t mangle -A ua2f -m connmark --mark 43 -j RETURN # 不处理标记为非 http 的流 (实验性)
iptables -t mangle -A ua2f -m set --set nohttp dst,dst -j RETURN
iptables -t mangle -A ua2f -j NFQUEUE --queue-num 10010

iptables -t mangle -A FORWARD -p tcp -m conntrack --ctdir ORIGINAL -j ua2f
iptables -t mangle -A FORWARD -p tcp -m conntrack --ctdir REPLY

Zxilly, 2021-07-13:
可能的原因太多了，没法帮你
比如iptables看下ua2f规则上有没有流量？