Zxilly/UA2F #27: 没反应

Subdue0, 2021-07-14:
我用的是主分支的最新版3.7。
我已经把它编译进固件中了，编译期间没有报错，但启动后系统日志里并没有UA2F相关日志。
像是程序没启动的样子，不知道是哪里出了问题。
打开UA检测网站，UA没有变化。

开机自启
ipset create nohttp hash:ip,port hashsize 16384 timeout 300

防火墙规则
iptables -t mangle -N ua2f
iptables -t mangle -A ua2f -d 10.0.0.0/8 -j RETURN
iptables -t mangle -A ua2f -d 127.0.0.0/8 -j RETURN
iptables -t mangle -A ua2f -d 192.168.0.0/16 -j RETURN # 不处理流向保留地址的包
iptables -t mangle -A ua2f -p tcp --dport 443 -j RETURN
iptables -t mangle -A ua2f -p tcp --dport 22 -j RETURN # 不处理 SSH 和 https
iptables -t mangle -A ua2f -p tcp --dport 80 -j CONNMARK --set-mark 44
iptables -t mangle -A ua2f -m connmark --mark 43 -j RETURN # 不处理标记为非 http 的流 (实验性)
iptables -t mangle -A ua2f -m set --set nohttp dst,dst -j RETURN
iptables -t mangle -A ua2f -j NFQUEUE --queue-num 10010

iptables -t mangle -A FORWARD -p tcp -m conntrack --ctdir ORIGINAL -j ua2f
iptables -t mangle -A FORWARD -p tcp -m conntrack --ctdir REPLY

Zxilly, 2021-07-14:
有个哥们给我提交了一些配置文件，我还没自己测试过，你研究一下他的两个PR?

Subdue0, 2021-07-14:
我先看看旧的反应如何吧

Subdue0, 2021-07-14:
UA2F版本是3.17.14
在Openwrt终端里，service ua2f start和service ua2f enable不能启动程序，
直接输入ua2f倒是可以启动程序，启动后程序挂起，有两个ua2f进程运行
这是修改后的UA："user-agent": "Mozilla/4.0 (compatible; MSIE 5.00; Windows 98)FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"

Subdue0, 2021-07-14:
ipset create nohttp hash:ip,port hashsize 16384 timeout 300 && ua2f，这个加入开机自启，程序可以运行，不过这个ua2f一运行就会挂起，阻断其他程序运行，典型的就是连reboot都不能运行了

Zxilly, 2021-07-14:
有两个UA2F进程运行是正常的，应该是那个哥们给我加的配置文件破坏了daemon，我回头修一下。

Subdue0, 2021-07-22:
啥时候有空修下bug

Zxilly, 2021-07-22:
反正最近是肯定没有的~ 测试的设备也没有

SunBK201, 2021-08-07:
@Subdue0
```bash
uci set ua2f.enabled.enabled=1
uci set ua2f.firewall.handle_fw=1
uci set ua2f.firewall.handle_tls=1
uci commit ua2f
service ua2f start
```

Class A user: 
```
iptables -t mangle -D ua2f 1
```

Class B user: 
```
iptables -t mangle -D ua2f 5
```

Class C user: 
```
iptables -t mangle -D ua2f 6
```

Subdue0, 2021-08-16:
你们有遇到不定时重启的问题吗？

Zxilly, 2021-08-16:
@Subdue0 确实会，但是至今没人给我提交引发崩溃的抓包文件，现在是靠自动重启硬抗的

Subdue0, 2021-08-19:
这包难抓，崩溃直接重启了

Zxilly, 2021-08-19:
等我有空弄个内置的错误处理吧

Zxilly, 2021-08-19:
如果你说的是路由器重启的话不是我的锅，ua2f不是内核程序，挂了就挂了不会连累系统的

lionop, 2021-09-18:
请教大佬，我安装了 https://github.com/DFlinmo/Actions-immortalWrt-UA2F/releases/tag/K2P-2021.08.30-1243 固件，应该是集成了ua2f了，刷机后ua2f不启动，系统日志和内核日志也没有信息，不知什么问题。启动ua2f需要特别设置吗？

SunBK201, 2021-09-18:
@lionop 这个固件采用的不是最新版本的ua2f，而是5月份版本的，里面预设了防火墙规则，你可以查看下日志ua2f是否启动了.

lionop, 2021-09-18:
@SunBK201  日志里没有ua2f信息，应该是没启动。需要命令吧应该：service ua2f start

DFlinmo, 2021-09-18:
> @lionop 这个固件采用的不是最新版本的ua2f（最新版本存在bug），而是5月份版本的，里面预设了防火墙规则，你可以查看下日志ua2f是否启动了.

防火墙预设命令我删了，本来想着新版本会自动配置，就删了，后面没改回来

SunBK201, 2021-09-18:
> > @lionop 这个固件采用的不是最新版本的ua2f（最新版本存在bug），而是5月份版本的，里面预设了防火墙规则，你可以查看下日志ua2f是否启动了.
> 
> 
> 
> 防火墙预设命令我删了，本来想着新版本会自动配置，就删了，后面没改回来

好吧,那最好还是加上吧

1715173329, 2021-09-25:
https://github.com/Zxilly/UA2F/blob/327f3129256c29cafdd2b7216e6dac0bdfa39a3f/files/ua2f.init#L29-L30
听说因为日志原因导致爆内存崩溃，可以试试把上面的 `1` 改为 `0`，直接关闭日志。
（procd 暂不支持把日志输出到自定义文件

SunBK201, 2021-09-25:
@1715173329 [image]

1715173329, 2021-09-25:
looks weird... 我国庆节看看是什么情况

SunBK201, 2021-09-25:
> looks weird... 我国庆节看看是什么情况

似乎是init方面出了些问题，导致进程重复了，daemon方面...

1715173329, 2021-09-25:
> 似乎是init方面出了些问题，导致进程重复了，daemon方面...

emmmm... 
在这下面插一个 `stop` 试试？
https://github.com/Zxilly/UA2F/blob/327f3129256c29cafdd2b7216e6dac0bdfa39a3f/files/ua2f.init#L14

1715173329, 2021-10-31:
非常迷惑的问题，在 procd 下面，ua2f 会被重复运行，并且 procd 并不认那个重复的进程。

1715173329, 2021-10-31:
> 似乎是init方面出了些问题，导致进程重复了，daemon方面...

看来这并不是 init 的问题，ua2f 本体在启动时会带动一个子进程，procd 在停止时会 kill 掉主进程，但子进程依然在继续运行。
如果因为改动配置而重启 ua2f daemon，就会导致错误。

SunBK201, 2021-10-31:
> > 似乎是init方面出了些问题，导致进程重复了，daemon方面...
> 
> 
> 
> 看来这并不是 init 的问题，ua2f 本体在启动时会带动一个子进程，procd 在停止时会 kill 掉主进程，但子进程依然在继续运行。
> 
> 如果因为改动配置而重启 ua2f daemon，就会导致错误。

是这样，主进程挂掉，工作子进程还在

Zxilly, 2021-10-31:
procd是发送哪个信号关闭进程的呢？我可以加个处理信号的函数来结束子进程。

1715173329, 2021-10-31:
> procd是发送哪个信号关闭进程的呢？我可以加个处理信号的函数来结束子进程。

应该是 SIGTERM。

Subdue0, 2021-11-01:
> @SunBK201 日志里没有ua2f信息，应该是没启动。需要命令吧应该：service ua2f start

这招是启动不了服务的，得直接输入ua2f启动

Zxilly, 2021-11-01:
我加了个回收的函数，不确定能不能工作，手头没设备测试，麻烦看一下

1715173329, 2021-11-02:
> 我加了个回收的函数，不确定能不能工作，手头没设备测试，麻烦看一下

最新的提交似乎导致 ua2f 无法回应 SIGTERM 信号。

Zxilly, 2021-11-02:
@1715173329 我的，我结束了子进程但是主进程忘记退出了。。。

1715173329, 2021-11-02:
> fix in [061d14a](https://github.com/Zxilly/UA2F/commit/061d14a04efe3d9317bcd07ed962d1186384143c)

非常感谢，现在测试 procd daemon 工作正常。

Zxilly, 2021-11-02:
暂时关闭了，观察一下最新版的release

closed 2021-11-02 by Zxilly