Zxilly/UA2F #31: 非计科专业 实在难以下手 请问有刷入即食的K2P固件吗 感谢各位作者辛勤付出！！

fanfan142, 2021-09-19:


SunBK201, 2021-09-19:
https://github.com/SunBK201/Actions-immortalWrt-UA2F

tonytt0624, 2021-09-20:
想问Zxilly有办法将ua2f加进到7620的Pandorabox固件中吗，由于pandorabox无源码，被迫只能SDK编译，但是可以说难度十分巨大，不知Zxiilly大佬是否可以帮帮忙，谢谢。http://downloads.pangubox.com:6380/pandorabox/18.12/targets/ralink/mt7620/

lionop, 2021-09-21:
非计科专业 实在难以下手 请问有刷入即食的K2P固件+1，感谢各位大佬。

affggh, 2022-03-11:
我编译了K2P的固件，从Lean大神仓库拉的源码，Ubuntu 18.04 LTS本地编译，immortalwrt ，源码来自https://github.com/immortalwrt/immortalwrt 根据作者的教程添加了UA2F编译而成
lanzou :     https://cutelei.lanzout.com/b0elkpwzi    密码:cqq

[immortalwrt_phicomm_k2p_ua2f.zip](https://github.com/Zxilly/UA2F/files/8232951/immortalwrt_phicomm_k2p_ua2f.zip)
[LEDE_phicomm_k2p_ua2f.zip](https://github.com/Zxilly/UA2F/files/8232952/LEDE_phicomm_k2p_ua2f.zip)

MADAXIAO, 2022-03-24:
问一下这个怎么用，直接在breed里刷压缩包里那openwrt个固件吗？直接刷就行了吗

closed 2022-06-01 by github-actions[bot]