Zxilly/UA2F #32: uci启动后，第一次访问网页可以成功修改ua 后面ua2f不再工作

Chordp, 2021-10-11:
uci启动后，第一次访问网页可以成功修改ua
第二次访问就变回去了

Zxilly, 2021-10-11:
请提供日志和抓包文件

Chordp, 2021-10-12:
> mnl_cb_run: Operation not permitted 最新版的bug，貌似进程重复。 你的日志应该会显示ua2f频繁重启，不过第二次访问就失效有些奇怪。 建议回退使用旧版本试试。

是的，有两个ua2f在运行

SunBK201, 2021-10-12:
> > mnl_cb_run: Operation not permitted 最新版的bug，貌似进程重复。 你的日志应该会显示ua2f频繁重启，不过第二次访问就失效有些奇怪。 建议回退使用旧版本试试。
> 
> 
> 
> 旧版是指哪个版本

6月份之前的

Chordp, 2021-10-12:
好像不是第一次有用，而是随机有效而且概率很小

Chordp, 2021-10-12:
Tue Oct 12 19:20:20 2021 user.notice : Try to start UA2F processor at [25842].
Tue Oct 12 19:20:20 2021 user.notice : UA2F processor start at [25842].
Tue Oct 12 19:20:20 2021 syslog.notice UA2F[25842]: Pipset inited.
Tue Oct 12 19:20:20 2021 syslog.notice UA2F[25842]: UA2F has inited successful.
Tue Oct 12 19:20:20 2021 syslog.err UA2F[25842]: Exit at breakpoint 10.

Brukamen, 2023-05-01:
# Copyright (C) 2020 Zxilly <zhouxinyu1001@gmail.com>
# Copyright (C) 2021 Tianling Shen <cnsztl@immortalwrt.org>
我的是这个版本，我在编译的时候添加了初始化脚本，让他开机自启，但是也是像你这个一样

Brukamen, 2023-05-01:
Mon May  1 16:33:23 2023 user.notice : UA2F processor start at [4197].
Mon May  1 16:33:23 2023 syslog.notice UA2F[4197]: Pipset inited.
Mon May  1 16:33:23 2023 syslog.notice UA2F[4197]: UA2F has inited successful.
日志倒是没有异常

Zxilly, 2023-05-01:
试试新版本，旧版本已经维护不了了

Brukamen, 2023-05-01:
> 

新版本怎么拉取，我不太会用github，望见谅

lucikap, 2023-07-24:
我制作了一个插件对ua2f的luci配置界面的支持，能提供给你吗？

Zxilly, 2023-07-24:
你可以交一个PR，或者你自己维护一个仓库，然后我提供一个指向的链接