Zxilly/UA2F #38: 可以支持自定义UA吗

SenkuuChin, 2021-11-29:
因为我们学校偶尔信息中心的老师就想抓一下人，而GS每次都没被，我发现GSWiFi的UA是个正常值的，所以我想定义为一个正常值，不然他那边一看，全都是正常值，我这个明显不正常了，就很容易被逮住。
如果支持自定义，要如何自定义？要是还没有支持，大大可以考虑这个进去嘛

Zxilly, 2021-11-29:
不会考虑支持，你可以改这个
https://github.com/Zxilly/UA2F/blob/4b06db820f3dec98533c6798a1ad12648ecb1b19/src/ua2f.c#L446
替换字符串，并修改后面的字符串长度
GS那个替换成正常UA应该是追踪了连接，开销更大，UA2F仅修改包，只能替换UA为等长字符串。

SenkuuChin, 2021-11-29:
我修改了的话，可以直接安装，比如说用 opkg 安装进去吗？ 我现在是已经编译好所有，在运行了。
还是说我修改这个源码后，必须重新编译才行？

SenkuuChin, 2021-11-29:
@Zxilly 我修改了的话，可以直接安装，比如说用 opkg 安装进去吗？ 我现在是已经编译好所有，在运行了。
还是说我修改这个源码后，必须重新编译才行？

Zxilly, 2021-11-29:
修改源码编译成ipk，opkg安装ipk

SenkuuChin, 2021-11-29:
@Zxilly OK，明白了。谢谢啦

closed 2021-11-29 by Zxilly

xia0028, 2025-06-17:
> [@Zxilly](https://github.com/Zxilly) OK，明白了。谢谢啦

你好能帮我编译一份吗？有偿。我们学校最近好像针对ua为一串F的用户（即使只有一台设备）做了限制，也会被视为多设备的情况

Zxilly, 2025-06-17:
这是很古老的版本了，现在你 `uci set ua2f.main.custom_ua="Test UA/1.0"` 就行，请看 README