Zxilly/UA2F #45: 运行ua2f，进程不断退出重启，UA未改变

unclear1427, 2022-03-15:
请问大佬，我将ua2f编译到固件内的，ua2f service enable开启了自启动，发现偶尔能改变ua为FFFFFF，我查看了日志后发现ua2f进程不断退出再restart，部分系统日志如下：
```
Tue Mar 15 19:31:24 2022 user.notice nlbwmon: Reloading nlbwmon due to ifup of loopback (lo)
Tue Mar 15 19:31:26 2022 daemon.notice procd: /etc/rc.d/S99ua2f: iptables: Chain already exists.
Tue Mar 15 19:31:26 2022 daemon.info procd: - init complete -
Tue Mar 15 19:31:26 2022 user.notice : Try to start UA2F processor at [5029].
Tue Mar 15 19:31:26 2022 user.notice : UA2F processor start at [5029].
Tue Mar 15 19:31:26 2022 syslog.notice UA2F[5029]: Pipset inited.
Tue Mar 15 19:31:26 2022 syslog.notice UA2F[5029]: UA2F has inited successful.
Tue Mar 15 19:31:35 2022 user.notice firewall: Reloading firewall due to ifup of wan (wan)
Tue Mar 15 19:31:36 2022 ### user.info : Received SIGTERM, kill child 5029
Tue Mar 15 19:31:36 2022 kern.info kernel: [   70.698050] do_page_fault(): sending SIGSEGV to ua2f for invalid read access from 00000000
Tue Mar 15 19:31:36 2022 kern.info kernel: [   70.706373] epc = 77d97681 in libmnl.so.0.2.0[77d96000+12000]
Tue Mar 15 19:31:36 2022 kern.info kernel: [   70.712229] ra  = 0040185b in ua2f[400000+3000]
Tue Mar 15 19:31:36 2022 user.notice : Try to start UA2F processor at [5171].
Tue Mar 15 19:31:36 2022 user.notice : UA2F processor start at [5171].
Tue Mar 15 19:31:36 2022 syslog.notice UA2F[5171]: Pipset inited.
Tue Mar 15 19:31:36 2022 syslog.notice UA2F[5171]: UA2F has inited successful.
Tue Mar 15 19:31:37 2022 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan (wan)
```
内核日志如下：
```
[   36.176372] br-lan: port 2(lan2) entered forwarding state
[   45.436549] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   45.443236] br-lan: port 4(wlan0) entered blocking state
[   45.448623] br-lan: port 4(wlan0) entered forwarding state
[   46.828434] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   46.835077] br-lan: port 3(wlan1) entered blocking state
[   46.840505] br-lan: port 3(wlan1) entered forwarding state
[   70.698050] do_page_fault(): sending SIGSEGV to ua2f for invalid read access from 00000000
[   70.706373] epc = 77d97681 in libmnl.so.0.2.0[77d96000+12000]
[   70.712229] ra  = 0040185b in ua2f[400000+3000]
[ 2932.133755] do_page_fault(): sending SIGSEGV to ### ua2f for invalid read access from 00000000
[ 2932.142089] epc = 77d90681 in libmnl.so.0.2.0[77d8f000+12000]
[ 2932.147908] ra  = 0040185b in ### ua2f[400000+3000]
```
[系统日志（全部）.txt](https://github.com/Zxilly/UA2F/files/8253517/default.txt)

Zxilly, 2022-03-15:
草，是ua2f炸了，等我琢磨下最近改啥了

Zxilly, 2022-04-21:
可能在 https://github.com/Zxilly/UA2F/commit/1afac6f4a31d729522c23643236186dfe17610b2 中被修复，麻烦看下最新版

closed 2022-06-28 by github-actions[bot]