Zxilly/UA2F #49: 几个问题希望可以得到开发者解惑

yaotthaha, 2022-04-17:
1：  MWAN3 会影响 UA2F 吗
2：  为什么 UA2F 挂在 FORWARD，不应该挂在 OUPUT 上吗

imguoliwei, 2022-04-17:
1.可能有影响。我改写的[UA2F-cpp](https://github.com/imguoliwei/UA2F-cpp)可以禁用CONNMARK，这样就没有影响。
2.我是挂在POSTROUTING的。一般情况下不应该挂在 OUPUT 。

yaotthaha, 2022-04-17:
1 可以细说有什么影响吗
2 我是觉得挂在 POSTROUTING 或者 OUTPUT 都行，当然 POSTROUING 会好一些，开发者挂在 FORWARD 是为了防止后面程序的冲突，才事先修改的吗

imguoliwei, 2022-04-17:
1 见这个[issue](https://github.com/Zxilly/UA2F/issues/13)
2 OUTPUT处理路由器自己发出的数据包，FORWARD处理转发的数据包。POSTROUTING处理OUTPUT和FORWARD的输出，具体看[这个图](https://wiki.nftables.org/wiki-nftables/index.php/Netfilter_hooks)

Zxilly, 2022-04-17:
1. 会有影响，因为我不是按位操作的，每次修改都会影响整个标记。
2. @imguoliwei 已经解释了，`POSTROUTING`也是可以的，我挂在 `FORWARD` 是因为 `UA2F` 的设计初衷是修改流经流量，路由器本身的请求是不修改的。既然你能理解 `netfilter`，你可以按你自己的需求来。
[image]

yaotthaha, 2022-04-18:
感谢解答

closed 2022-04-18 by yaotthaha