Zxilly/UA2F #50: 电脑连接WIFI之后，UA2F崩溃

anna5543whi, 2022-04-20:
单纯手机连接WIFI并不会崩溃，电脑一连接有网络活动的话校园网会被秒踢下线。看了日志才判断到是UA2F崩溃了，麻烦大佬看看是什么情况。

**系统日志如下：**
```
Wed Apr 20 18:14:03 2022 user.info : Received SIGTERM, kill child 12662
Wed Apr 20 18:14:03 2022 kern.info kernel: [15642.452772] do_page_fault(): sending SIGSEGV to ua2f for invalid read access from 00000000
Wed Apr 20 18:14:03 2022 kern.info kernel: [15642.469341] epc = 77db0681 in libmnl.so.0.2.0[77daf000+12000]
Wed Apr 20 18:14:03 2022 kern.info kernel: [15642.480916] ra  = 0040185b in ua2f[400000+3000]
Wed Apr 20 18:14:03 2022 user.notice : Try to start UA2F processor at [12875].
Wed Apr 20 18:14:03 2022 user.notice : UA2F processor start at [12875].
Wed Apr 20 18:14:03 2022 syslog.notice UA2F[12875]: Pipset inited.
Wed Apr 20 18:14:03 2022 syslog.notice UA2F[12875]: UA2F has inited successful.
Wed Apr 20 18:14:03 2022 user.notice firewall: Reloading firewall due to ifup of wan (wan)
Wed Apr 20 18:14:04 2022 user.info : Received SIGTERM, kill child 12875
Wed Apr 20 18:14:04 2022 kern.info kernel: [15643.265057] do_page_fault(): sending SIGSEGV to ua2f for invalid read access from 00000000
Wed Apr 20 18:14:04 2022 kern.info kernel: [15643.281676] epc = 77d45681 in libmnl.so.0.2.0[77d44000+12000]
Wed Apr 20 18:14:04 2022 kern.info kernel: [15643.293240] ra  = 0040185b in ua2f[400000+3000]
Wed Apr 20 18:14:04 2022 user.notice : Try to start UA2F processor at [13036].
Wed Apr 20 18:14:04 2022 user.notice : UA2F processor start at [13036].
Wed Apr 20 18:14:04 2022 syslog.notice UA2F[13036]: Pipset inited.
Wed Apr 20 18:14:04 2022 syslog.notice UA2F[13036]: UA2F has inited successful.
```

**内核日志如下：**
```
[12430.315482] do_page_fault(): sending SIGSEGV to ua2f for invalid read access from 00000000
[12430.332029] epc = 77dce681 in libmnl.so.0.2.0[77dcd000+12000]
[12430.343721] ra  = 0040185b in ua2f[400000+3000]
[12433.915654] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control off
[12433.930260] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
[12434.380179] do_page_fault(): sending SIGSEGV to ua2f for invalid read access from 00000000
[12434.396721] epc = 77dc8681 in libmnl.so.0.2.0[77dc7000+12000]
[12434.408344] ra  = 0040185b in ua2f[400000+3000]
[12435.185687] do_page_fault(): sending SIGSEGV to ua2f for invalid read access from 00000000
[12435.202279] epc = 77dee681 in libmnl.so.0.2.0[77ded000+12000]
[12435.213961] ra  = 0040185b in ua2f[400000+3000]
```

[系统日志(全部).txt](https://github.com/Zxilly/UA2F/files/8520646/default.txt)
[内核日志(全部).txt](https://github.com/Zxilly/UA2F/files/8520647/default.txt)

Zxilly, 2022-04-20:
应该是最近的更改引入了Bug，能不能抓一下包呢？有助于确定问题。

anna5543whi, 2022-04-20:
tcpdump抓包到断网才停止，https://wwt.lanzoub.com/ihlrz03ic9le

Zxilly, 2022-04-21:
可能在 https://github.com/Zxilly/UA2F/commit/1afac6f4a31d729522c23643236186dfe17610b2 中被修复，麻烦看下最新版

anna5543whi, 2022-04-21:
麻烦了，等过两天我回去了再测试看看。

imguoliwei, 2022-04-21:
这个issue提到了两个问题：
1.程序意外地收到了SIGTERM信号
2.收到SIGTERM信号后出现了非法内存访问

我没有遇到第一个问题，我感觉可能是内存不够之类的问题，未必是代码的问题
我经常遇到第二个问题，具体情况是，只要我手动结束UA2F进程，就有一定概率出现上面的非法内存访问。

我最近删除了killChild函数中的`mnl_socket_close(nl);`，发现非法内存访问的现象消失了。无论我重启多少次UA2F，似乎都不会出现非法内存访问。查阅[文档](https://man7.org/linux/man-pages/man2/kill.2.html)发现，kill函数会在发送信号后返回，但是并没有说在进程结束后才返回。因此，当`mnl_socket_close(nl);`关闭socket时，UA2F子进程可能还在读写socket，这样就产生了非法内存访问。
我通过unique_ptr在进程结束时关闭socket。C语言可以考虑使用atexit函数。

Zxilly, 2022-04-21:
@imguoliwei 我能确定的bug就是刚刚改的这个，在UA长度为0时会出现野指针。然后跑了下address sanitize也没看到什么问题。
这个问题我回头看看。看了之前的page fault还真有可能是这个。

anna5543whi, 2022-04-23:
3.8-17版本还是跟之前一样的情况
[系统日志(全部)0423.txt](https://github.com/Zxilly/UA2F/files/8547411/0423.txt)
[内核日志(全部)0423.txt](https://github.com/Zxilly/UA2F/files/8547412/0423.txt)

imguoliwei, 2022-04-24:
我注意到日志中 `user.info : Received SIGTERM, kill child`前面经常出现这样的记录
`Tue Mar 15 19:31:35 2022 user.notice firewall: Reloading firewall due to ifup of wan (wan)`
`Fri Apr 22 22:56:53 2022 user.notice firewall: Reloading firewall due to ifup of wan6 (eth0.2)`
`Fri Apr 22 22:57:09 2022 daemon.notice netifd: Interface 'wan' is now up`
因此UA2F重启很可能与启动脚本设置的某些触发条件有关

可以尝试禁用UA2F的启动脚本，直接通过命令行运行`/usr/bin/ua2f  &`，手动配置防火墙，看看UA2F还会不会重启。

Zxilly, 2022-04-24:
应该是解决 `mnl_socket_close(nl);` 的问题就好了

MADAXIAO, 2022-04-24:
Sun Apr 24 19:27:45 2022 syslog.notice UA2F[6816]: UA2F has inited successful.
Sun Apr 24 19:28:41 2022 syslog.info UA2F[6816]: UA2F has handled 8 ua http, 1351 tcp. Set 0 mark and 7 noUA mark in 56 seconds
Sun Apr 24 19:29:40 2022 syslog.info UA2F[6816]: UA2F has handled 16 ua http, 2419 tcp. Set 6 mark and 14 noUA mark in 1 minutes and 55 seconds
Sun Apr 24 19:29:40 2022 syslog.info UA2F[6816]: UA2F has handled 32 ua http, 2669 tcp. Set 6 mark and 14 noUA mark in 1 minutes and 55 seconds
Sun Apr 24 19:30:23 2022 syslog.info UA2F[6816]: UA2F has handled 64 ua http, 4149 tcp. Set 6 mark and 29 noUA mark in 2 minutes and 38 seconds
Sun Apr 24 19:48:24 2022 syslog.info UA2F[6816]: UA2F has handled 128 ua http, 13606 tcp. Set 13 mark and 139 noUA mark in 20 minutes and 39 seconds
Sun Apr 24 20:07:23 2022 syslog.info UA2F[6816]: UA2F has handled 256 ua http, 24523 tcp. Set 25 mark and 224 noUA mark in 39 minutes and 38 seconds
Sun Apr 24 20:20:09 2022 syslog.info UA2F[6816]: UA2F has handled 512 ua http, 60717 tcp. Set 48 mark and 300 noUA mark in 52 minutes and 24 seconds
Sun Apr 24 20:26:43 2022 syslog.info UA2F[6816]: UA2F has handled 1024 ua http, 69745 tcp. Set 60 mark and 357 noUA mark in 58 minutes and 58 seconds
Sun Apr 24 20:32:51 2022 syslog.info UA2F[6816]: UA2F has handled 2048 ua http, 83466 tcp. Set 127 mark and 506 noUA mark in 1 hours, 5 minutes and 6 seconds
Sun Apr 24 20:52:48 2022 syslog.info UA2F[6816]: UA2F has handled 4096 ua http, 199570 tcp. Set 288 mark and 983 noUA mark in 1 hours, 25 minutes and 3 seconds
Sun Apr 24 21:23:57 2022 user.info : Received SIGTERM, kill child 6816
Sun Apr 24 21:23:57 2022 user.notice : Try to start UA2F processor at [13923].
Sun Apr 24 21:23:57 2022 user.notice : UA2F processor start at [13923].
Sun Apr 24 21:23:57 2022 syslog.notice UA2F[13923]: Pipset inited.
Sun Apr 24 21:23:57 2022 syslog.notice UA2F[13923]: UA2F has inited successful.
Sun Apr 24 21:23:58 2022 user.info : Restarting firewall on custom /etc/firewall.user change
Sun Apr 24 21:24:54 2022 syslog.info UA2F[13923]: UA2F has handled 8 ua http, 1932 tcp. Set 0 mark and 39 noUA mark in 57 seconds
Sun Apr 24 21:25:03 2022 syslog.info UA2F[13923]: UA2F has handled 16 ua http, 3232 tcp. Set 1 mark and 44 noUA mark in 1 minutes and 6 seconds
Sun Apr 24 21:25:33 2022 syslog.info UA2F[13923]: UA2F has handled 32 ua http, 3842 tcp. Set 1 mark and 45 noUA mark in 1 minutes and 36 seconds
Sun Apr 24 21:26:28 2022 syslog.info UA2F[13923]: UA2F has handled 64 ua http, 7941 tcp. Set 4 mark and 82 noUA mark in 2 minutes and 31 seconds
Sun Apr 24 21:30:42 2022 syslog.info UA2F[13923]: UA2F has handled 128 ua http, 53604 tcp. Set 17 mark and 166 noUA mark in 6 minutes and 45 seconds
Sun Apr 24 21:49:43 2022 syslog.info UA2F[13923]: UA2F has handled 256 ua http, 68434 tcp. Set 31 mark and 280 noUA mark in 25 minutes and 46 seconds
大佬我用的3.8-17的貌似也会出现这个问题，就是突然断一下网，日志是这样的

MADAXIAO, 2022-04-24:
我是按照教程手动添加了防火墙配置，然后运行了# Enable the daemon
uci set ua2f.enabled.enabled=1
# At your option set fw rules
uci set ua2f.firewall.handle_fw=1
uci set ua2f.firewall.handle_tls=1
uci set ua2f.firewall.handle_mmtls=1
uci set ua2f.firewall.handle_intranet=1

# Apply your modifications
uci commit ua2f

service ua2f enable
# Start the daemon
service ua2f start

imguoliwei, 2022-04-25:
我发现我需要更正我对于`mnl_socket_close(nl);`问题的分析。实际情况是，对fork函数返回的两个进程的全局变量进行赋值是独立的，因此主进程中nl的值总是为NULL，mnl_socket_close函数尝试解引用NULL，就发生了异常。
在代码中使用全局变量要特别小心，尤其是`child_status`和`nl`。这两个变量在主进程和子进程都可能被访问，但是它们的值有两个版本。
如果确实要关闭socket，无论通过signal还是atexit，基本上还得通过全局变量来传递参数。鉴于子进程可能会继承fork前主进程的退出处理函数，如果在fork前设置退出处理函数，函数的代码需要区分是在哪个进程。
似乎也可以不关闭socket。貌似只要进程结束了，socket就会自动关闭。

imguoliwei, 2022-04-25:
由于C语言没有RAII，代码重复出现了`pktb_free(pktb);`，维护这样的代码不得不非常小心。
我目前发现两种解决方案，一种是[Implementing RAII in pure C](https://stackoverflow.com/questions/368385/implementing-raii-in-pure-c)，另外一种是使用[goto](https://www.netfilter.org/projects/libnetfilter_queue/doxygen/html/libnetfilter__queue_8c_source.html#l00396)

另外，其实我当时提交 #42 PR的时候就意识到代码中有两个nfq_send_verdict不是很好。我考虑过goto，但最终决定把那些代码放在一个子程序里面然后return，在编译时加上`-O3`选项以自动开启`-finline-functions`

Zxilly, 2022-04-25:
看起来问题是莫名其妙收到的SIGTERM，我去找找啥发的这个。。。

wlshdjj, 2022-08-20:
同样发现这个问题。
[ 1798.276670] ua2f[11457]: segfault at 0 ip 00007f9d36a96277 sp 00007ffd9ddd9f90 error 4 in libmnl.so.0.2.0[7f9d36a96000+1000]
[ 1798.279237] Code: c7 00 1c 00 00 00 eb 13 83 7c 24 30 0c 74 10 ff 15 3e 3d 00 00 c7 00 16 00 00 00 48 83 c8 ff 48 83 c4 68 c3 55 53 48 89 fb 50 <8b> 3f ff 15 59 3d 00 00 48 89 df 89 c5 ff 15 56 3d 00 00 89 e8 5a

问题不在于是什么原因导致的kill，而是进程不能被正常终止。
至于楼主这个问题，是由于重新拨号引起firewall restart导致ua2f启动脚本重启，是由于不完善的ua2f.include处理方案造成的不良后果。启动脚本有些地方也不对，运行有逻辑报错，我受不了，大刀阔斧修改了启动脚本，楼主这个问题是肯定解决了（实际上firewall restart时不需要重启ua2f，只需要重新添加iptables规则即可），另外我写了luci。我用不惯github，不知道怎么提交，作者如有看到麻烦联系我一下，我发给你。

Zxilly, 2022-08-20:
@wlshdjj 可以把patch发出来，拿个pastebin都行

wlshdjj, 2022-08-22:
链接: https://pan.baidu.com/s/1x32RcNzV4FJ0aOuIC8AzLg?pwd=378c 提取码: 378c
直接放网盘吧。config get是过时的命令集，uci get是用于取代它的新一代命令，它是单命令，运行更加高效不易出错。
 [ "$handle_intranet" -eq 1 ]，如果handle_intranet为空时即逻辑报错， [ "$handle_intranet" == 1 ]就不会报错。
ipset destroy nohttp前需要先flush。
实际上ua2f运行后基本没有需要重启的或者关闭的时候，因此把socket注销命令去掉了，免得内核爆炸，除非你非要去重启或者关闭，否则系统运行中遇到接口重启、fw重启只需要管iptables那一部分。
不同于发行版linux，直接在start中用exit来处理enabled，在openwrt中有一定几率会引起启动失败，特别是在luci中配置enabled后按下应用，程序无法运行，非得去启动项里start才能运行，这是写了多年脚本深受其害才找到的原因。

Zxilly, 2022-08-27:
@wlshdjj https://github.com/Zxilly/UA2F/commit/c2388e31865d64d879e79cba0e238cfc4692471d 已经加上了

closed 2022-11-04 by github-actions[bot]