Zxilly/UA2F #56: UA2F正常running但UA未修改

reamger, 2022-09-22:
# lean大神的源码编译的openwrt，按照博客的教材编译安装ua2f
1. ikuai+op双路由，ikuai(10.10.10.2/24)多线拨号，op(10.10.10.10/24)用来多设备伪装
2. op只有一个lan口，网关是10.10.10.2，开启DHCP (通告的网关和dns服务器都是op)
3. op已关闭Turbo ACC 网络加速，开启了smartdns，未开启魔法

笔记本电脑通过AP连接op上网，测试ua未修改QwQ
 
op的防火墙如图所示：
[image]

系统日志：
[image]

搞了一天了求求大佬帮助ORZ

Apromise01, 2022-09-25:
大佬我也是没改变ua，
[image]
syslog.info UA2F[5992]: UA2F has handled 8 ua http, 155 tcp. Set 0 mark and 0 noUA mark in 1 minutes and 19 seconds

然后手动启动显示Set cannot be created: set with the same name already exists
iptables: Chain already exists.

Zxilly, 2022-09-26:
@Apromise01 你上次退出失败了，手动删 iptables chain 和 ipset

Apromise01, 2022-09-26:
大佬方便说下具体吗，我是跟着教程走的小白
是在openwrt后台操作还是ssh连接输入代码

reamger, 2022-09-27:
> 

那个图片里一直在启动是我手动service ua2f restart重启的，不过现在UA2F好像确实在自动重启了QwQ
现在的logread（挂着运行了三天）：
[image]

尝试手动运行ua2f：
[image]

Zxilly, 2022-09-27:
对啊，Operation not permitted 说明对应的内核功能不支持。
https://github.com/Zxilly/UA2F/blob/965fd28811527ef30eb53e3689be44a95a0dbcd1/src/ua2f.c#L413-L414

reamger, 2022-09-27:
有没有可能是因为我是第二次编译，而没有执行make clean导致内核没有重新编译？

reamger, 2022-09-27:
不行呀，make clean之后再编译也还是` mnl_cb_run: Operation not permitted`
难道得 make dirclean ?

Zxilly, 2022-09-28:
配置看起来没什么毛病，应该是你本地编译的时候有什么问题吧

Zxilly, 2022-09-28:
我注意到你这个是 Not permitted，如果同时有两个UA2F在运行会出现这个。你测试的时候确定 service 关闭了吗？
ps aux | grep ua2f 看一下

reamger, 2022-09-28:
测试的时候service确实没有关闭，关闭之后测试：
[image]

直接运行ua2f没有输出，在直接运行ua2f时访问 http://ua.233996.xyz/ UA没有更改。

启动service时如下情况：
[image]
UA仍没有修改，日志如下：
[image]

Zxilly, 2022-09-28:
那就是默认的防火墙不适用于你的配置，你得自己写规则。

`iptables -t mangle -A ua2f -j NFQUEUE --queue-num 10010`

最后把所有流量打到 `NFQUEUE` 的 10010 就行

Butyllithium, 2022-11-06:
我遇到了和你一样的问题，我通过更改该选项为“接受"解决了
[image]

[image]

menglilongxia, 2022-11-06:
为啥我这个非标准端口测试可以，标准就不行了

Butyllithium, 2022-11-06:
> 为啥我这个非标准端口测试可以，标准就不行了

防火墙规则没设置好吧，试试把作者博客里的规则复制过去

menglilongxia, 2022-11-06:
> > 为啥我这个非标准端口测试可以，标准就不行了
> 
> 防火墙规则没设置好，尝试把博客作者的规则复制过去

哦哦，原来是我挂的梯子，破案了

reamger, 2022-11-26:
因为ipv6的原因把网络拓扑改了下，现在是openwrt直接拨号了，没做任何其他修改UA成功变F
特此留下记录已期帮助后来者

closed 2022-11-26 by reamger

bobandbrony, 2024-06-16:
[image]
这样是正常的吗，ua是成功修改了的