Zxilly/UA2F #62: https网页无法修改ua

Chordp, 2022-12-05:
近期学校的校园网又不能用了，搞了好久突然发小他ios弹窗证书失效 显示是drcom。
突然想到可能是tls用https://ua2.233996.xyz/  测试了一下 果然 http和ip:port都没问题 https 出问题了

编译的是最新的ua2f，启动了handle_tls 
```
# Enable the daemon
uci set ua2f.enabled.enabled=1
# At your option set fw rules
uci set ua2f.firewall.handle_fw=1
uci set ua2f.firewall.handle_tls=1
uci set ua2f.firewall.handle_mmtls=1
uci set ua2f.firewall.handle_intranet=1

# Apply your modifications
uci commit ua2f

service ua2f enable
# Start the daemon
service ua2f start
```

```
你的真实UA是(服务器获取的UA)
Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/107.0.0.0 Safari/537.36 Edg/107.0.1418.62
你的浏览器UA是
Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/107.0.0.0 Safari/537.36 Edg/107.0.1418.62
```

Zxilly, 2022-12-05:
https本身就是安全的，你是不是装了学校的CA被中间人了

Chordp, 2022-12-05:
> https本身就是安全的，你是不是装了学校的CA被中间人了

很有可能 ，今天我手机弹证书到期就很奇怪
[image]

Chordp, 2022-12-05:
> > https本身就是安全的，你是不是装了学校的CA被中间人了
> 
> 很有可能 ，今天我手机弹证书到期就很奇怪

[image]

Zxilly, 2022-12-05:
这就是中间人攻击了，但是这么干的后果应该是你打不开任何 https 的网页，同时也不会泄露自己的 UA。
因为在发送 http header 之前的 ssl handshake 就失败了

Zxilly, 2022-12-05:
建议你检查下自己是不是装了啥乱七八糟的证书，CA 不能乱装的

Chordp, 2022-12-05:
> 建议你检查下自己是不是装了啥乱七八糟的证书，CA 不能乱装的

我并没有乱装ca，反正很奇怪

Chordp, 2022-12-05:
> 建议你检查下自己是不是装了啥乱七八糟的证书，CA 不能乱装的

上面的证书并不是我安装的

closed 2023-02-11 by github-actions[bot]