Zxilly/UA2F #70: 并非UA2F问题，但觉得可以进来讨论下

Kyaru2333, 2023-02-01:
在经过改UA，TCPTimestamp，IPID，TTL等方法无效后，我又进行了一个学期的探索，总算把学校判断共享的策略搞明白了。
-----------------------------废话-------------------------------
某一天半夜凌晨三点起床，发现校园网后台依然被检测到了共享。但是当时我确信所有人都已经不在使用网络（仅有手机自动连接）。于是我重新登录校园网并且路由器开始抓包。不出意料，两分钟过去后浏览器再一次提示多用户共享。仔细分析抓包，发现除了generate_204的请求以外，还有微信相关的请求。于是，关闭路由器wifi，重新登陆校园网，用电脑模拟不同手机的generate_204请求，都没有被ban。所以我重新打开路由器wifi，继续抓包，发现依然是generate_204和微信相关连接。所以我开始怀疑是和微信有关。
后来研究校园网脚本登录的时候，发现API包含"panabit_login"字样，经过网上的资料搜寻以及panabit论坛的截图，基本确定学校用Panabir防火墙
------------------------------废话结束---------------------------
后来在panabit论坛得知可以自己安装panabit，于是虚拟机安装了panabit并且单独分析了panabit的共享检测插件，发现原理是通过检查微信连接会话数（还有还有LOL端游和王者荣耀手游的）（panabit官方管这叫tcpflow_ttl)，从而判定是否存在共享行为。然后就进行实验，根据Panabit后台的数据，只要是微信的IP，哪怕我就是用浏览器GET一下都会被判断为微信连接。所以panabit会话数认定里还加了tcpflow_ttl认证。用浏览器GET产生的tcpflow_ttl值是很小的（个位数），而如果遇到真微信的话这个值通常有1400。所以我推测这个值是微信的长连接时间，这个值会随着时间流逝而变小。
另外王者荣耀手游和LOL端游的检测原理大致一样。

所以现在问题是：
1. tcpflow_ttl究竟是个啥
2. 这个值能不能修改
若是能在路由器改这个值就基本上能起飞了

Zxilly, 2023-02-01:
应该可以conntrack追踪连接，然后在 `tcpflow_ttl` 置 0 之前向双端发送 `TCP_RESET` 来打断？

但是游戏如果用这个策略，可能会带来比较恼火的体验。

Zxilly, 2023-02-01:
openwrt 的话，可以通过连接老化来粗略的处理。

```
net.netfilter.nf_conntrack_tcp_timeout_established=300
```

当然，体验会差一些

Kyaru2333, 2023-02-01:
> 应该可以 `conntrack` 追踪连接，然后在 `tcpflow_ttl` 置 0 之前向双端发送 `TCP_RESET` 来打断？
> 
> 但是游戏如果用这个策略，可能会带来比较恼火的体验。

实在抱歉，我刚刚又看了一下，他那个ttl应该是tcp已经建立连接的时长，是不断增加的

Kyaru2333, 2023-02-01:
> openwrt 的话，可以通过连接老化来粗略的处理。
> 
> ```
> net.netfilter.nf_conntrack_tcp_timeout_established=300
> ```
> 
> 当然，体验会差一些

我也不太清楚他是怎么工作的，就是那个“tcpflow_ttl”，似乎一建立连接就会有个很高的值，将net.netfilter.nf_conntrack_tcp_timeout_established=300加进去后panabit后台显示那个值一样是很高

Kyaru2333, 2023-02-01:
另外对于游戏的处理他也有一个udpflow_ttl值

Zxilly, 2023-02-01:
指的应该就是单个tcp连接的持续时间，udp没有连接的概念，指的应该是固定的一对三元组。

Zxilly, 2023-02-01:
`nf_conntrack_tcp_timeout_established` 不会主动打断连接，应该是微信发心跳包了

Kyaru2333, 2023-02-01:
> 指的应该就是单个 tcp 连接的持续时间，udp 没有连接的概念，指的应该是固定的一对三元组。

那对于这种是不是除了用代理就无解了？
（panabit也是真的狠，他好像把微信的ip都记录了个遍…）

Zxilly, 2023-02-01:
udp 应该没办法了，tcp 可以尝试我前面说的 ·在 tcpflow_ttl 置 0 之前向双端发送 TCP_RESET 来打断·

Kyaru2333, 2023-02-01:
> udp 应该没办法了，tcp 可以尝试我前面说的 `在 tcpflow_ttl 置 0 之前向双端发送 TCP_RESET 来打断`

他也不是说置0就ban，他是统计所有微信聊天的连接中“tcpflow_ttl”这一项大于某个设定值的连接的数量（判定为微信会话）大于管理员的设定值就会ban
（但是这个值在panabit的详细信息里是递减的，一秒会减1）
所以这样看起来RESET也没用？
我的想法是在发送之前就更改这个所谓“tcpflow_tcp”值，但是我抓包也没发现这个值，或者是这个值的近似值，也不知道他是怎么判断的

Zxilly, 2023-02-01:
`tcpflow_ttl` 本身就是一个统计值，指的是一个 tcp 连接已经被建立了多久，递减的情况下，指的是还有多少这个连接应该过期或者被视为长连接。
只要在一个连接到达限制之前就打断它，那么就永远不会有连接超出限制了。

Kyaru2333, 2023-02-01:
感谢解答！
因为打断RESET这些实在不会写，我目前考虑硬碰硬去解决，将QQ和微信的消息，王者LOL这四个检测项目用爱快调度到移动数据。

closed 2023-02-01 by Kyaru2333