Zxilly/UA2F #71: lean源码编译出错

goodwinY, 2023-04-14:
make[3]: Entering directory '/home/goodwin/openwrt/bohao/lede/package/UA2F'
rm -f /home/goodwin/openwrt/bohao/lede/build_dir/target-x86_64_musl/UA2F-3.10/.built
touch /home/goodwin/openwrt/bohao/lede/build_dir/target-x86_64_musl/UA2F-3.10/.built_check
x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/goodwin/openwrt/bohao/lede/build_dir/target-x86_64_musl/UA2F-3.10=UA2F-3.10 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -L/home/goodwin/openwrt/bohao/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/goodwin/openwrt/bohao/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro -lmnl -lnetfilter_queue -lpthread /home/goodwin/openwrt/bohao/lede/build_dir/target-x86_64_musl/UA2F-3.10/ua2f.c /home/goodwin/openwrt/bohao/lede/build_dir/target-x86_64_musl/UA2F-3.10/statistics.c /home/goodwin/openwrt/bohao/lede/build_dir/target-x86_64_musl/UA2F-3.10/child.c /home/goodwin/openwrt/bohao/lede/build_dir/target-x86_64_musl/UA2F-3.10/util.c /home/goodwin/openwrt/bohao/lede/build_dir/target-x86_64_musl/UA2F-3.10/cache.c -o /home/goodwin/openwrt/bohao/lede/build_dir/target-x86_64_musl/UA2F-3.10/ua2f
/home/goodwin/openwrt/bohao/lede/build_dir/target-x86_64_musl/UA2F-3.10/ua2f.c: In function 'main':
/home/goodwin/openwrt/bohao/lede/build_dir/target-x86_64_musl/UA2F-3.10/ua2f.c:303:44: error: 'sizeof_buf' undeclared (first use in this function)
  303 |         ret = mnl_socket_recvfrom(nl, buf, sizeof_buf);
      |                                            ^~~~~~~~~~
/home/goodwin/openwrt/bohao/lede/build_dir/target-x86_64_musl/UA2F-3.10/ua2f.c:303:44: note: each undeclared identifier is reported only once for each function it appears in
/home/goodwin/openwrt/bohao/lede/build_dir/target-x86_64_musl/UA2F-3.10/child.c:9:8: error: unknown type name '__pid_t'
    9 | static __pid_t child_pid = 0;
      |        ^~~~~~~
/home/goodwin/openwrt/bohao/lede/build_dir/target-x86_64_musl/UA2F-3.10/child.c:10:8: error: unknown type name 'u_int8_t'
   10 | static u_int8_t failure_count = 0;
      |        ^~~~~~~~
/home/goodwin/openwrt/bohao/lede/build_dir/target-x86_64_musl/UA2F-3.10/child.c:12:7: error: unknown type name 'u_int8_t'
   12 | const u_int8_t MAX_RETRY_COUNT = 8;
      |       ^~~~~~~~
make[3]: *** [Makefile:61: /home/goodwin/openwrt/bohao/lede/build_dir/target-x86_64_musl/UA2F-3.10/.built] Error 1
make[3]: Leaving directory '/home/goodwin/openwrt/bohao/lede/package/UA2F'
time: package/UA2F/compile#0.11#0.02#0.12
    ERROR: package/UA2F failed to build.
make[2]: *** [package/Makefile:116: package/UA2F/compile] Error 1
make[2]: Leaving directory '/home/goodwin/openwrt/bohao/lede'
make[1]: *** [package/Makefile:110: /home/goodwin/openwrt/bohao/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/goodwin/openwrt/bohao/lede'
make: *** [/home/goodwin/openwrt/bohao/lede/include/toplevel.mk:231：world] 错误 2

Zxilly, 2023-04-14:
emmm...我在做一些重构，现在你可以 checkout 这个版本
https://github.com/Zxilly/UA2F/commit/da8ed678810e34b74f4a974f346a56a6f195cca9

goodwinY, 2023-04-14:
我不急，我只是例行升级，我推迟了升级，你的回复是真滴快呀

Zxilly, 2023-04-21:
更新完了，测试一下v4能不能用

goodwinY, 2023-04-21:
> 更新完成了，测试一下v4不能用

正在云编译

closed 2023-04-22 by goodwinY

Zxilly, 2023-04-22:
不用急着关，v4基本完全重写了，看下是不是有啥其他的问题