Zxilly/UA2F #76: 22.03 似乎在UA2F启动后一段时间会删除现有的iptables链

Ljzd-PRO, 2023-05-08:
虽然 22.03 用的是 nftables，但是似乎还保留了 iptables，只是不推荐两个同时使用
> **检测到旧版规则**
> 系统上存在旧版 iptables 规则。 不鼓励混合使用 iptables 和 nftables 规则，这可能会导致流量过滤不完整。

如果是直接用 `ua2f` 或 `service ua2f start` 启动的话，似乎无法成功自动创建防火墙规则，UA没有被修改。

手动添加 iptables 链以后再启动的话，手动添加的会消失，似乎是被删除了，UA还是没有被修改

先启动 UA2F，再添加 iptables 链的话，发现可以生效，UA会被修改。

Zxilly, 2023-05-08:
应该是init脚本的问题，我检查下。
nftables的支持过几天加吧，实在是不太熟悉

Ljzd-PRO, 2023-05-08:
突然发现 `service ua2f start` 无效是因为UCI配置文件里没有开启UA2F，`enable`设为了0😂

可能不能正常运行的是直接使用 `ua2f` 命令，可能是和这个有关？
https://github.com/Zxilly/UA2F/blob/58943101d73e5b58d7e1fc099ff58ea5881df911/openwrt/files/ua2f.uci#L3-L10

22.03 以后防火墙配置里的include要求用nftables的规则
https://openwrt.org/docs/guide-user/firewall/firewall_configuration#includes_for_2203_and_later_with_fw4

closed 2023-05-09 by Zxilly