Zxilly/UA2F #78: 小概率会出现漏UA问题

Po810C, 2023-05-08:
通过长时间的抓包（一下午）发现，部分APP的UA会出现不被处理的情况，drcom防代理网关会因此立即检测出多设备，且阿里系应用有两个UA字段（虽然未被处理的UA不影响防检测效果），UA2F为最新commit编译

[image]
爱奇艺

[image]
淘宝

[image]
阿里云

Po810C, 2023-05-08:
且此情况在mips和arm设备上较为常见，x86出现较少

Zxilly, 2023-05-09:
truncated的包确实是无法处理的，而且当前的架构下也处理不了。
两个UA的。。。我想想怎么处理

Zxilly, 2023-05-09:
如果可以的话能把truncated的包发出来看一下吗，我想想能怎么处理

Po810C, 2023-05-09:
> 如果可以的话能把truncated的tcp包发出来看一下吗，我想想能怎么处理

已将抓包文件发送至你的邮箱

Zxilly, 2023-05-10:
已经确定问题了，有非标准的User-Agent字段，可以加个特判解决。

commit 8c44ffa7 (2023-05-10): chore: bump version to 4.0.5

Close: #78
closed 2023-05-10 by Zxilly