Zxilly/UA2F #82: 启动服务后很多网页打不开，还有两个错误提示

ArcadiaYang, 2023-05-14:
[image]
不知道是什么问题

Zxilly, 2023-05-14:
你的ip6tables没有connmark模块

ArcadiaYang, 2023-05-14:
me 已收到，我会尽快回复你。  闪 ~~~

ArcadiaYang, 2023-05-14:
> 你的ip6tables没有connmark模块

抱歉，请问已经编译好且刷入路由器的固件怎么添加connmark模块呢

ArcadiaYang, 2023-05-14:
这个是系统日志，不太会分析
Sun May 14 14:35:28 2023 daemon.err ua2f[10677]: Error relocating /usr/bin/ua2f: __difftime64: symbol not found
Sun May 14 14:35:28 2023 daemon.err ua2f[10677]: Error relocating /usr/bin/ua2f: __pselect_time64: symbol not found
Sun May 14 14:35:28 2023 daemon.err ua2f[10677]: Error relocating /usr/bin/ua2f: __time64: symbol not found
Sun May 14 14:35:28 2023 daemon.err ua2f[10677]: Error relocating /usr/bin/ua2f: __clock_gettime64: symbol not found
Sun May 14 14:35:33 2023 daemon.err ua2f[10699]: Error relocating /usr/bin/ua2f: __difftime64: symbol not found
Sun May 14 14:35:33 2023 daemon.err ua2f[10699]: Error relocating /usr/bin/ua2f: __pselect_time64: symbol not found
Sun May 14 14:35:33 2023 daemon.err ua2f[10699]: Error relocating /usr/bin/ua2f: __time64: symbol not found
Sun May 14 14:35:33 2023 daemon.err ua2f[10699]: Error relocating /usr/bin/ua2f: __clock_gettime64: symbol not found
Sun May 14 14:35:33 2023 daemon.info procd: Instance ua2f::ua2f s in a crash loop 6 crashes, 0 seconds since last crash

ArcadiaYang, 2023-05-14:
> 这个是系统日志，不太会分析 Sun May 14 14:35:28 2023 daemon.err ua2f[10677]: Error relocating /usr/bin/ua2f: __difftime64: symbol not found Sun May 14 14:35:28 2023 daemon.err ua2f[10677]: Error relocating [...]

启用后，只有https网页可以正常打开，http网页无法打开

Zxilly, 2023-05-15:
你这系统怎么没有libc的。。。

Zxilly, 2023-05-15:
执行这个命令看一下
`ldd /usr/bin/ua2f`

ArcadiaYang, 2023-05-15:
> 执行这个命令看一下 `ldd /usr/bin/ua2f`

> 执行这个命令看一下 `ldd /usr/bin/ua2f`

[image]
z这样的

Zxilly, 2023-05-16:
你是从release下载的，还是一起编译的

Zxilly, 2023-05-16:
看起来像是因为musl1.2和1.1的abi不同导致的

1715173329, 2023-05-21:
目前 openwrt 三个 release 的 libc 版本均不一致，不是主线的话只能靠自己编译了

closed 2023-09-06 by Zxilly