Zxilly/UA2F #87: 4.1.2版本非标准端口http流量的UA不会被修改

Ljzd-PRO, 2023-05-27:
日志经常显示例如
```
Sun May 28 00:05:10 2023 syslog.warn UA2F[9448]: Unexpected connmark value: 296, Maybe other program has changed connmark?
```
的警告。4.0.3版本也有一样的警告，但是可以正常使用，只是疑似极少数情况会失效（校园网提示了多设备连接）

Ljzd-PRO, 2023-05-27:
启动时还有如下的日志输出
```
Sun May 28 00:15:45 2023 syslog.warn UA2F[10625]: Kernel does not support conntrack
Sun May 28 00:15:45 2023 syslog.err UA2F[10625]: Packet has no conntrack.
Sun May 28 00:15:50 2023 syslog.info UA2F[10877]: Statistics initialized.
Sun May 28 00:15:50 2023 syslog.info UA2F[10877]: Handler initialized.
Sun May 28 00:15:50 2023 syslog.info UA2F[10877]: Cache lock initialized
Sun May 28 00:15:50 2023 syslog.info UA2F[10877]: Cleanup thread created
```
编译的时候有设置内核的 `NETFILTER_NETLINK_GLUE_CT` 选项。4.0.3版本也没有提示`Kernel does not support conntrack`。

Ljzd-PRO, 2023-05-27:
一段时间后80端口的UA也不会被修改了

Zxilly, 2023-05-27:
说明有别的东西也在修改connmark导致冲突了，如果只有UA2F在修改，这个值不应该超过43

closed 2023-07-27 by Ljzd-PRO