Zxilly/UA2F #88: 一些没被修改的UA

Po810C, 2023-05-28:
[image]

已将相关抓包文件发送至您的邮箱

Zxilly, 2023-05-28:
已经收到，但是没看出有啥问题，估计不太好修

Po810C, 2023-05-28:
部分app会被 drcom 防代理网关检测，目前发现 驾考宝典、到梦空间 会被检测出手机型号，暂未发现是通过何种途径检测出的，还需抓包

Po810C, 2023-09-02:
从mumu模拟器抓到一些没被修改的ua，已通过邮箱发送给您

Zxilly, 2023-09-02:
已收到，看起来是因为内网流量默认不会处理，可以修改ua2f.firewall.handle_intranet改变这一行为

Po810C, 2023-09-02:
> 已收到，看起来是因为内网流量默认不会处理，可以修改ua2f.firewall.handle_intranet改变这一行为

ua2f.firewall.handle_intranet 选项已启用，内网仅有此地址会不对UA进行处理，其他内网地址的请求UA修改均正常

Po810C, 2023-09-02:
这个地址是drcom的登录认证地址，猜测之前会“漏包”的app也是因为这种方式导致被检测

Po810C, 2023-09-04:
可能需要弄清楚为什么UA2F不会修改drcom登录地址的UA的原因

Zxilly, 2023-09-04:
我大概知道了，handle_intranet没有被应用到防火墙规则上

Zxilly, 2023-09-04:
Resolved in https://github.com/Zxilly/UA2F/commit/c244bfbe6796c738e11bd18f9dfecb8ce7b39c5e

Po810C, 2023-09-04:
最新commit无法运行，报错如图
[image]

Zxilly, 2023-09-05:
写错了，等我修一下

Zxilly, 2023-09-05:
看一下最新版本

Po810C, 2023-09-06:
最新版本已可正常启动且解决之前所述问题，但使用mumu模拟器任会被检测，通过抓包发现任有含明文ua的包未处理，抓包文件已发至您的邮箱。

Zxilly, 2023-09-06:
抓包时的log可以提供一下吗

Zxilly, 2023-09-06:
如果可以的话，麻烦把抓包时的conntrack信息也给一下，可以用 cat /proc/net/ip_conntrack

Po810C, 2023-09-09:
> 如果可以的话，麻烦把抓包时的conntrack信息也给一下，可以用 cat /proc/net/ip_conntrack 或者 conntrack -L

已将抓包文件发送至您的邮箱，cat /proc/net/ip_conntrack 或 conntrack -L 执行报错，可能是我没有编译进固件

closed 2023-11-17 by github-actions[bot]

Po810C, 2023-11-19:
通过对不同源码不同设备的实验，基本可以确定这些极少数的漏包是因为mtk hnat，大多数情况下ua2f在有mtk hnat时工作正常，最容易复现此问题的情况是启动mumu模拟器，必定会出现漏包并被drcom系统检测。

测试设备：
JCG Q20 MT7621 MIPS
360 T7  MT7981 ARM
软路由         X86

测试源码：
https://github.com/padavanonly/immortalwrt 有mtk hnat
https://github.com/immortalwrt/immortalwrt 无mtk hnat

Po810C, 2023-11-19:
已将抓包文件发送至您的邮箱

Zxilly, 2023-11-19:
硬件加速这种问题UA2F处理不了，程序完全依赖系统的防火墙来进行包的获取的。