Zxilly/UA2F #90: 启用内核 conntrack 支持后反复 crash，同时存在多个进程，且 UA 没有变化

Rachel030219, 2023-08-31:
设备是红米 AX6，感觉可能与 OpenWrt 换用 nftables 有关，安装了 iptables-nft，还启用了一些可能有用的其它特性，但仍然报如下错误：

```
Thu Aug 31 20:58:40 2023 syslog.info UA2F[4924]: Statistics initialized.
Thu Aug 31 20:58:40 2023 syslog.info UA2F[4924]: Handler initialized.
Thu Aug 31 20:58:40 2023 syslog.info UA2F[4924]: Cache lock initialized
Thu Aug 31 20:58:40 2023 syslog.info UA2F[4924]: Cleanup thread created
Thu Aug 31 20:58:50 2023 syslog.warn UA2F[4924]: Kernel does not support conntrack
Thu Aug 31 20:58:50 2023 syslog.err UA2F[4924]: Packet has no conntrack.
```

然后它就自己反复 crash，但有一个 pid 不同的进程一直在后台运行：

```
Thu Aug 31 20:46:05 2023 syslog.info UA2F[4839]: Statistics initialized.
Thu Aug 31 20:46:05 2023 syslog.info UA2F[4839]: Handler initialized.
Thu Aug 31 20:46:05 2023 syslog.info UA2F[4839]: Cache lock initialized
Thu Aug 31 20:46:05 2023 syslog.info UA2F[4839]: Cleanup thread created
……
Thu Aug 31 20:54:39 2023 syslog.info UA2F[4839]: UA2F has handled 16 ua http, 95 tcp. 95 ipv4, 0 ipv6 packets in 8 minutes and 34 seconds.
```

尽管如此，发出的 HTTP 包的 UA 并没有变化，完整的系统日志如下： https://pastebin.com/BaEz8J7R

大部分有变动的内核配置：

[image]

可能相关的已安装软件包：

```
root@RachelRt:~# opkg list
ip6tables-mod-fullconenat - 2023-01-01-74c5e6f3-1
ip6tables-mod-nat - 1.8.8-1
ip6tables-nft - 1.8.8-1
ipq-wifi-redmi_ax6 - 2023-06-03-cd9c30ca-2
iptables-mod-conntrack-extra - 1.8.8-1
iptables-mod-fullconenat - 2023-01-01-74c5e6f3-1
iptables-mod-nflog - 1.8.8-1
iptables-mod-nfqueue - 1.8.8-1
iptables-nft - 1.8.8-1
ipv6helper - 5
kmod-ip6tables - 5.15.127-1
kmod-ipt-conntrack - 5.15.127-1
kmod-ipt-conntrack-extra - 5.15.127-1
kmod-ipt-core - 5.15.127-1
kmod-ipt-fullconenat - 5.15.127+2023-01-01-74c5e6f3-1
kmod-ipt-nat - 5.15.127-1
kmod-ipt-nat6 - 5.15.127-1
kmod-ipt-nflog - 5.15.127-1
kmod-ipt-nfqueue - 5.15.127-1
kmod-iptunnel - 5.15.127-1
kmod-iptunnel4 - 5.15.127-1
kmod-nf-conncount - 5.15.127-1
kmod-nf-conntrack - 5.15.127-1
kmod-nf-conntrack-netlink - 5.15.127-1
kmod-nf-conntrack6 - 5.15.127-1
kmod-nf-flow - 5.15.127-1
kmod-nf-ipt - 5.15.127-1
kmod-nf-ipt6 - 5.15.127-1
kmod-nf-log - 5.15.127-1
kmod-nf-log6 - 5.15.127-1
kmod-nf-nat - 5.15.127-1
kmod-nf-nat6 - 5.15.127-1
kmod-nf-nathelper - 5.15.127-1
kmod-nf-nathelper-extra - 5.15.127-1
kmod-nf-reject - 5.15.127-1
kmod-nf-reject6 - 5.15.127-1
kmod-nfnetlink - 5.15.127-1
kmod-nfnetlink-log - 5.15.127-1
kmod-nfnetlink-queue - 5.15.127-1
kmod-nft-bridge - 5.15.127-1
kmod-nft-compat - 5.15.127-1
kmod-nft-core - 5.15.127-1
kmod-nft-fib - 5.15.127-1
kmod-nft-fullcone - 5.15.127+2023-01-10-95ad79bc-1
kmod-nft-nat - 5.15.127-1
kmod-nft-netdev - 5.15.127-1
kmod-nft-offload - 5.15.127-1
kmod-nft-queue - 5.15.127-1
kmod-qca-nss-dp - 5.15.127+2022-04-30-72e9ec41-1
kmod-qca-ssdk - 5.15.127+2022-09-12-628b22bc-2
libmnl0 - 1.0.5-1
libnatpmp1 - 20150609-3
libnetfilter-conntrack3 - 1.0.9-2
libnetfilter-queue1 - 1.0.5-4
libnettle8 - 3.9.1-1
libnfnetlink0 - 1.0.2-1
libnftnl11 - 1.2.6-1
libxtables12 - 1.8.8-1
miniupnpd-nftables - 2.3.3-2
mtd - 26
netifd - 2023-06-04-ec9dba72-1
nft-qos - 1.0.6-4
nftables-json - 1.0.8-1
qca-ssdk-shell - 2022-09-12-b43fba2d-1
ua2f - 4.1.2-1
xtables-nft - 1.8.8-1
```

完整的已安装软件包列表：  https://pastebin.com/jgVZRxHP

`uci show ua2f` 的结果：

```
root@RachelRt:~# uci show ua2f
ua2f.enabled=ua2f
ua2f.enabled.enabled='1'
ua2f.firewall=ua2f
ua2f.firewall.handle_fw='1'
ua2f.firewall.handle_tls='1'
ua2f.firewall.handle_intranet='1'
ua2f.firewall.handle_mmtls='1'
```

Zxilly, 2023-08-31:
nftables的支持是没有测试过的，如果运行不了最近也没有办法

Zxilly, 2023-08-31:
我加一个conntrack可选的特性吧

Zxilly, 2023-08-31:
看一下 https://github.com/Zxilly/UA2F/commit/e21821c4c0f693285544245a62d9ea5447f3ae08 能不能解决问题，CI 里可以下构建好的 ipk

Rachel030219, 2023-08-31:
坏了，close 早了，好像还是不太对劲，它只在一开始正常工作了一会儿，过了几分钟又会疯狂启动，log 如下：

```
Thu Aug 31 21:51:28 2023 syslog.info UA2F[3672]: Statistics initialized.
Thu Aug 31 21:51:28 2023 syslog.info UA2F[3672]: Handler initialized.
Thu Aug 31 21:51:28 2023 syslog.info UA2F[3672]: Cache lock initialized
Thu Aug 31 21:51:28 2023 syslog.info UA2F[3672]: Cleanup thread created
………
Thu Aug 31 23:16:42 2023 syslog.info UA2F[3672]: UA2F has handled 8 ua http, 58 tcp. 58 ipv4, 0 ipv6 packets in 1 hours, 25 minutes and 14 seconds.
………
Thu Aug 31 23:20:33 2023 syslog.warn UA2F[3672]: Kernel does not support conntrack
Thu Aug 31 23:20:33 2023 syslog.warn UA2F[3672]: Packet has no conntrack. Cache will be disabled. This may cause performance issue.
Thu Aug 31 23:20:33 2023 syslog.info UA2F[3672]: Disabling cache
Thu Aug 31 23:20:33 2023 syslog.info UA2F[3672]: Cleanup thread canceled
Thu Aug 31 23:20:33 2023 syslog.err UA2F[3672]: Unknown ip version: 0
```

似乎它正确处理了几个包（验证也的确如此），然后在此之后就继续疯狂重启，类似于：

<details>
<summary>点我看致死量 log </summary>
<pre><code>
Thu Aug 31 23:20:33 2023 syslog.info UA2F[3672]: Cleanup thread canceled
Thu Aug 31 23:20:33 2023 syslog.err UA2F[3672]: Unknown ip version: 0
Thu Aug 31 23:20:38 2023 syslog.info UA2F[5707]: Statistics initialized.
Thu Aug 31 23:20:38 2023 syslog.info UA2F[5707]: Handler initialized.
Thu Aug 31 23:20:38 2023 syslog.info UA2F[5707]: Cache lock initialized
Thu Aug 31 23:20:38 2023 syslog.info UA2F[5707]: Cleanup thread created
Thu Aug 31 23:20:41 2023 syslog.warn UA2F[5707]: Kernel does not support conntrack
Thu Aug 31 23:20:41 2023 syslog.warn UA2F[5707]: Packet has no conntrack. Cache will be disabled. This may cause performance issue.
Thu Aug 31 23:20:41 2023 syslog.info UA2F[5707]: Disabling cache
Thu Aug 31 23:20:41 2023 syslog.info UA2F[5707]: Cleanup thread canceled
Thu Aug 31 23:20:41 2023 syslog.err UA2F[5707]: Unknown ip version: 0
Thu Aug 31 23:20:46 2023 syslog.info UA2F[5709]: Statistics initialized.
Thu Aug 31 23:20:46 2023 syslog.info UA2F[5709]: Handler initialized.
Thu Aug 31 23:20:46 2023 syslog.info UA2F[5709]: Cache lock initialized
Thu Aug 31 23:20:46 2023 syslog.info UA2F[5709]: Cleanup thread created
Thu Aug 31 23:21:13 2023 syslog.warn UA2F[5709]: Kernel does not support conntrack
Thu Aug 31 23:21:13 2023 syslog.warn UA2F[5709]: Packet has no conntrack. Cache will be disabled. This may cause performance issue.
Thu Aug 31 23:21:13 2023 syslog.info UA2F[5709]: Disabling cache
Thu Aug 31 23:21:13 2023 syslog.info UA2F[5709]: Cleanup thread canceled
Thu Aug 31 23:21:13 2023 syslog.err UA2F[5709]: Unknown ip version: 0
Thu Aug 31 23:21:18 2023 syslog.info UA2F[5712]: Statistics initialized.
Thu Aug 31 23:21:18 2023 syslog.info UA2F[5712]: Handler initialized.
Thu Aug 31 23:21:18 2023 syslog.info UA2F[5712]: Cache lock initialized
Thu Aug 31 23:21:18 2023 syslog.info UA2F[5712]: Cleanup thread created
Thu Aug 31 23:21:44 2023 syslog.warn UA2F[5712]: Kernel does not support conntrack
Thu Aug 31 23:21:44 2023 syslog.warn UA2F[5712]: Packet has no conntrack. Cache will be disabled. This may cause performance issue.
Thu Aug 31 23:21:44 2023 syslog.info UA2F[5712]: Disabling cache
Thu Aug 31 23:21:44 2023 syslog.info UA2F[5712]: Cleanup thread canceled
Thu Aug 31 23:21:44 2023 syslog.err UA2F[5712]: Unknown ip version: 0
Thu Aug 31 23:21:49 2023 syslog.info UA2F[5714]: Statistics initialized.
Thu Aug 31 23:21:49 2023 syslog.info UA2F[5714]: Handler initialized.
Thu Aug 31 23:21:49 2023 syslog.info UA2F[5714]: Cache lock initialized
Thu Aug 31 23:21:49 2023 syslog.info UA2F[5714]: Cleanup thread created
Thu Aug 31 23:22:16 2023 syslog.warn UA2F[5714]: Kernel does not support conntrack
Thu Aug 31 23:22:16 2023 syslog.warn UA2F[5714]: Packet has no conntrack. Cache will be disabled. This may cause performance issue.
Thu Aug 31 23:22:16 2023 syslog.info UA2F[5714]: Disabling cache
Thu Aug 31 23:22:16 2023 syslog.info UA2F[5714]: Cleanup thread canceled
Thu Aug 31 23:22:16 2023 syslog.err UA2F[5714]: Unknown ip version: 0
Thu Aug 31 23:22:21 2023 syslog.info UA2F[5717]: Statistics initialized.
Thu Aug 31 23:22:21 2023 syslog.info UA2F[5717]: Handler initialized.
Thu Aug 31 23:22:21 2023 syslog.info UA2F[5717]: Cache lock initialized
Thu Aug 31 23:22:21 2023 syslog.info UA2F[5717]: Cleanup thread created
</code></pre>
</details>

看起来问题似乎出在 `Thu Aug 31 23:22:16 2023 syslog.err UA2F[5714]: Unknown ip version: 0` ，但我不太确定这个问题的来源，也许仍然是 nftables 的兼容性问题，有什么方法能够定位它吗…？

Zxilly, 2023-08-31:
看起来仍然是因为conntrack无法正常工作，所以ua2f无法识别一个包是ipv4的还是ipv6的

Rachel030219, 2023-09-01:
感谢跟进！这次只有一个 UA2F 了，但是防火墙规则似乎出了点问题：

```
Fri Sep  1 08:27:12 2023 daemon.notice procd: /etc/rc.d/S99ua2f: /dev/stdin:37:13-19: Error: syntax error, unexpected counter
Fri Sep  1 08:27:12 2023 daemon.notice procd: /etc/rc.d/S99ua2f: ct original counter queue num 10010 bypass;
Fri Sep  1 08:27:12 2023 daemon.notice procd: /etc/rc.d/S99ua2f:             ^^^^^^^
Fri Sep  1 08:27:12 2023 syslog.info UA2F[3629]: Statistics initialized.
Fri Sep  1 08:27:12 2023 syslog.info UA2F[3629]: Handler initialized.
Fri Sep  1 08:27:12 2023 syslog.info UA2F[3629]: Cache lock initialized
Fri Sep  1 08:27:12 2023 syslog.info UA2F[3629]: Cleanup thread created
```

在此之后 UA2F 在后台运行，但 `nftables list ruleset` 没有相关规则；我对 nftables 也是一点不熟悉，试了一下用 iptables 手动输入 `/etc/init.d/ua2f` 里面那些 iptables 语句虽然会（看似）正确地在 nftables 中生成 `table ip mangle` 以及相应规则，但实际上没有生效；问了 ChatGPT 后它推荐我把 `ct original counter queue num 10010 bypass;` 改成 `queue num 10010` ，改完重新使用 nftables 导入规则后，防火墙倒是似乎生效了，但是网断没了，而且又开始反复重启：

<details><summary>点我看导入后 <code>nft list ruleset</code> 显示的规则</summary>
<pre><code>
table inet ua2f {
        set localaddr_v4 {
                type ipv4_addr
                flags interval
                auto-merge
                elements = { 0.0.0.0/8, 100.64.0.0/10,
                             127.0.0.0/8, 169.254.0.0/16,
                             172.16.0.0/12, 192.0.0.0/24,
                             192.0.2.0/24, 192.31.196.0/24,
                             192.52.193.0/24, 192.88.99.0/24,
                             192.168.0.0/16, 192.175.48.0/24,
                             198.18.0.0/15, 198.51.100.0/24,
                             203.0.113.0/24, 224.0.0.0/3 }
        }

        set localaddr_v6 {
                type ipv6_addr
                flags interval
                auto-merge
                elements = { ::/127,
                             ::ffff:0.0.0.0/96,
                             64:ff9b::/96,
                             100::/64,
                             2001::/32,
                             2001:10::-2001:2f:ffff:ffff:ffff:ffff:ffff:ffff,
                             2001:db0::/28,
                             2002::/16,
                             fc00::/7,
                             fe80::/10,
                             ff00::/8 }
        }

        chain prerouting {
                type filter hook prerouting priority mangle - 5; policy accept;
                ip daddr @localaddr_v4 counter packets 262 bytes 23302 return
                ip6 daddr @localaddr_v6 counter packets 46 bytes 10800 return
                tcp dport 22 counter packets 0 bytes 0 return comment "!ua2f: bypass SSH"
                tcp dport 80 counter packets 9 bytes 432 ct mark set 0x0000002c
                ct mark 0x0000002b counter packets 1736 bytes 2555110 return comment "!ua2f: bypass non-http stream"
                queue to 10010
        }
}
</code></pre>
</details>

<details><summary>点我看致死量 log</summary>
<pre><code>
Fri Sep  1 09:33:52 2023 syslog.info UA2F[9395]: Statistics initialized.
Fri Sep  1 09:33:52 2023 syslog.info UA2F[9395]: Handler initialized.
Fri Sep  1 09:33:52 2023 syslog.info UA2F[9395]: Cache lock initialized
Fri Sep  1 09:33:52 2023 syslog.info UA2F[9395]: Cleanup thread created
Fri Sep  1 09:33:52 2023 syslog.warn UA2F[9395]: Kernel does not support conntrack
Fri Sep  1 09:33:52 2023 syslog.err UA2F[9395]: Packet has no conntrack.
Fri Sep  1 09:33:57 2023 syslog.info UA2F[9397]: Statistics initialized.
Fri Sep  1 09:33:57 2023 syslog.info UA2F[9397]: Handler initialized.
Fri Sep  1 09:33:57 2023 syslog.info UA2F[9397]: Cache lock initialized
Fri Sep  1 09:33:57 2023 syslog.info UA2F[9397]: Cleanup thread created
Fri Sep  1 09:34:17 2023 syslog.info UA2F[9403]: Statistics initialized.
Fri Sep  1 09:34:17 2023 syslog.info UA2F[9403]: Handler initialized.
Fri Sep  1 09:34:17 2023 syslog.info UA2F[9403]: Cache lock initialized
Fri Sep  1 09:34:17 2023 syslog.info UA2F[9403]: Cleanup thread created
Fri Sep  1 09:34:19 2023 syslog.warn UA2F[9403]: Kernel does not support conntrack
Fri Sep  1 09:34:19 2023 syslog.err UA2F[9403]: Packet has no conntrack.
Fri Sep  1 09:34:24 2023 syslog.info UA2F[9405]: Statistics initialized.
Fri Sep  1 09:34:24 2023 syslog.info UA2F[9405]: Handler initialized.
Fri Sep  1 09:34:24 2023 syslog.info UA2F[9405]: Cache lock initialized
Fri Sep  1 09:34:24 2023 syslog.info UA2F[9405]: Cleanup thread created
Fri Sep  1 09:34:26 2023 syslog.warn UA2F[9405]: Kernel does not support conntrack
Fri Sep  1 09:34:26 2023 syslog.err UA2F[9405]: Packet has no conntrack.
Fri Sep  1 09:34:26 2023 daemon.info procd: Instance ua2f::ua2f s in a crash loop 6 crashes, 1 seconds since last crash
</code></pre>
</details>

头有点大了…

Zxilly, 2023-09-01:
试一下 https://github.com/Zxilly/UA2F/commit/2fb0b6febbf0f433ab561fef61f49c76cabc63c5，我手上没有nftables的设备

Rachel030219, 2023-09-01:
感谢！这个新 commit 编译出来的版本能够正常兼容 nftables 了，只是跑起来依旧反复重启，这次并没有在日志里输出问题，检查了一下发现似乎是 libnetfilter-queue 存在 segmentation fault：

```
root@RachelRt:~# valgrind ua2f
==6114== Memcheck, a memory error detector
==6114== Copyright (C) 2002-2022, and GNU GPL'd, by Julian Seward et al.
==6114== Using Valgrind-3.21.0 and LibVEX; rerun with -h for copyright info
==6114== Command: ua2f
==6114==
==6114== Invalid read of size 8
==6114==    at 0x491A0B0: ??? (in /usr/lib/libnetfilter_queue.so.1.5.0)
==6114==  Address 0x8 is not stack'd, malloc'd or (recently) free'd
==6114==
==6114==
==6114== Process terminating with default action of signal 11 (SIGSEGV)
==6114==  Access not within mapped region at address 0x8
==6114==    at 0x491A0B0: ??? (in /usr/lib/libnetfilter_queue.so.1.5.0)
==6114==  If you believe this happened as a result of a stack
==6114==  overflow in your program's main thread (unlikely but
==6114==  possible), you can try to increase the size of the
==6114==  main thread stack using the --main-stacksize= flag.
==6114==  The main thread stack size used in this run was 8388608.
==6114==
==6114== HEAP SUMMARY:
==6114==     in use at exit: 0 bytes in 0 blocks
==6114==   total heap usage: 0 allocs, 0 frees, 0 bytes allocated
==6114==
==6114== All heap blocks were freed -- no leaks are possible
==6114==
==6114== For lists of detected and suppressed errors, rerun with: -s
==6114== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
Segmentation fault
```

很怪，感觉应该不是 UA2F 的问题…？总之先把这个 issue close 掉，我自己研究研究好了，多谢作者的贡献！

Zxilly, 2023-09-01:
不，如果还是相同的日志的话，还是ua2f读不到conntrack主动关闭的，具体的代码是
https://github.com/Zxilly/UA2F/blob/2fb0b6febbf0f433ab561fef61f49c76cabc63c5/src/handler.c#L167-L170
但是ua2f同时支持v4和v6需要conntrack给的标记，所以暂时没有什么好的解决办法，先开着，看下什么时候能深入看一下

Zxilly, 2023-09-01:
如果是libnetfilter-queue炸了的话。。。我再看下具体的实现

Zxilly, 2023-09-01:
可以的话还是麻烦提供一下日志，我好追一下是断在哪里了

Rachel030219, 2023-09-01:
好的，截取出来的一段日志如下：

```
……
Fri Sep  1 08:27:28 2023 syslog.info UA2F[5463]: Statistics initialized.
Fri Sep  1 08:27:28 2023 syslog.info UA2F[5463]: Handler initialized.
Fri Sep  1 08:27:28 2023 syslog.info UA2F[5463]: Cache lock initialized
Fri Sep  1 08:27:28 2023 syslog.info UA2F[5463]: Cleanup thread created
Fri Sep  1 08:27:30 2023 kern.info kernel: [   32.486328] l11: disabling
Fri Sep  1 17:17:09 2023 syslog.info UA2F[5480]: Statistics initialized.
Fri Sep  1 17:17:09 2023 syslog.info UA2F[5480]: Handler initialized.
Fri Sep  1 17:17:09 2023 syslog.info UA2F[5480]: Cache lock initialized
Fri Sep  1 17:17:09 2023 syslog.info UA2F[5480]: Cleanup thread created
Fri Sep  1 17:17:14 2023 syslog.info UA2F[5480]: UA2F has handled 8 ua http, 74 tcp. 74 ipv4, 0 ipv6 packets in 5 seconds.
Fri Sep  1 17:17:34 2023 syslog.info UA2F[5486]: Statistics initialized.
Fri Sep  1 17:17:34 2023 syslog.info UA2F[5486]: Handler initialized.
Fri Sep  1 17:17:34 2023 syslog.info UA2F[5486]: Cache lock initialized
Fri Sep  1 17:17:34 2023 syslog.info UA2F[5486]: Cleanup thread created
Fri Sep  1 17:17:44 2023 syslog.info UA2F[5526]: Statistics initialized.
Fri Sep  1 17:17:44 2023 syslog.info UA2F[5526]: Handler initialized.
Fri Sep  1 17:17:44 2023 syslog.info UA2F[5526]: Cache lock initialized
Fri Sep  1 17:17:44 2023 syslog.info UA2F[5526]: Cleanup thread created
Fri Sep  1 17:17:53 2023 daemon.notice hostapd: phy0-ap0: DFS-CAC-COMPLETED success=1 freq=5180 ht_enabled=0 chan_offset=0 chan_width=5 cf1=5250 cf2=0
Fri Sep  1 17:17:53 2023 daemon.warn hostapd: Can't set DFS state for freq 5180 MHz
Fri Sep  1 17:17:53 2023 daemon.warn hostapd: Can't set DFS state for freq 5200 MHz
Fri Sep  1 17:17:53 2023 daemon.warn hostapd: Can't set DFS state for freq 5220 MHz
Fri Sep  1 17:17:53 2023 daemon.warn hostapd: Can't set DFS state for freq 5240 MHz
Fri Sep  1 17:17:53 2023 kern.info kernel: [   79.953671] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
Fri Sep  1 17:17:53 2023 kern.info kernel: [   79.953935] br-lan: port 4(phy0-ap0) entered blocking state
Fri Sep  1 17:17:53 2023 kern.info kernel: [   79.959143] br-lan: port 4(phy0-ap0) entered forwarding state
Fri Sep  1 17:17:53 2023 daemon.notice netifd: Network device 'phy0-ap0' link is up
Fri Sep  1 17:17:53 2023 daemon.notice hostapd: phy0-ap0: interface state DFS->ENABLED
Fri Sep  1 17:17:53 2023 daemon.notice hostapd: phy0-ap0: AP-ENABLED
Fri Sep  1 17:18:06 2023 syslog.info UA2F[5545]: Statistics initialized.
Fri Sep  1 17:18:06 2023 syslog.info UA2F[5545]: Handler initialized.
Fri Sep  1 17:18:06 2023 syslog.info UA2F[5545]: Cache lock initialized
Fri Sep  1 17:18:06 2023 syslog.info UA2F[5545]: Cleanup thread created
Fri Sep  1 17:18:11 2023 daemon.info procd: Instance ua2f::ua2f s in a crash loop 6 crashes, 4 seconds since last crash
……
```

并没有出现类似于 `Packet has no conntrack.` 的内容，所以我才会感觉不是 UA2F 的问题…

Zxilly, 2023-09-01:
因为有个别的bug，nftables把非tcp的包放行了，我加了条规则

Rachel030219, 2023-09-01:
感谢！现在正常工作了，跑了十几分钟没有问题，UA 也能正常被修改，这个 issue 应该可以 close 掉了

另外为以后找到这个 issue 的人指一下，如果 UA2F 正常工作但是 UA 还是没有变动的话，可以检查一下 `网络 -> 防火墙` 里的 `流量分载` ，我检查了好久才想起来这个 offloading 会影响本机对包的处理，必须全部关掉才行

closed 2023-09-01 by Rachel030219