Zxilly/UA2F #92: 关于使用Github Actions编译包含UA2F固件的报错

yuos-bit, 2023-09-26:
大佬您好，请问您这边的CMAKE编译环境是怎么样的呢？我这边使用Github Actions时，按照原生配置进行编译时，报错：
```shell
C_STANDARD is set to invalid value '17' 
```
[image]

我在查阅网上资料后加入
```shell
# 确定cmake最低版本
# sed -i 's/set(CMAKE_C_STANDARD 17)/set(CMAKE_C_STANDARD 11)/g' package/yuos/UA2F/CMakeLists.txt
sed -i '4a set(CMAKE_C_STANDARD_REQUIRED ON)' package/yuos/UA2F/CMakeLists.txt
sed -i '5a SET_TARGET_PROPERTIES(target PROPERTIES name 17)' package/yuos/UA2F/CMakeLists.txt
```
然后提示：
```shell
CMake Error at CMakeLists.txt:6 (set_target_properties):
  set_target_properties called with illegal arguments, maybe missing a
  PROPERTIES specifier?
```
我这边查询相应资料无果，特向大佬请教一二。
以下是我的openwrt 21.02 配置文件以及Github Actions地址
配置文件：https://github.com/yuos-bit/AutoBuild-OpenWrt/blob/main/ruijie-1.sh
Github Actions地址:https://github.com/yuos-bit/AutoBuild-OpenWrt/blob/main/.github/workflows/Build_Openwrt-ruijie.yml
GCC以及CMAKE版本已更新
[image]

Zxilly, 2023-09-26:
你改的东西太多了，我不确定问题在哪

yuos-bit, 2023-09-26:
感谢大佬这么晚还回复，如题，我这边最开始就是遇到了c 17这个报错，只需要解决这个报错就可以了。

后续的一些操作都是为了解决这个报错而魔改的。

yuos-bit, 2023-09-26:
> 你改的东西太多了，我不确定问题在哪

[image]
```shell
-- Using version string: 4.3.3
-- AddressSanitizer is disabled.
-- Using custom user agent string: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/112.0.0.0 Safari/537.36 Edg/112.0.1722.68
-- Tests are disabled.
-- Configuring done
CMake Error at CMakeLists.txt:72 (add_executable):
  C_STANDARD is set to invalid value '17'
```
这是在不改动任何配置，编译openwrt 21.02的报错
参阅：https://github.com/yuos-bit/AutoBuild-OpenWrt/actions/runs/6294950021/job/17087524921
[image]

yuos-bit, 2023-09-26:
> 建议你什么都别改，默认的cmake版本够了
> 
> https://github.com/actions/runner-images/blob/main/images/linux/Ubuntu2204-Readme.md
> 
> `CMake 3.27.5`

好的，感谢大佬，我这边默认的编译环境是`ubuntu-20.04`，我试试迭代到`ubuntu-22.04`
https://github.com/actions/runner-images/blob/main/images/linux/Ubuntu2204-Readme.md

yuos-bit, 2023-09-27:
> 建议你什么都别改，默认的cmake版本够了
> 
> https://github.com/actions/runner-images/blob/main/images/linux/Ubuntu2204-Readme.md
> 
> `CMake 3.27.5`
大佬您好，通过测试，默认的编译环境迭代到`ubuntu-22.04`，报错依旧
而我查阅论坛后将
```shell
set(CMAKE_C_STANDARD 17)
```
更改为
```shell
set(CMAKE_C_STANDARD 11)
```
之后，问题解决。
编译成功案例如下：
https://github.com/yuos-bit/AutoBuild-OpenWrt/actions/runs/6321797320

bash脚本内容如下：
```shell
## 修复 C_STANDARD is set to invalid value '17' 报错
## 参阅：https://cmake.org/cmake/help/v3.27/prop_tgt/CXX_STANDARD.html
# sed -i 's/VERSION 3.16/VERSION 3.27.6/g' package/yuos/UA2F/CMakeLists.txt
# 确定cmake最低版本
sed -i 's/set(CMAKE_C_STANDARD 17)/set(CMAKE_C_STANDARD 11)/g' package/yuos/UA2F/CMakeLists.txt
# sed -i '4a set(CMAKE_C_STANDARD_REQUIRED ON)' package/yuos/UA2F/CMakeLists.txt
# sed -i '5a SET_TARGET_PROPERTIES(target PROPERTIES name 17)' package/yuos/UA2F/CMakeLists.txt
```
供后来人参考

云编译成功的包如下：
[image]

Zxilly, 2023-09-27:
我不保证在C11下编译通过