Zxilly/UA2F #99: 4.1.2之后的版本似乎在 红米ac2100  23.05  上不起作用，最新版本4.3.3重新编译固件也一样。

PC12138-Z, 2023-10-16:
[image]
一个多月前最新版本是4.1.2时编译的就能用，在此基础上升级到4.3.3也不起作用，甚至尝试升级到4.2.2也不起作用
[image]

Zxilly, 2023-10-16:
开下coredump，然后把生成的文件给我

PC12138-Z, 2023-10-17:
> 然后运行一下 `nft list ruleset`，把返回值发一下

找原因了， nft 没起作用 ，把东西补全就好了，直接升级，跟重新编译也能起作用了，感谢提供思路

Zxilly, 2023-10-17:
建议说一下改了什么，可以给别人参考

PC12138-Z, 2023-10-17:
> 建议说一下改了什么，可以给别人参考

最简单就是直接安装firewall4，这个能用了，nft基本没问题

PC12138-Z, 2023-10-17:
> 建议说一下改了什么，可以给别人参考

最简单就是直接安装firewall4，这个能用了，nft基本没问

Zxilly, 2023-10-17:
但是如果没有firewall4的话应该会自动回退到iptables的，不应该出问题

YeWenxuan64, 2023-10-17:
我这里也复现并解决了，4.3.3的ua2f，在云编译时没加多少自带软件包，所以编译出来用不了，然后小女子一气之下装上firewall4和全套iptables开头的软件包就能用了

YeWenxuan64, 2023-10-17:
尤其iptables-nft

Zxilly, 2023-10-17:
firewall4 应该是默认安装的吧，有点奇怪了

Zxilly, 2023-10-17:
你们运行 `/etc/init.d/ua2f start` 会报错吗，在手动安装 `firewall4` 之前？

YeWenxuan64, 2023-10-17:
云编译嘛，小女子不才，没选上firewall4。运行我这里（newwifi3-d2）假如没有firewall4时运行ua2f start只会报告statistics initialized，handler initialized，Cache lock initialized，Cleanup thread created，然后就不处理流量，没返回“已处理多少条tcp”啥的，去ua检测网站也是没改ua。安装firewall4后就能用了。（补充一点，有关于广告屏蔽插件与ua2f的兼容性测试报告吗，这点我这在初期部署openwrt时貌似有问题）

YeWenxuan64, 2023-10-17:
?神奇，我这里是先安装firewall4然后ua2f就起作用了

Zxilly, 2023-10-17:
4.3.4 已经添加修复

closed 2023-10-17 by Zxilly