Zxilly/UA2F #100: 这是一个非法的Issues（doge）。ios上的appstore貌似会绕过UA2F

YeWenxuan64, 2023-10-17:
喵喵喵~(*/ω＼*)，在被拉黑之前，老娘又来提问题了ヾ(≧▽≦*)o
我的富哥舍友在打开他的苹果全家桶的appstore时貌似UA2F就不行了。
出现过四次，两次偶然，两次手动实验的。
嘤嘤嘤，小女子学识不高，还不知道怎么部署抓包的东西，没能提供数据，抱歉啦`(*>﹏<*)′

Zxilly, 2023-10-18:
缺乏必要的数据，无法修复

Zxilly, 2023-10-18:
苹果在主要的服务上都使用了ssl，可能是其他的dpi策略做了检测，ua2f无法处理这种情况

YeWenxuan64, 2023-10-18:
弱弱地问一下，小女子还有一台x86设备（工控机，一个lan，一个wlan），怎样才能部署抓包加装了ua2f的路由器呢

YeWenxuan64, 2023-10-29:
经过一个多星期的测试，证明只能在自编译的固件稳定实现ua2f，关键点在于以下内核设置（吧）（因为在云编译弄不了这个）

Networking support -> Networking options -> Network packet filtering framework (Netfilter)（先选中再进去） -> Core Netfilter Configuration -> 选中：
•	Netfilter NFNETLINK interface
•	Netfilter LOG over NFNETLINK interface
•	Netfilter connection tracking support
•	Connection tracking netlink interface
•	NFQUEUE and NFLOG integration with Connection Tracking （打开）

也就是在readme所提到的   "由于新加入的 CONNMARK 影响，编译内核时需要添加 NETFILTER_NETLINK_GLUE_CT flag"
(吧，应该是，小女子不懂）

（对于我的学校的网）同时我的另一位同学（他虽然会点计算机，但不会弄本地编译）是用了云编译的带ua2f的固件，跟我的区别是没弄那个内核设置，然后他天天封，但发现开着passwall飞到香港才能缓解（代理，加密嘛）

综上所述：请把那个 "由于新加入的 CONNMARK 影响，编译内核时需要添加 NETFILTER_NETLINK_GLUE_CT flag" 弄显眼一点，或多写一点我上面那个的

☆*: .｡. o(≧▽≦)o .｡.:*☆

Zxilly, 2023-10-29:
https://github.com/Zxilly/UA2F/blob/659e17f888583b22364e8de6d9520ec0b88fe8ef/src/handler.c#L171-L178

发现不支持的时候应该会自动降级到低性能方案而不是出错，有看到这条提示吗

lfznnzq, 2023-11-26:
> 经过一个多星期的测试，证明只能在自编译的加固稳定实现ua2f，关键点位于以下内核设置（吧）（因为在云编译弄不了这个）
> 
> 网络支持 -> 网络选项 -> 网络数据包过滤框架 (Netfilter)（先选中再进去） -> 核心 Netfilter 配置 -> 选中： • Netfilter NFNETLINK 接口 • Netfilter LOG over NFNETLINK 接口 • Netfilter 连接跟踪支持 • 连接跟踪 netlink 接口 • NFQUEUE 和 NFLOG 与连接跟踪集成（打开）
> 
> readme中提到的“由于新加入的CONNMARK影响，编译内核时需要添加NETFILTER_NETLINK_GLUE_CT标志” （吧，应该是，小女子不懂）
> 
> （对于我学校的网）同时我的另一个同学（他虽然会点，但不会弄本地编译）是用云编译的计算机带ua2f的固件，我的区别是没弄那个内核设置，然后他天天封，但发现开着passwall飞到香港才干（代理，加密嘛）
> 
> 综上说：请把那个“由于新加入的CONNMARK影响，编译编译时需要添加NETFILTER_NETLINK_GLUE_CT标志”弄显眼一点，婴儿写一点我上面那个的
> 
> ☆*：.｡。o(≧▽≦)o.｡.:*☆

不要云编译。
直接使用openwrt官方固件，建议用新版本（现在是23.05.x），直接安装ua2f。
然后配置一下。
`uci set ua2f.enabled.enabled=1
uci set ua2f.firewall.handle_fw=1
uci set ua2f.firewall.handle_tls=1
uci set ua2f.firewall.handle_intranet=1
uci set ua2f.main.custom_ua="Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/112.0.0.0 Safari/537.36 Edg/112.0.1722.68"
uci commit ua2f
service ua2f enable
service ua2f start
`

YeWenxuan64, 2024-01-01:
发现这确实是正常的，因为是加密协议，改不了。当然，反代理检测也检测不了，所以无害。

closed 2024-01-01 by YeWenxuan64