Zxilly/UA2F #101: j1900编译openwrt无法开启ua2f

forgotfish, 2023-10-20:
遇到一个非常奇怪的问题，编译成功，ua2f初始化成功，但是“service ua2f start”无反应，网页端查询ua2f已启动,本地终端查询service反馈为“/etc/init.d/ua2f enabled stopped”内核日志无报错，系统日志查询“ua2f”为无
[image]
[image]
[image]
[image]

Zxilly, 2023-10-20:
系统日志你应该发最后那段而不是系统还在初始化的日志

Zxilly, 2023-10-20:
`service ua2f start` 本来就没有回显

forgotfish, 2023-10-20:
确实没有生效,同样的固件在上一台x86_64设备上生效了（上一台烧坏了，才换的这台）

[image]

这张是系统日志的最后一段
[image]

Zxilly, 2023-10-20:
`command -v nft`看下输出

forgotfish, 2023-10-20:
[image]
没有反馈...

Zxilly, 2023-10-20:
那 `cat /etc/config/ua2f`

forgotfish, 2023-10-20:
[image]
如图

Zxilly, 2023-10-20:
你没启用 ua2f
```
uci set ua2f.enabled.enabled=1
uci commit
serivce ua2f restart
```

forgotfish, 2023-10-20:
[image]
好奇怪...

Zxilly, 2023-10-20:
service 打错了，你自己改一下

closed 2023-10-20 by Zxilly

forgotfish, 2023-10-20:
哦哦哦，才看到，service

forgotfish, 2023-10-20:
感谢感谢