Zxilly/UA2F #103: Unexpected connmark value: 258, Maybe other program has changed connmark?

jiuyan660, 2023-10-27:
最新版UA2F系统日志一直刷新这个

Fri Oct 27 22:43:55 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 298, Maybe other program has changed connmark?
Fri Oct 27 22:43:55 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 298, Maybe other program has changed connmark?
Fri Oct 27 22:43:55 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 259, Maybe other program has changed connmark?
Fri Oct 27 22:43:55 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 259, Maybe other program has changed connmark?
Fri Oct 27 22:43:55 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 259, Maybe other program has changed connmark?
Fri Oct 27 22:43:55 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 257, Maybe other program has changed connmark?
Fri Oct 27 22:43:55 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 257, Maybe other program has changed connmark?
Fri Oct 27 22:43:55 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 260, Maybe other program has changed connmark?
Fri Oct 27 22:43:55 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 260, Maybe other program has changed connmark?
Fri Oct 27 22:43:55 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 260, Maybe other program has changed connmark?
Fri Oct 27 22:43:55 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 258, Maybe other program has changed connmark?
Fri Oct 27 22:43:55 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 259, Maybe other program has changed connmark?
Fri Oct 27 22:43:55 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 259, Maybe other program has changed connmark?
Fri Oct 27 22:43:55 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 259, Maybe other program has changed connmark?
Fri Oct 27 22:43:55 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 257, Maybe other program has changed connmark?
Fri Oct 27 22:43:55 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 257, Maybe other program has changed connmark?
Fri Oct 27 22:43:55 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 258, Maybe other program has changed connmark?
Fri Oct 27 22:43:55 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 259, Maybe other program has changed connmark?
Fri Oct 27 22:43:55 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 259, Maybe other program has changed connmark?
Fri Oct 27 22:43:55 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 261, Maybe other program has changed connmark?
Fri Oct 27 22:43:55 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 260, Maybe other program has changed connmark?
Fri Oct 27 22:43:55 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 260, Maybe other program has changed connmark?
Fri Oct 27 22:43:55 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 261, Maybe other program has changed connmark?
Fri Oct 27 22:43:55 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 261, Maybe other program has changed connmark?
Fri Oct 27 22:43:55 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 273, Maybe other program has changed connmark?
Fri Oct 27 22:43:56 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 265, Maybe other program has changed connmark?
Fri Oct 27 22:43:56 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 265, Maybe other program has changed connmark?
Fri Oct 27 22:43:56 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 265, Maybe other program has changed connmark?
Fri Oct 27 22:43:56 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 266, Maybe other program has changed connmark?
Fri Oct 27 22:43:56 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 266, Maybe other program has changed connmark?
Fri Oct 27 22:43:56 2023 syslog.warn UA2F[8944]: Unexpected connmark value: 296, Maybe other program has changed connmark?

Zxilly, 2023-10-27:
mwan3, v2ray, clash之类的应用可能会修改这个，自己排查吧

jiuyan660, 2023-10-27:
Kernel does not support packet timestamps
还有这个问题，不知道是什么原因，编译的时候已经添加NETFILTER_NETLINK_GLUE_CT flag

Zxilly, 2023-10-27:
不要紧，就提示下，没用到

closed 2024-01-03 by github-actions[bot]