Zxilly/UA2F #105: ua2f s in a crash loop

Lc113, 2023-11-04:
ua2f无法启动，系统日志出现：
Sat Nov  4 02:54:09 2023 daemon.info procd: Instance ua2f::ua2f s in a crash loop 6 crashes, 0 seconds since last crash

Lc113, 2023-11-04:
安装的包：
root@OpenWrt:~# opkg list-installed
arptables-nft - 1.8.8-1
base-files - 1545-r24283-280d9dd758
bash - 5.2.15-1
busybox - 1.36.1-1
ca-bundle - 20230311-1
ca-certificates - 20230311-1
cgi-io - 2022-08-10-901b0f04-21
coreutils - 9.3-1
coreutils-nohup - 9.3-1
curl - 8.4.0-2
dnsmasq-full - 2.89-6
dropbear - 2022.82-5
ebtables-nft - 1.8.8-1
firewall4 - 2023-09-01-598d9fbb-1
fstools - 2023-02-28-bfe882d5-1
fwtool - 2019-11-12-8f7fe925-1
getrandom - 2022-08-13-4c7b720b-2
hostapd-common - 2023-09-08-e5ccbfc6-5
ip-full - 6.5.0-1
ipset - 7.17-1
iptables-mod-conntrack-extra - 1.8.8-1
iptables-mod-ipopt - 1.8.8-1
iptables-nft - 1.8.8-1
iw - 5.19-1
iwinfo - 2023-07-01-ca79f641-1
jansson4 - 2.14-3
jshn - 2023-05-23-75a3b870-1
jsonfilter - 2018-02-04-c7e938d6-1
kernel - 5.15.137-1-1e96c0adc826e7e9d7d8c1c19b5a5ebc
kmod-arptables - 5.15.137-1
kmod-cfg80211 - 5.15.137+6.5-1
kmod-crypto-aead - 5.15.137-1
kmod-crypto-ccm - 5.15.137-1
kmod-crypto-cmac - 5.15.137-1
kmod-crypto-crc32c - 5.15.137-1
kmod-crypto-ctr - 5.15.137-1
kmod-crypto-gcm - 5.15.137-1
kmod-crypto-gf128 - 5.15.137-1
kmod-crypto-ghash - 5.15.137-1
kmod-crypto-hash - 5.15.137-1
kmod-crypto-hmac - 5.15.137-1
kmod-crypto-manager - 5.15.137-1
kmod-crypto-null - 5.15.137-1
kmod-crypto-rng - 5.15.137-1
kmod-crypto-seqiv - 5.15.137-1
kmod-crypto-sha512 - 5.15.137-1
kmod-ebtables - 5.15.137-1
kmod-gpio-button-hotplug - 5.15.137-3
kmod-ip6tables - 5.15.137-1
kmod-ipt-conntrack - 5.15.137-1
kmod-ipt-conntrack-extra - 5.15.137-1
kmod-ipt-core - 5.15.137-1
kmod-ipt-ipopt - 5.15.137-1
kmod-ipt-ipset - 5.15.137-1
kmod-leds-gpio - 5.15.137-1
kmod-lib-crc-ccitt - 5.15.137-1
kmod-lib-crc32c - 5.15.137-1
kmod-mac80211 - 5.15.137+6.5-1
kmod-mt76-core - 5.15.137+2023-09-18-2afc7285-1
kmod-mt7603 - 5.15.137+2023-09-18-2afc7285-1
kmod-nf-conncount - 5.15.137-1
kmod-nf-conntrack - 5.15.137-1
kmod-nf-conntrack-netlink - 5.15.137-1
kmod-nf-conntrack6 - 5.15.137-1
kmod-nf-flow - 5.15.137-1
kmod-nf-ipt - 5.15.137-1
kmod-nf-ipt6 - 5.15.137-1
kmod-nf-log - 5.15.137-1
[... 123 lines cut ...]
ubox - 2022-08-13-4c7b720b-2
ubus - 2023-06-05-f787c97b-1
ubusd - 2023-06-05-f787c97b-1
uci - 2023-08-10-5781664d-1
uclient-fetch - 2023-04-13-007d9454-1
ucode - 2023-10-19-07c03173-1
ucode-mod-fs - 2023-10-19-07c03173-1
ucode-mod-html - 1
ucode-mod-lua - 1
ucode-mod-math - 2023-10-19-07c03173-1
ucode-mod-nl80211 - 2023-10-19-07c03173-1
ucode-mod-rtnl - 2023-10-19-07c03173-1
ucode-mod-ubus - 2023-10-19-07c03173-1
ucode-mod-uci - 2023-10-19-07c03173-1
ucode-mod-uloop - 2023-10-19-07c03173-1
uhttpd - 2023-06-25-34a8a74d-1
uhttpd-mod-ubus - 2023-06-25-34a8a74d-1
unzip - 6.0-8
urandom-seed - 3
urngd - 2023-07-25-7aefb47b-1
usign - 2020-05-23-f1f65026-1
wireless-regdb - 2023.09.01-1
wpad-basic-mbedtls - 2023-09-08-e5ccbfc6-5
xtables-nft - 1.8.8-1
zlib - 1.3-1

Lc113, 2023-11-04:
Openwrt23.05，小米路由器4C

Zxilly, 2023-11-04:
`ulimit -c unlimited` 开启 coredump, 崩溃发生后上传 /tmp 目录下的 core 文件

Lc113, 2023-11-04:
[image]
[image]
/tmp下无core文件

Zxilly, 2023-11-04:
`cat /proc/sys/kernel/core_pattern` 看一下

Zxilly, 2023-11-04:
然后打印一下配置文件 `cat /etc/config/ua2f`

Lc113, 2023-11-04:
[image]

Zxilly, 2023-11-04:
`ua2f --version` 执行正常吗，这是在判断配置读取是不是有问题

Lc113, 2023-11-04:
确实
root@OpenWrt:~# ua2f --version
/usr/bin/ua2f: line 1: syntax error: unexpected "("

Zxilly, 2023-11-04:
`ldd /usr/bin/ua2f` 看看，感觉你的这个文件根本不是可执行文件
[image]

Lc113, 2023-11-04:
[image]
惹啊，的确不是

closed 2024-01-11 by github-actions[bot]