Zxilly/UA2F #118: ua2f启动，但网站ua不变的情况

zhuan-ye, 2024-03-04:
Mon Mar 4 22:43:23 2024 syslog.info UA2F[4078]: Statistics initialized.
Mon Mar 4 22:43:23 2024 syslog.info UA2F[4078]: Custom user agent string not set, using default F-string.
Mon Mar 4 22:43:23 2024 syslog.info UA2F[4078]: Handler initialized
日志内容

turbo acc和NAT均关闭

Zxilly, 2024-03-04:
提供的信息均正常，需要更多细节来判断具体错误

zhuan-ye, 2024-03-05:
luci-app-advancedplus
luci-app-argon-config
luci-app-autoreboot
luci-app-cpufreq
luci-app-fan
luci-app-firewall
luci-app-mwan3
luci-app-opkg
luci-app-passwall
luci-app-ttydover the web	
luci-app-turboaccShortcut-FE	
luci-app-ua2f
luci-app-upnp
luci-app-wizard
luci-i18n-ttyd-zh-cn
这是插件页，有相冲突的吗

Zxilly, 2024-03-05:
turboacc mwan3冲突

zhuan-ye, 2024-03-05:
删除这些插件了，但是ua不变，有可能是拨号上网的问题吗

zhuan-ye, 2024-03-05:
用的是openwrt-ai的默认包，软件包里下载的ua2f

Zxilly, 2024-03-05:
没有看到有意义的信息，问问发行版的维护者吧🤷‍♂

Zxilly, 2024-03-05:
还有没有啥hwnat/nss啥的，也找找

closed 2024-03-05 by zhuan-ye