Zxilly/UA2F #132: 设置完之后，UA检测网站进不去

2856845160, 2024-07-20:
开发者您好，我是一个小白。非常白的那种。
问题背景：
一、我自己使用23分支编译了一个op固件，在其中加入了ipid和ua2f模块。
二、刷入后，目前仅按照您文档设置了ua2f。除此之外并未设置任何防火墙以及ipid等模块规则。
三、目前使用的是wifi连接手机热点，再通过lan口给电脑供网。
问题：现在https://ua.233996.xyz/这个UA检测网站进不去了。但如果开代理是能打开的。
我试过使用service ua2f stop命令后，依然还是进不去。
这可能是什么原因呢？需要如何配置？还是说
[image]
[image]

Zxilly, 2024-07-20:
logread | grep UA2F 看看，别的http网站能打开吗？比如 http://httpforever.com/？

2856845160, 2024-07-20:
> `logread | grep UA2F` 看看，别的http网站能打开吗？比如 http://httpforever.com/ ？

真滴很感谢回复，刚才试过啦，能打开(不挂代理的情况下)

Zxilly, 2024-07-20:
那你试试不用浏览器，直接 curl http://ua.233996.xyz/ 呢？

Zxilly, 2024-07-20:
https://www.boce.com/m/http/ua.233996.xyz
测了一下，这个网站直连也就打不开，和开不开UA2F没啥关系

2856845160, 2024-07-20:
> https://www.boce.com/m/http/ua.233996.xyz 测了一下，这个网站直连就打不开，和开不开UA2F没啥关系

嗯嗯，刚才试了一下curl http//ua.233996.xyz/，也不行，无论开关UA2F都不行。提示curl: (6) Could not resolve host: http

2856845160, 2024-07-20:
> https://www.boce.com/m/http/ua.233996.xyz 测了一下，这个网站直连就打不开，和开不开UA2F没啥关系

想再麻烦问一下您，这种情况可能是什么原因呢(为什么不连接路由器网络，换为其他网络，例如手机热点，即可直接访问和curl)，如果不管这个的话，如何检测UA2F是否配置好了呢？

Zxilly, 2024-07-20:
你出的错是因为你少打了一个分号，是 `curl http://ua.233996.xyz/`

我重新测了一下看起来是正常的
https://www.boce.com/http/ua.233996.xyz

你用 logread | grep UA2F 看下日志有没有报错。

2856845160, 2024-07-20:
> 你出的错是因为你少打了一个分号，是 `curl http://ua.233996.xyz/`
> 
> 我重新测了一下看起来是正常的 https://www.boce.com/http/ua.233996.xyz
> 
> 你用 `logread | grep UA2F` 看下日志有没有报错。

哇，非常感谢您的提醒！
刚才又试了一下，突然通过网页能打开了，并且curl也可以了。
但是，具体显示并没有预期中那样显示"真实UA"为FFFFFFFFFFFFF。
日志似乎也没有报错。
[image]
[image]

2856845160, 2024-07-20:
> 你出的错是因为你少打了一个分号，是 `curl http://ua.233996.xyz/`
> 
> 我重新测了一下看起来是正常的 https://www.boce.com/http/ua.233996.xyz
> 
> 你用 `logread | grep UA2F` 看下日志有没有报错。

刚才我用您文档中提到的# 使用自定义 User-Agent
uci set ua2f.main.custom_ua="Test UA/1.0"
网页是能进去的，但真实UA那一行也是不显示所设置的这个test UA

如果是路由器其他方面配置出了问题，您能给俺大概说一下是哪方面吗？我先去了解学习一下，就不一直打扰您了。
[image]

Zxilly, 2024-07-20:
[image]

这个服务已经不好使了，加了hsts头会强制跳转到https，失去测试的意义了，我自己弄一个吧。

2856845160, 2024-07-20:
> ![image](https://private-user-images.githubusercontent.com/31370133/350700848-a4182caf-2503-442d-a102-77609a8da3b0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

嗯嗯，俺也照着试了一下，输出是一样的。期待您的新成果！

Zxilly, 2024-07-20:
试试 http://ua-check.stagoh.com/

2856845160, 2024-07-21:
> 试试 http://ua-check.stagoh.com/

非常感谢开发者的新测试网站，刚才试了一下，是可以的。
在op后台stop和start UA2F后是能检测到的。
就是不知道没有显示“Test UA/1.0”是否是正常的？
这个没法显示test ua 是不是指的就是您文章中提到的

**UA2F 不会修改包的大小，因此即使自定义了 User-Agent， 运行时实际的 User-Agent 会是一个从 custom ua 中截取的长度与原始 User-Agent 相同的子串，长度不足时会在末尾补空格。**

[image]

Zxilly, 2024-07-21:
配置优先级有点小问题，修一下

Zxilly, 2024-07-21:
https://github.com/Zxilly/UA2F/releases/tag/v4.7.1

closed 2024-07-21 by Zxilly

hutaodog, 2025-03-20:
> https://github.com/Zxilly/UA2F/releases/tag/v4.7.1

为什么现在这个网址进不去了呢，进去就显示502

hutaodog, 2025-03-20:
> 试试 http://ua-check.stagoh.com/

为什么现在这个网址进不去了呢，进去就显示502

Zxilly, 2025-03-20:
@hutaodog 修好了

hutaodog, 2025-03-20:
> [@hutaodog](https://github.com/hutaodog) 修好了

可以正常访问了，太效率了，感谢你