Zxilly/UA2F #133: 视频有时加载不出来

lzd7707, 2024-08-07:
设备是ac2100，图片加载正常，视频有时会加载不出来，关闭ua2f就正常了。因为没在学校，是在家里测试的，会不会和这个有关？

lzd7707, 2024-08-07:
> 设备是ac2100，图片加载正常，视频有时会加载不出来，关闭ua2f就正常了。因为没在学校，是在家里测试的

Zxilly, 2024-08-07:
缺乏必要信息，无法复现

lzd7707, 2024-08-07:
> 缺乏必要信息，无法复现

挂梯子就好了，应该是ua的问题，可是现在我想改原ua改不了，日志如下Wed Aug  7 16:44:36 2024 syslog.info UA2F[6482]: Statistics initialized.
Wed Aug  7 16:44:36 2024 syslog.info UA2F[6482]: Using config user agent string: Mozilla/5.0 (Linux; Android 13; M2012K11AC Build/TKQ1.220829.002) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/104.0.5112.97 Mobile Safari/537.36
Wed Aug  7 16:44:36 2024 syslog.info UA2F[6482]: Using embed user agent string: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/112.0.0.0 Safari/537.36 Edg/112.0.1722.68

Zxilly, 2024-08-07:
显示错误，最新版已经修了

lzd7707, 2024-08-07:
> 显示错误，最新版已经修了

哦，好的谢谢

lzd7707, 2024-08-07:
> 显示错误，最新版已经修了

改了ua 现在b站的视频能正常看了，微信的视频还和之前一样，挂全局梯子才能正常看🤔

Zxilly, 2024-08-07:
那就是微信做了额外的检测，这个没办法的

lzd7707, 2024-08-07:
> 那就是微信做了额外的检测，这个没办法的
好吧，谢谢

closed 2024-08-07 by Zxilly