Zxilly/UA2F #134: Talk about your views on “proxy-det campus broadband” and “anti proxy-det”

YeWenxuan64, 2024-08-20:
Sorry, I couldn't find your contact information.
Dear Zxilly,
  In the last chapter of my video Facing the proxy-det of campus networks. My solution is…
As one of those who have stood up and fought with practical actions before, I want to invite you to talk about your views on “proxy-det campus broadband” and “anti proxy-det”.

  At first I encountered “proxy-det net” at the beginning of my college lives in 2023, and as a “full stack engineer” proficient in electrical, computer science, music, animation, and photography, I believe I can find solutions. After several days of collecting for information, I discovered your blog and code. I never expected that senpai had already fought for it for years. In today's world where dialectical perspectives and the spirit of sharing have become rare treasures. Thanks to your spirit, after fully mastering the relevant knowledge and skills, I started making a video to briefly describe the relevant knowledge and viewpoints. During this year, I have learned a lot, a lot…
  
  Due to the current chaotic online environment, my "training set" has been contaminated, and I am not able to complete the summary of the final chapter of the video in an absolutely fair and objective manner. Therefore, I need the help of senpai to share your thoughts as a reference. Can you talk about your views on “proxy-det campus broadband” and “anti proxy-det” please?
  
  I accept any form of expression, whether it's text, voice, or video. Let's convey meaningful knowledge and correct viewpoints to “future generations” at the end of the video! Looking forward to your early reply and I will provide more information about my work.

Yours,
Wenxuan Ye

Zxilly, 2024-08-20:
I meet the problem, I solve the problem, I don't see what to discuss.
Also, not a good idea to send message like this to issue, this may confuse anyone watch this repo.

closed 2024-08-20 by Zxilly

YeWenxuan64, 2024-08-20:
I offer my sincerest apologies to you.

I feel ashamed that I am unable to find any your contact information like e-mail or telegram.
Please don't be angry about my mistake.
Hope you will have a nice day.