Zxilly/UA2F #136: 报错 Signal Segmentation fault received, exiting...

PC12138-Z, 2024-09-12:
新版本UA2F  开启之后会拦截经过路由的所有数据，我关了ua2f就正常上网了，不知道是不是不兼容的问题，我的另一个路由就是没事，是正常的，之前旧版本（应该是4.6或者4.7版本）就没事两个路由都能正常使用。

我发现这问题之后我以为是新版本缺少依赖，所以我编译新固件的时候把ua2f也编译进去了，结果问题依旧，查log显示正常开启，实际上也是上不了网，里面有一条报错Kernel does not support packet timestamps，不知道是不是内核的原因

[image]
[image]

Zxilly, 2024-09-12:
有哪里出bug了，但是不太清楚什么原因

Zxilly, 2024-09-12:
不过你是怎么编译的，为啥git相关的信息都是空的

PC12138-Z, 2024-09-12:
> 不过你是怎么编译的，为啥git相关的信息都是空的

问题就是打开ua2f就上不了网 ，我重新安装了一遍，第一张图的复现不了了，log全部都是第二张图那样  Signal Segmentation fault received, exiting... ，也是打开就上不了网

PC12138-Z, 2024-09-12:
> 不过你是怎么编译的，为啥git相关的信息都是空的

我把4.8.3到4.8.0版本都装了一遍，就4.8.0的不会显示git为空，4.8.3git为全空，4.8.2-4.8.1为Git branch 空。

最后一张为4.8.0，但是我安装4.8.0之后正常运行一会还是会报错 Signal Segmentation fault received, exiting...

[image]
[image]
[image]
[image]

PC12138-Z, 2024-09-12:
> 有哪里出bug了，但是不太清

换成4.7.1就正常了，4.8.0以上版本都有问题，启用ua2f后，路由之后的设备会断网，但是路由没断网，估计是把经过路由的数据都拦截了。
不知道是不是 Kernel does not support packet timestamps 的原因，因为4.7.1没有报错Kernel does not support packet timestamps，4.8.0的git不会为空，但是会报Kernel does not support packet timestamps

Zxilly, 2024-09-12:
`ulimit -c unlimited` 开启 coredump, 崩溃发生后上传 /tmp 目录下的 core 文件

Zxilly, 2024-09-12:
4.8.0 以上主要是加了一个fallback支持，可能哪里没写好

PC12138-Z, 2024-09-12:
> 命令行执行 `ulimit -c unlimited` 开启 coredump, 崩溃发生后上传 `/tmp` 目录下的 `.core` 文件和 `/usr/bin/ua2f`，打个 zip 包可以传 GitHub Issue 里

coredump 没有抓到东西，先把ua2f文件给你发了
[ua2f.zip](https://github.com/user-attachments/files/16979494/ua2f.zip)

Zxilly, 2024-09-12:
```
ulimit -c unlimited
echo "/tmp/%e-%p-%t.core" > /proc/sys/kernel/core_pattern
echo "1" > /proc/sys/fs/suid_dumpable
```

如果没有core文件，试试以上命令

PC12138-Z, 2024-09-12:
> ```
> ulimit -c unlimited
> echo "/tmp/%e-%p-%t.core" > /proc/sys/kernel/core_pattern
> echo "1" > /proc/sys/fs/suid_dumpable
> ```
> 
> 如果没有core文件，试试以上命令
还是没有core文件

Zxilly, 2024-09-12:
用最新分支上的代码重新编译一下吧，之前错误的捕获了Segmentation fault信号

PC12138-Z, 2024-09-13:
> 麻烦用最新分支上的代码重新编译一下吧，之前错误的捕获了Segmentation fault信号

试了还是报错，我想会不会是内核的原因，

我就把我另一个路由器刷了最新的分支编译的固件，内核从5.15.158升级到了5.15.164，装的ua2f最新版本，在没升级之前是不会报 git空 跟 Kernel does not support packet timestamps，升级之后就如下图，但是ua2f是正常起作用的，试了一下抓core文件也是没有

[image]

之前出问题的路由器内核也是最新5.15.164，估计是arm平台的问题了。

Zxilly, 2024-09-13:
等我有空了加个内置的错误追踪吧

JinXinYa, 2024-09-23:
同样的问题

Zxilly, 2024-09-23:
@JinXinYa 请上传 coredump 文件

JinXinYa, 2024-09-23:
没有coredump文件
[image]
[image]

Zxilly, 2024-09-23:
用最新的master分支上的代码编译不会捕获错误

JinXinYa, 2024-09-23:
好的好的，刚刚又重新编译了一份，之前git信息是空的，现在有了。目前没有问题
[image]

848657685, 2024-11-22:
> > 有哪里出bug了，但是不太清
> 
> 换成4.7.1就正常了，4.8.0以上版本都有问题，启用ua2f后，路由之后的设备会断网，但是路由没断网，估计是把经过路由的数据都拦截了。 不知道是不是 Kernel does not support packet timestamps 的原因，因为4.7.1没有报错Kernel does not support packet timestamps，4.8.0的git不会为空，但是会报Kernel does not support packet timestamps

感谢提供思路，安装旧版本就好了，跟59r 50m的东西 斗智斗勇几个晚上了

Zxilly, 2024-11-22:
有能力的话还是传一下coredump，我这里无法复现，bug还是得修

closed 2024-12-10 by Zxilly