Zxilly/UA2F #137: 可以加个...填上补齐后面的内容吗

alphime, 2024-09-17:
可以安装括号外判断空格有几个，一旦检测到...就补上空格后面的ua

alphime, 2024-09-17:
我尝试实现下，但是发现在handle_packet里面的ua内容始终都是第一次登录的ua信息，而不是来自发送请求的UA，这是为什么

alphime, 2024-09-17:
我是用的是函数里面的ua_start变量

Zxilly, 2024-09-17:
能不能再解释下你想要什么，你是想要按照语义分割UA？

alphime, 2024-09-17:
差不多，我自己实现了下遇到了个很奇怪的问题，不然你加我Q2328589870

alphime, 2024-09-17:
[image]
我是想自定义UA设置成Mozilla/5.0 (Windows NT 10.0; Win64; x64) 加请求UA的第一个括号`)`之后的内容，
但是我可以开始推测你代码ua_start变量是请求的ua内容，可我经验证之后，发现第一次连上网络的UA且一直保持不变
[image]
[image]

Zxilly, 2024-09-17:
你没理解怎么处理的，去看看libnetfilter-queue的例子吧

alphime, 2024-09-17:
你写的这个代码都是第一次的记录吗，要手动获取头部信息吗

closed 2024-09-18 by alphime