Zxilly/UA2F #138: 我尝试编译新版本使用出现如下问题

alphime, 2024-09-20:
[image]

Zxilly, 2024-09-20:
请参照 https://github.com/Zxilly/UA2F/issues/136 收集 core 文件上传

alphime, 2024-09-20:
大佬，不然你帮我完善修复下我新写的新功能吧，本人实在是尽力了，排查了很久都没解决

Zxilly, 2024-09-20:
我需要 coredump 文件才能确认错误

alphime, 2024-09-20:
https://github.com/alphime/UA2F/commit/e2c8e4624a047ca679a45da55dea3f2b4b9beb3d

alphime, 2024-09-20:
我感觉可能是我编译环境问题，你那边编译试下

Zxilly, 2024-09-20:
替换的字符串必须长度一致

closed 2024-09-20 by Zxilly

alphime, 2024-09-20:
> 替换的字符串必须长度一致

我是后面补空格，这个方案不行吗？你开辟的是65536空间，后面应该读的到吧

Zxilly, 2024-09-20:
请学习TCP协议中的ACK部分

alphime, 2024-09-20:
[image]
[image]
[image]

好像不是这个问题啊，一样出现问题

Zxilly, 2024-09-20:
你写的代码你问我干啥，你自己写的空指针

alphime, 2024-09-20:
好吧想寻求你帮助下，我自己来吧