Zxilly/UA2F #144: UA2F启动后短时间内异常退出

expliyh, 2024-11-30:
启动服务后立刻测试UA是改了的，等一小会（大概不到30s）就会退出，日志如下
Sun Dec  1 00:12:22 2024 syslog.info UA2F[31289]: UA2F version: 4.9.2
Sun Dec  1 00:12:22 2024 syslog.info UA2F[31289]: Git commit: 9ad2a3f1a2b08fe3fc7cc4d022c18136085c4c7d
Sun Dec  1 00:12:22 2024 syslog.info UA2F[31289]: Git branch: unknown
Sun Dec  1 00:12:22 2024 syslog.info UA2F[31289]: Git tag: v4.9.2
Sun Dec  1 00:12:22 2024 syslog.info UA2F[31289]: Statistics initialized.
Sun Dec  1 00:12:22 2024 syslog.info UA2F[31289]: Custom user agent string not set, using default F-string.
Sun Dec  1 00:12:22 2024 syslog.info UA2F[31289]: Handler initialized.
Sun Dec  1 00:12:22 2024 syslog.warn UA2F[31289]: Kernel does not support packet timestamps
Sun Dec  1 00:12:22 2024 syslog.info UA2F[31289]: Cache lock initialized
Sun Dec  1 00:12:22 2024 syslog.info UA2F[31289]: Cleanup thread created
Sun Dec  1 00:12:23 2024 daemon.info procd: Instance ua2f::ua2f s in a crash loop 11 crashes, 1 seconds since last crash

系统信息如图
[image]

Zxilly, 2024-11-30:
请参照 https://github.com/Zxilly/UA2F/issues/136 的评论收集 coredump 文件上传

Zxilly, 2024-11-30:
```
touch /.init_enable_core
sysctl -w "kernel.core_pattern=/tmp/%e.%p.%s.%t.core"
```

expliyh, 2024-12-01:
没有抓到coredump文件，系统日志和内核日志也没有错误

Zxilly, 2024-12-01:
那很难排查了

expliyh, 2024-12-01:
我在命令行里直接运行UA2F获取到了 coredump 文件，文件与命令行截图如下：

[image]
[ua2f.22775.11.1733055311.zip](https://github.com/user-attachments/files/17968879/ua2f.22775.11.1733055311.zip)

Zxilly, 2024-12-01:
如果路由器上能安装 gdb 的话，执行一下以下命令截图

```
gdb /usr/bin/ua2f ua2f.22775.11.1733055311.core
bt
```

expliyh, 2024-12-01:
[image]

[ua2f.zip](https://github.com/user-attachments/files/17969099/ua2f.zip)

Zxilly, 2024-12-01:
在最新的代码上启用 UA2F_DEBUG_BUILD 构建后调试试试

expliyh, 2024-12-01:
我 Fork 仓库后在CMakeList.txt中加入了 OPTION(UA2F_DEBUG_BUILD "Enable Debug" ON) 后使用Github Action编译后并重新运行，但是结果好像没有什么区别

[image]
[image]

Zxilly, 2024-12-02:
用现在最新的代码编译后应该能直接看到错误栈了

expliyh, 2024-12-02:
不好意思，我实在是不会编译这个包，直接用最新的Actions里面的测试输出和原来的一样，下面是编译的日志，图片是用最新Actions里面的包的结果

buildbot@0aeed9579568:~$ make package/ua2f/ compile -j1 V=s or V=sc
make: 'package/ua2f/' is up to date.
WARNING: Makefile 'package/feeds/packages/libssh/Makefile' has a dependency on 'zlib', which does not exist
WARNING: Makefile 'package/feeds/packages/libssh/Makefile' has a dependency on 'libmbedtls', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'eip197-mini-firmware', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'airoha-en8811h-firmware', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'r8169-firmware', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'e100-firmware', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'bnx2-firmware', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'bnx2x-firmware', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'ar3k-firmware', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'mwifiex-sdio-firmware', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-phy-bcm-ns-usb2', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'edgeport-firmware', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'r8152-firmware', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-phy-bcm-ns-usb3', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'amdgpu-firmware', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'i915-firmware-dmc', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'radeon-firmware', which does not exist
WARNING: Makefile 'package/ua2f/openwrt/Makefile' has a dependency on 'libnetfilter-conntrack', which does not exist
WARNING: Makefile 'package/ua2f/openwrt/Makefile' has a dependency on 'libnetfilter-queue', which does not exist
WARNING: Makefile 'package/ua2f/openwrt/Makefile' has a dependency on 'libuci', which does not exist
WARNING: Makefile 'package/ua2f/openwrt/Makefile' has a dependency on 'iptables-mod-conntrack-extra', which does not exist
WARNING: Makefile 'package/ua2f/openwrt/Makefile' has a dependency on 'iptables-mod-filter', which does not exist
WARNING: Makefile 'package/ua2f/openwrt/Makefile' has a dependency on 'iptables-mod-nfqueue', which does not exist
WARNING: Makefile 'package/feeds/packages/tmate/Makefile' has a dependency on 'libevent2', which does not exist
WARNING: Makefile 'package/feeds/packages/tmate/Makefile' has a dependency on 'libncurses', which does not exist
WARNING: Makefile 'package/feeds/packages/tmate/Makefile' has a dependency on 'terminfo', which does not exist
WARNING: Makefile 'package/feeds/packages/tmate/Makefile' has a build dependency on 'ncurses', which does not exist
make[2]: Entering directory '/builder/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/builder/scripts/config'
tmp/.config-package.in:66:warning: ignoring type redefinition of 'PACKAGE_libc' from 'bool' to 'tristate'
tmp/.config-package.in:93:warning: ignoring type redefinition of 'PACKAGE_libgcc' from 'bool' to 'tristate'
tmp/.config-package.in:207:warning: ignoring type redefinition of 'PACKAGE_libpthread' from 'bool' to 'tristate'
tmp/.config-package.in:262:warning: ignoring type redefinition of 'PACKAGE_librt' from 'bool' to 'tristate'
Config-build.in:1447:warning: defaults for choice values not supported
Config-build.in:1451:warning: defaults for choice values not supported
Config-build.in:1455:warning: defaults for choice values not supported
#
# No change to .config
#
make[1]: Entering directory '/builder'
make[1]: *** No rule to make target 'compile'.  Stop.
make[1]: Leaving directory '/builder'
make: *** [/builder/include/toplevel.mk:241: compile] Error 2

[image]

Zxilly, 2024-12-02:
[aarch64_generic-97f863becdb50ed2d0cb2f8f95e4e516321e4c8a-packages.zip](https://github.com/user-attachments/files/17978980/aarch64_generic-97f863becdb50ed2d0cb2f8f95e4e516321e4c8a-packages.zip)

用这个zip里的ipk

expliyh, 2024-12-03:
下面第一张截图是运行结果，此外第二章截图是我在重新安装UA2F时收到的提示

[image]

[image]

Zxilly, 2024-12-03:
执行一下 `ldd /usr/bin/ua2f`

expliyh, 2024-12-03:
[root@Expliyh-RTOP:05:29 PM ~] # ldd /usr/bin/ua2f 
        /lib/ld-musl-aarch64.so.1 (0xffffbd371000)
        libmnl.so.0 => /usr/lib/libmnl.so.0 (0xffffbd350000)
        libnetfilter_queue.so.1 => /usr/lib/libnetfilter_queue.so.1 (0xffffbd32f000)
        libnfnetlink.so.0 => /usr/lib/libnfnetlink.so.0 (0xffffbd30f000)
        libuci.so => /lib/libuci.so (0xffffbd2ee000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0xffffbd2bd000)
        libc.so => /lib/ld-musl-aarch64.so.1 (0xffffbd371000)
        libubox.so.20240329 => /lib/libubox.so.20240329 (0xffffbd29c000)

Zxilly, 2024-12-03:
```
uci set ua2f.main.disable_connmark=1
uci commit ua2f
```

看看有没有变化

expliyh, 2024-12-03:
没有变化

[image]

Zxilly, 2024-12-03:
崩溃以后，set $pc = $lr，然后 bt 看看能不能恢复现场

expliyh, 2024-12-03:
[image]

Zxilly, 2024-12-03:
https://github.com/Zxilly/UA2F/actions/runs/12138205207/artifacts/2266761123

改了一些，试一下新版本，感觉是 arm64 特定的 bug，x64 上的测试压根测不出来

expliyh, 2024-12-03:
我刚刚试了一下，不闪退了。明早再看看情况，感谢

expliyh, 2024-12-06:
没有再出现问题了

closed 2024-12-06 by expliyh