Zxilly/UA2F #145: 关于启动UA2F 报错 iptables: Chain already exists. ip6tables: Chain already exists.问题

3509903755, 2024-12-14:
我现在使用的是lede最新的源码来进行编译。编译完成后，软件包是有ua2f的，进入系统我发现ua2f没有启动，我就去手动启动，输入命令之后显示两行代码：iptables: Chain already exists.  和  ip6tables: Chain already exists.   目前不知道什么问题

Zxilly, 2024-12-14:
之前创建的链没有清理干净

3509903755, 2024-12-14:
服务里面没有UA2F这正常吗？能读取到UA2F的日志，但是没效果

Zxilly, 2024-12-14:
不正常，你应该检查自己的防火墙规则

3509903755, 2024-12-14:
现在有新问题了，SSH是显示UA2F没有运行，服务里面显示UA2F正在运行，测试网页UA为：FFFFFFFFFFFFFFFFFFFF，但是过一会就会被检测到共享网络，我们学校用的是哆点的验证

c1738348785, 2025-01-03:
> 之前创建的链没有清理干净

这个老版本更新后启动ua2f就会这样。会提示Received non-tcp packet. You may set wrong firewall rules.这个规则不知道是哪个/(ㄒoㄒ)/~~

Zxilly, 2025-01-03:
@c1738348785 你是哪个版本

c1738348785, 2025-01-03:
> @c1738348785 你是哪个版本
4.9.6,使用最新版就会防火墙自动配置不了，start会提示iptables: Chain already exists. ip6tables: Chain already exists.，后台日志就直接Received non-tcp packet. You may set wrong firewall rules.我使用老版本4.8.3就不会出现这个问题

closed 2025-03-13 by github-actions[bot]