Zxilly/UA2F #149: ua2f在immwrt24.10不生效，无报错，就是ua没没有被修改

AviderMin, 2025-02-16:
不止我一个有这个情况，不知道是不是ua2f的问题

Zxilly, 2025-02-16:
给的信息太少了，修不了

AviderMin, 2025-02-16:
设备是小米路由器4A千兆版，版本是24.10.0 (r32824-6a73dae98c9c)，ua2f是4.9.6-r1，在使用命令启动service ua2f start，命令行内无报错，但是ua未被修改，如何去找log日志，请告诉我，我去看看

AviderMin, 2025-02-16:
我是在imm的软件包里直接安装ua2f的，防火墙用的是nftables

AviderMin, 2025-02-16:
已经关闭了关闭全锥形NAT和软件流量分载

Zxilly, 2025-02-16:
启动ua2f以后把syslog打出来看看

AviderMin, 2025-02-17:
我现在在23.05，ua2f正常，我找时间到24.10复现

Yalimasnei, 2025-02-17:
我之前编译的时候也这样，后面切23.05再编译就正常了

DragonYH, 2025-02-23:
我也在24.10遇到了问题
`Sun Feb 23 15:38:25 2025 syslog.info UA2F[7023]: UA2F version: 4.9.6
Sun Feb 23 15:38:25 2025 syslog.info UA2F[7023]: Git commit: unknown
Sun Feb 23 15:38:25 2025 syslog.info UA2F[7023]: Git branch: unknown
Sun Feb 23 15:38:25 2025 syslog.info UA2F[7023]: Git tag: unknown
Sun Feb 23 15:38:25 2025 syslog.info UA2F[7023]: Statistics initialized.
Sun Feb 23 15:38:25 2025 syslog.info UA2F[7023]: Using embed user agent string: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/112.0.0.0 Safari/537.36 Edg/112.0.1722.68
Sun Feb 23 15:38:25 2025 syslog.info UA2F[7023]: Handler initialized.
Sun Feb 23 15:38:27 2025 syslog.warn UA2F[7023]: Kernel does not support packet timestamps
Sun Feb 23 15:38:27 2025 syslog.info UA2F[7023]: Cache lock initialized
Sun Feb 23 15:38:27 2025 syslog.info UA2F[7023]: Cleanup thread created
Sun Feb 23 15:38:27 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.
Sun Feb 23 15:38:31 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.
Sun Feb 23 15:38:31 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.
Sun Feb 23 15:38:37 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.
Sun Feb 23 15:38:43 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.
Sun Feb 23 15:38:43 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.
Sun Feb 23 15:38:43 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.
Sun Feb 23 15:38:43 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.
Sun Feb 23 15:38:43 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.
Sun Feb 23 15:38:43 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.
Sun Feb 23 15:38:44 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.
Sun Feb 23 15:38:44 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.
Sun Feb 23 15:38:45 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.`

DragonYH, 2025-02-23:
> 我也在24.10遇到了问题
> 
> ```
> Sun Feb 23 15:38:25 2025 syslog.info UA2F[7023]: Git commit: unknown
> Sun Feb 23 15:38:25 2025 syslog.info UA2F[7023]: Git branch: unknown
> Sun Feb 23 15:38:25 2025 syslog.info UA2F[7023]: Git tag: unknown
> Sun Feb 23 15:38:25 2025 syslog.info UA2F[7023]: Statistics initialized.
> Sun Feb 23 15:38:25 2025 syslog.info UA2F[7023]: Using embed user agent string: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/112.0.0.0 Safari/537.36 Edg/112.0.1722.68
> Sun Feb 23 15:38:25 2025 syslog.info UA2F[7023]: Handler initialized.
> Sun Feb 23 15:38:27 2025 syslog.warn UA2F[7023]: Kernel does not support packet timestamps
> Sun Feb 23 15:38:27 2025 syslog.info UA2F[7023]: Cache lock initialized
> Sun Feb 23 15:38:27 2025 syslog.info UA2F[7023]: Cleanup thread created
> Sun Feb 23 15:38:27 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.
> Sun Feb 23 15:38:31 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.
> Sun Feb 23 15:38:31 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.
> Sun Feb 23 15:38:37 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.
> Sun Feb 23 15:38:43 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.
> Sun Feb 23 15:38:43 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.
> Sun Feb 23 15:38:43 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.
> Sun Feb 23 15:38:43 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.
> Sun Feb 23 15:38:43 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.
> Sun Feb 23 15:38:43 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.
> Sun Feb 23 15:38:44 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.
> Sun Feb 23 15:38:44 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.
> Sun Feb 23 15:38:45 2025 syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.
> ```

似乎是新版本的防火墙改为了nftables，而不是老版本的iptables

Zxilly, 2025-02-23:
但是应该能够继续正常工作才对，只是会打印出警告，我做了额外的检测的。
而且为啥你这版本字符串全是空的？
24.10 是不是不用 opkg 了改用 apk 了？我怀疑是这个的问题。

DragonYH, 2025-02-23:
我直接在immortalwrt编译的，具体什么原因我也不清楚。opkg还是能用的

mcshijian, 2025-02-23:
我用kwrt编译的24.10日志也是一直syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.

Zxilly, 2025-02-26:
应该是24.10的兼容问题，但是我最近应该没空修，看看有哪个路过的有心情研究一下吧

mcshijian, 2025-02-26:
试了下在24.10中使用ua2f 4.9.6版本就会报syslog.warn UA2F[7023]: Received non-tcp packet. You may set wrong firewall rules.   而使用4.8.3就正常使用且没报错

Zxilly, 2025-02-27:
逻辑都是一样的，只是以前没有打这条错误日志

mcshijian, 2025-02-27:
但在24.10上4.8.3确实就能正常改ua了

Zxilly, 2025-02-27:
能正常工作吗？那我回头检查下

Yalimasnei, 2025-02-27:
当时我测的时候是运行了一下马上挂了

Keduoli03, 2025-03-01:
我也是，24.10无法使用，更换到4.8.3就可以更改ua成功了。最新包报错为上面的防火墙规则错误，不知道为什么

Zxilly, 2025-03-02:
试一下最新的 4.9.7，看看在 24.10 能不能正常工作。如果还是不行的话发一下 syslog

yigekuyou, 2025-03-03:
> 试一下最新的 4.9.7，看看在 24.10 能不能正常工作。如果还是不行的话发一下 syslog

syslog.err UA2F[12391]: Transport header was not set.
syslog.warn UA2F[12391]: Received non-tcp packet. You may set wrong firewall rules.

Zxilly, 2025-03-05:
现在最新的commit试一下，提供一下完整的日志

Zxilly, 2025-03-05:
在 `4.10.0` 中修复了

YeWenxuan64, 2025-03-05:
> 逻辑都是一样的，只是以前没有打这条错误日志

能不能加个设置隐藏此类log呀，小女子受够了它刷屏嘤嘤嘤，影响人家看日志喵

YeWenxuan64, 2025-03-05:
在非4.10.0版本的最新（和倒数第2，3，4，……新）的版本，7981的机器（在24.10版本的wrt）感觉ua2f工作得都不太行，正常工作的复现率很低（24年第四季度后基本不成功了，24.10怀疑增加）

Zxilly, 2025-03-05:
已经修复了

yigekuyou, 2025-03-06:
> 已经修复了，你可以自己看commit

过了一会就崩溃了

Thu Mar  6 08:08:03 2025 syslog.info UA2F[9242]: UA2F version: 4.9.7
Thu Mar  6 08:08:03 2025 syslog.info UA2F[9242]: Git commit: c7ffb02159d5ee3ef9c43f890d6b07307c9b5540
Thu Mar  6 08:08:03 2025 syslog.info UA2F[9242]: Git branch: unknown
Thu Mar  6 08:08:03 2025 syslog.info UA2F[9242]: Git tag: v4.10.0
Thu Mar  6 08:08:03 2025 syslog.info UA2F[9242]: Statistics initialized.
Thu Mar  6 08:08:03 2025 syslog.info UA2F[9242]: Using config user agent string: Test UA/1.0
Thu Mar  6 08:08:03 2025 syslog.info UA2F[9242]: Handler initialized.
Thu Mar  6 08:08:03 2025 syslog.warn UA2F[9242]: Kernel does not support packet timestamps
Thu Mar  6 08:08:03 2025 syslog.info UA2F[9242]: Cache lock initialized
Thu Mar  6 08:08:03 2025 syslog.info UA2F[9242]: Cleanup thread created
Thu Mar  6 08:08:28 2025 authpriv.info 
Thu Mar  6 08:08:35 2025 syslog.info UA2F[9242]: UA2F has handled 2 ua http, 20 http, 20 tcp. 91 ipv4, 0 ipv6 packets in 32 seconds.
Thu Mar  6 08:08:35 2025 syslog.info UA2F[9242]: UA2F has handled 4 ua http, 22 http, 22 tcp. 107 ipv4, 0 ipv6 packets in 32 seconds.
Thu Mar  6 08:08:36 2025 syslog.info UA2F[9242]: UA2F has handled 8 ua http, 34 http, 34 tcp. 146 ipv4, 0 ipv6 packets in 33 seconds.
Thu Mar  6 08:08:36 2025 syslog.info UA2F[9242]: UA2F has handled 16 ua http, 57 http, 57 tcp. 191 ipv4, 0 ipv6 packets in 33 seconds.
Thu Mar  6 08:08:45 2025 syslog.info UA2F[9242]: UA2F has handled 32 ua http, 94 http, 94 tcp. 422 ipv4, 0 ipv6 packets in 42 seconds.
Thu Mar  6 08:08:50 2025 daemon.info procd: Instance ua2f::ua2f s in a crash loop 6 crashes, 48 seconds since last crash
Thu Mar  6 08:08:50 2025 kern.info kernel: [ 7717.944348] do_page_fault(): sending SIGSEGV to ua2f for invalid write access to 00000000
Thu Mar  6 08:08:50 2025 kern.info kernel: [ 7717.944402] epc = 77e5c1b4 in libc.so[77e2a000+b2000]
Thu Mar  6 08:08:50 2025 kern.info kernel: [ 7717.944473] ra  = 77e5c710 in libc.so[77e2a000+b2000]

Zxilly, 2025-03-06:
wtf, 怎么还 page fault 了

Zxilly, 2025-03-06:
参照一下 https://github.com/Zxilly/UA2F/issues/144， 看看能不能收集到 coredump 文件

Zxilly, 2025-03-06:
试一下最新的commit，如果还是崩溃提供一下完整的log

yigekuyou, 2025-03-07:
> 试一下最新的commit，如果还是崩溃提供一下完整的log

没有崩溃了.

Zxilly, 2025-03-10:
暂时没有看到进一步回报，先关闭了

closed 2025-03-10 by Zxilly