Zxilly/UA2F #151: ua2f在immwrt24.10不生效，无报错，后台日志分析不出来问题出现在哪

BaicaiD, 2025-03-10:
从最新版换回ua2f | 4.8.3-r1就正常使用了
Mon Mar 10 22:17:42 2025 daemon.err ua2f[6946]: Received signal Segmentation fault (11)
Mon Mar 10 22:17:42 2025 daemon.err ua2f[6946]: Backtrace:
Mon Mar 10 22:17:49 2025 daemon.err ua2f[6949]: Received signal Segmentation fault (11)
Mon Mar 10 22:17:49 2025 daemon.err ua2f[6949]: Backtrace:
Mon Mar 10 22:17:56 2025 daemon.err ua2f[6953]: Received signal Segmentation fault (11)
Mon Mar 10 22:17:56 2025 daemon.err ua2f[6953]: Backtrace:
Mon Mar 10 22:18:11 2025 daemon.err ua2f[6956]: Received signal Segmentation fault (11)
Mon Mar 10 22:18:11 2025 daemon.err ua2f[6956]: Backtrace:
Mon Mar 10 22:18:21 2025 daemon.err ua2f[6961]: Received signal Segmentation fault (11)
Mon Mar 10 22:18:21 2025 daemon.err ua2f[6961]: Backtrace:
Mon Mar 10 22:18:21 2025 daemon.info procd: Instance ua2f::ua2f s in a crash loop 6 crashes, 5 seconds since last crash
Mon Mar 10 22:17:40 2025 syslog.warn UA2F[6946]: Kernel does not support packet timestamps
这个是在最新版本不生效时我出现的问题，我不是很懂这个也算是反馈一下吧（sos）。换为4.8.3就好使了

Zxilly, 2025-03-10:
[image]

Backtrace 是空的吗，应该是这样的才对。

Zxilly, 2025-03-10:
执行一下 `file /usr/bin/ua2f` 看一下

BaicaiD, 2025-03-10:
好的，我仅查看了一下系统日志，
具体报错，等今天上完班，再来复现一下。
到时候发给您。：）

BaicaiD, 2025-03-11:
[image]

Zxilly, 2025-03-11:
挺奇怪的，看着像是启动就失败了，你是过滤了logread的日志等级吗

Zxilly, 2025-03-11:
然后你file没装，opkg update 然后 opkg install file再执行

BaicaiD, 2025-03-11:
下午回去看看，反正就是频繁崩溃尝试启动又崩溃。ok我是说怎么看不了堆栈。不清楚会不会可能配置文件错误，我回去捞堆栈给你看ദ്ദി˶•̀֊•́)✧

BaicaiD, 2025-03-11:
/usr/bin/ua2f: ELF 32-bit LSB executable, MIPS, MIPS32 rel2 version 1 (SYSV), dynamically linked, interpreter /lib/ld-musl-mipsel-sf.so.1, no section header

Zxilly, 2025-03-11:
你怎么构建的，这不太对，没有调试信息了。

BaicaiD, 2025-03-12:
emmmm，也许是我个例吧，我换回老的版本确实好使了，不排除我设备兼容性？有明显找到问题我再提交吧

Zxilly, 2025-03-12:
开着吧，我怀疑是32位的问题，但我现在没设备

Zxilly, 2025-04-14:
错误是第三方构建导致的

closed 2025-04-14 by Zxilly