Zxilly/UA2F #152: openwrt反复报错Received signal Segmentation fault (11)

hmho000, 2025-04-13:
在openwrt运行ua2f反复如图重启
[image]

使用固件如下
[image]

hmho000, 2025-04-14:
[image]使用的小米AX6

> 系统是什么版本和架构的？

Zxilly, 2025-04-14:
非官方发行版上的不太好复现，libbacktrace也寄了

Zxilly, 2025-04-14:
你是编译安装的还是从GitHub下载的ipk？

hmho000, 2025-04-14:
> 你是编译安装的还是从GitHub下载的ipk？

很抱歉现在才回复，之前在上课，我是在[openwrt.ai](https://openwrt.ai/?target=qualcommax%2Fipq807x&id=redmi_ax6)上直接[编译的软件包](https://wwb.lanzoum.com/ieJcs2tk4msb)，在配置好校园网验证之后我从opkg进行了ua2f的软件包升级，opkg配置如下

[image]

在禁用wan口的情况下打开ua2f如下

[image]

[image]

打开wan口后ua2f开始重启

[image]

Zxilly, 2025-04-14:
什么叫做 “从opkg进行了ua2f的软件包升级”，ua2f 并不在 openwrt 的官方源中

Zxilly, 2025-04-14:
我怀疑你应该联系 `openwrt.ai`，我不清楚他是怎么编译的 UA2F

Zxilly, 2025-04-14:
我明白了

openwrt.ai 的包是 https://github.com/kiddin9/kwrt-packages，其中的 UA2F 是从 https://github.com/immortalwrt/packages 同步的，同步时只强行更新了版本，而没有处理 Makefile 的变化，因此无法正常工作。

cc @1715173329

1715173329, 2025-04-14:
immortalwrt 仓库的包已经更新，至于用某 ai 的只能祝他们幸福。

closed 2025-06-21 by github-actions[bot]